Bodmer/TFT_eSPI #470: TFT_eSPI.h:665:3: error: stray '\302' in program

KlausMu, 2019-11-11:
I'm using the Arduino IDE 1.8.10 with TFT_eSPI 1.4.20

I get the following error:
`C:\Users\xxxxx\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:665:3: error: stray '\302' in program
   TFT_eSPI(int16_t _W = TFT_WIDTH, int16_t _H = TFT_HEIGHT);
   ^
C:\Users\xxxxx\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:665:3: error: stray '\240' in program`

I know that strays are coming from weird characters (wrong encoding), but I can't find any unusual character at that position. I'm using an unchanged version of the file "TFT_eSPI.h".

Bodmer, 2019-11-11:
I installed a new copy of IDE 1.8.10 and a new copy of TFT_eSPI 1.4.20 from the Github zip file and  found examples compiled without any error.

I see on the web that others have seen this issue in other libraries but it is not clear why. Some problems seem to have been resolved by reloading the library. Did you use the IDE download or a direct zip copy off Github?

KlausMu, 2019-11-11:
Thanks for your reply. I tried again with a fresh install - and it worked.
The reason for the error was a change made by me in the file "User_Setup.h" where some unsupported characters where introduced by copy and paste in the line "#define TFT_WIDTH  80".

closed 2019-11-11 by KlausMu

Bodmer, 2019-11-11:
I use Notepad++ for editing, it is very good and has syntax highlighting. Also it does not let illegal characters like a non-breaking space to be inserted.

Glad it is solved.

KlausMu, 2019-11-11:
I'm also using Notepad++, but I copied text from a browser. In that case Notepad++ seems to let them in.