Bodmer/TFT_eSPI #471: 2 Display

alwan12e3, 2019-11-13:
Hi Bodmer,

Im working on the project which include controlling 2 tft 2.8 inch IL9341 display using esp 32. To do this you once said to use the same pin for both display except the CS pin, but,

How can we actually control the chip select?

I've try to commented out the TFT_CS defnition in user_setup and move it in esp32 program in two ways. First is using integer variable called TFT_CS and use #define TCT_CS. Both didnt work, only a blank white screen.

I'm assuming we need to tft.init for every changes in both display with each CS?
Can we just tft.init once for both display? Because tft.init restarts the display and gives a noticeable blank white screen pause for each iteration

Bodmer, 2019-11-13:
Connect the chip select lines of the TFTs to different GPIO pins. In you sketch drive those pins using the standard Adruino digitalWrite function. To select the TFT drive that TFT chip select pin LOW, then issue the tft library comand you want to send to that TFT, then drive the chip select high.

In setup you can initialise both displays simultaneously by driving both chip selects LOW, then run tft.init() then drive both or one HIGH. BUT in this case you MUST leave the MISO lines from the TFT unconnected. The MISO lines only need to be connected if your read the display, in most applications you do not need to read the display memory. You can init() one then the other if you wish to connect the MISO signals, but then NEVER drive both chip selects low, otherwise the two MISO outputs will be driving each other, which is not desirable.

If you read up on how SPI works then you will understand what you are trying to do and why the above rules apply.
You only have to run init()

alwan12e3, 2019-11-14:
About the chip select, how should I define TFT_CS in user_setup.h? 
About the GPIO for CS pins,  when I want to drive one of them, set the undesired pin to HIGH and desired pin to LOW, after that issue the tft command and right after the command, I set the desired CS pin to HIGH and the undesired pin to LOW?

Bodmer, 2019-11-14:
Do not define TFT_CS at all in the setup.

Set both chip selects high, drive one low only when you want to talk to that display, then drive high immediately afterwards.

When you are not talking to the display BOTH chip selects should be high, this allows other SPI devices to be connected to the SPI bus and used, eg touch controller or SD card.

closed 2019-11-14 by Bodmer