Bodmer/TFT_eSPI #472: SPI Pins

hasenradball, 2019-11-16:
Hi,
just a short question.
How doe I make it clear that I am using the Hardware SPI Pins on the ESP8266 and the ESP32.
I recognized that ot is an difference if I use:
```// Use hardware SPI (on Uno, #13, #12, #11) and the above for CS/DC
Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC);```
or this
```
// If using the breakout, change pins as desired
Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_MOSI, TFT_CLK, TFT_RST, TFT_MISO);```

Bodmer, 2019-11-16:
You are using an Adafruit library, I do not provide support for it.

TFT_eSPI only uses hardware SPI.

closed 2019-11-16 by Bodmer

hasenradball, 2019-11-17:
Ok thank you, I just want to know if the eSPi Lib uses Hardware or Software SPI.

Can you say if the following is possible:
define a specified Display area with:
- background
- temperature and Humidity values ans so on
-> define first the hole area without drawing, and if it is done draw the complete area with special drawing methods like:
 fade in, fade out, drwa it from left to write and so on...

Like define an small object but draw it then with special effects...