Bodmer/TFT_eSPI #477: ESP-32F Development Board with 1.44" 128x128 TFT LCD white screen

poky, 2019-11-22:
I have a [ESP-32F dev kit which came with the TFT LCD](https://www.aliexpress.com/item/33025684641.html) from Aliexpress

The docs pin as follow:
[image]

So I over-write the User_Setup.h setting as:

```
#define TFT_CS   26
#define TFT_DC   17
#define TFT_RST  16
#define TFT_MISO -1
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_BL 13
```

Even though the bag that came with the screen says it has ST7735 driver, but I can't get it to lit up.
Finally, I've tried all the listed driver, but only 
`#define ST7789_DRIVER`
I can see the white screen, but nothing else...
[image]

Can anyone suggest what I can try next? Many Thanks!

Bodmer, 2019-11-27:
Run the [diagnostic test](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Test%20and%20diagnostics/Read_User_Setup) and see if you have the correct setup being picked up by the compiler. Output is to serial port.

poky, 2019-11-27:
Finally got it sort out!
The problem was that the ST7789_DRIVER doesn't have the TFT_BL enable, the image is there, but can't lit up.
I have added the def TFT_BL to ST7735_Init.h, and sent you a pull request, this issue may closed, thanks!

closed 2019-11-27 by poky