Bodmer/TFT_eSPI #483: draw gradient line

tobozo, 2019-11-28:
Hey @Bodmer and thanks for your marvelous library.

Would you be interested in a PR that implements gradient lines?

https://github.com/tobozo/ESP32-Chimera-Core/pull/6/commits/1c2c182e6ea6670c7973b8908c44ce2a18ca572e

[image]

Bodmer, 2019-11-29:
Yes, I am interested but not within the TFT_eSPI library. A good place would be in the graphics extension library [TFT_eFEX.](https://github.com/Bodmer/TFT_eFEX) via a pull request. A simple example will also be needed to show users how to use the functions.

tobozo, 2019-11-29:
ok the example has been created and a PR ic available here https://github.com/Bodmer/TFT_eFEX/pull/10 

but I'm not happy with the speed so there's probably room for improvement

closed 2019-11-30 by Bodmer