Bodmer/TFT_eSPI #484: Connection to Wemos D1

stm01, 2019-11-28:
Tried several Pins but it was never exactly clear to me, which connections must be made if i want to use the i.e. ILI9488 to a ESP8266 on a Wemos D1.
Read already through the User Setups but afaik 3 Pins only dont seem to work for SPI.
Can someone please provide help?

Bodmer, 2019-11-29:
Some Wemos board have different pin mappings to typical ESP8266 boards.

Try using Dx instead of PIN_Dx for the pin allocations in the setup file.

Bodmer, 2019-11-29:
[These are the mappings ](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L129-L149)used by TFT_eSPI when PIN_Dx is used for pin names. Check against your boards documentation.

Bodmer, 2019-11-29:
[This setup is a good starting point](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup20_ILI9488.h). You need to connect up MOSI and SCK too [(see this list)](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L98-L106) but do not define those pins in the setup file.

MISO is not needed unless you wish to read from the display.

stm01, 2019-11-29:
Thanks in advance for the information you provided.
Tried my 3,5" ILI9488 today w/ a Nodemcu but could not get anything to display.
Checked all the UserSetup Defines and connections more than twice, but no luck for me.
Will have to dig into this over christmas holidays again.
If anyone is interested, the display im using is exactly this one: 
http://www.lcdwiki.com/3.5inch_SPI_Module_ILI9488_SKU:MSP3520

Thanks for providing the infos above :)

Bodmer, 2019-11-29:
Thanks for the link. An unusual vendor because they provide good information and simple Arduino sketches to test the display. Some vendors get information wrong. I have not used an ILI9488 in a while so to check the library is still OK I tested my ILI9488 ([ER-TFTM035-6](https://www.buydisplay.com/default/lcd-3-5-inch-320x480-tft-display-module-optl-touch-screen-w-breakout-board)) with an ESP32 (setup 21) and a ESP8266 NodeMCU (Setup 20).  The NodeMCU was connected as follows:
```
// Display SDO/MISO  to NodeMCU pin D6 (or leave disconnected if not reading TFT)
// Display LED       to NodeMCU pin VIN (or 5V, see below)
// Display SCK       to NodeMCU pin D5
// Display SDI/MOSI  to NodeMCU pin D7
// Display DC (RS/AO)to NodeMCU pin D3
// Display RESET     to NodeMCU pin D4 (or RST, see below)
// Display CS        to NodeMCU pin D8 (or GND, see below)
// Display GND       to NodeMCU pin GND (0V)
// Display VCC       to NodeMCU 5V or 3.3V
//
```
The "TFT_Rainbow480" test sketch is shown running OK.
[image]

Try the Read_User_Setup sketch to verify the compiler is picking up your setup file. Output is to the Serial Monitor window only.

stm01, 2019-12-02:
Interesting pic btw, yours seem to have much more pins than mine (14 in total).
Will do testing this week :)

Bodmer, 2019-12-15:
I assume testing went well, so closing this issue.

closed 2019-12-15 by Bodmer