Bodmer/TFT_eSPI #485: Sd card (bmp, jpg) during drawing,turns off

Andr1961, 2019-11-29:
Hi,
Thank you for the excellent library,
But there is a problem with the Sd card,
I run the same example from bmp to
Library adafruit and everything works,
fnd on the tft_espi on the first file draws a little, and then probably SD card is disconnected and writes the "file is not found".
Sketch bmp is the same only in adafruit - tft.writePixel(tft.color565(r,g,b));
And here tft. pushColor (tft.color565 (r, g, b));
Can we fix this somehow?
[image]

Bodmer, 2019-11-30:
I need to know what sketch you are trying to run, if you have created or modified a sketch then I need to have a copy to test.
What hardware are you using? ESP32 or ESP8266. What setup or pins are you using to the display and SD card. What version of the board package? What version of the TFT_eSPI library?

Andr1961, 2019-11-30:
[TFT_SD_BMP.ZIP](https://github.com/Bodmer/TFT_eSPI/files/3906971/TFT_SD_BMP.ZIP)

sketch attached, it just choose library
#define tft_ips
//#define adafruit
with ESP 32 does not work, now I try on ESP8266
Setup1_ILI9341.h :
#define TXT_MOSI D7
#define TFT_MISO D6
#define TFT_SCLK D5
#define TFT_CS PIN_D8 // Chip select control pin D8
#define TFT_DC PIN_D3 // Data Command control pin
/ / #define TFT_RST PIN_D4 / / Reset pin (could connect to NodeMCU RST, seek next line)
#define TT_ST -1 // St FT_RST to -1 if the display RESET is connected to Node MSU Rector of 3.3 V
the Board version 2.5.2--> TFT_ESPI 1.4.4
the Board version 2.6.2--> TFT_ESPI 1.4.21
[TFT_eSPI.zip](https://github.com/Bodmer/TFT_eSPI/files/3906975/TFT_eSPI.zip)
[adafruit.zip](https://github.com/Bodmer/TFT_eSPI/files/3906976/adafruit.zip)

Bodmer, 2019-12-01:
I loaded your sketch, configured TFT_eSPI for an ESP32, loaded the parrot image on the SD card, uploaded the sketch and it worked straight away.
To advise further I need to see the back of the TFT board and your wiring to the display, the reason is I want to see what circuitry is on the back of the display.
Also provide the output of the  "Test and diagnostics"  "Read_User_Setup" example.
Did the ESP8266 work?

Andr1961, 2019-12-03:
Hi, thank you for the quick response,
Did not respond immediately should have checked the ESP32.
------ESP32-----
1.
sd hspi
tft hspi
Not working.
2.
sd vspi
tft hspi
Not working.
3.
sd vspi
tft vspi
Works.
----ESP8266------
If SD is earlier, SD ok, SD does not see files.
If the TFT is earlier, SD ok, draws a block from the first file, no further.

"Test and diagnostics"
[code]
TFT_eSPI ver = 1.4.21
Processor    = ESP8266
Frequency    = 80 MHz
Voltage      = 3.23 V
Transactions = No
Interface    = SPI
SPI overlap  = No

Display driver = 9341
Display width  = 240
Display height = 320

MOSI    = D7 (GPIO 13)
MISO = D6 (GPIO 12) (disabled)
SCK     = D5 (GPIO 14)
TFT_CS   = D8 (GPIO 15)
TFT_DC   = D3 (GPIO 0)
Display SPI frequency = 40.0 MHz
[/code]

SD Pins
MOSI    = D7 (GPIO 13)
MISO    = D6 (GPIO 12)
SCK     = D5 (GPIO 14)
SD_CS  D1 (GPIO 5)
---------
TFT_CS,TFT_DC,SD_CS changed, didn 't help,
I have TFT ILI9341 and ILI9488,
The PINs on them are distributed equally.
Tried both do not work, on ESP32 work (on VSPI)

This project I still have for Setup13_ILI9481_Parallel.h,
Has not yet tried
[file.zip](https://github.com/Bodmer/TFT_eSPI/files/3918441/file.zip)

Bodmer, 2019-12-03:
OK, if I understand your comments - you now have the ESP32 working OK. The VSPI port is the default for TFT_eSPI and it is likely the SD library will only work on this port. TFT_eSPI will work on either port. As [noted here](url), the HSPI port should only be used with TFT_eSPI if the VSPI port is in use for other purposes. The VSPI port is the default used by the supporting Arduino SPI library for the ESP32.

Thanks for the ESP8266 setup report. The ESP6266 needs transaction support enabled to work with the SD library. [Enable this in the setup ](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L283)by removing //

Andr1961, 2019-12-03:
Just super!
Thank you very much for your help and
Excellent library,
good luck

closed 2019-12-03 by Andr1961