Bodmer/TFT_eSPI #486: ST7796S driven 4.0" LCD Display support

Hans007a, 2019-12-01:
Hello,

I have this display: [MHS-4.0](http://www.lcdwiki.com/MHS-4.0inch_Display-B).

Can You add support for this please?

Thank You!

Hans

Bodmer, 2019-12-01:
The good news is the answer is yes, I will add support.

The bad news is that I am very busy with paid for work at the moment and thus it will probably be January before I have time to release a test version.

I am also in the process of completing a major update to TFT_eSPI to support other processors and provide DMA transfer for SMT32 processors, and that task has priority at the moment.

peppeve, 2019-12-20:
Any update? @Hans007a  you have solved?

Bodmer, 2020-01-10:
Good news. I have a working version on the ESP32. Performance is very good for a SPI display as it is working fine at 80MHz SPI clock rate (which is the maximum SPI clock rate for an ESP32).

I anticipate I will be able to push the changes to Github hopefully within 24hrs after testing with the ESP8266.

Bodmer, 2020-01-11:
Display support now added to the Master branch.

Please raise any problems as a NEW issue.

closed 2020-01-11 by Bodmer