Bodmer/TFT_eSPI #487: Character background issue.

Fredy12335, 2019-12-03:
First of all thanks for the lib. The best I've ever used. I like it!
When I overwrite another character that is already on the screen, the background of the current character, or glyph, is not cleared, and so the previous character shows through...
 I set to background color black tft.setTextColor(TFT_WHITE, TFT_BLACK); I generated font with Processing 3. To get a better view, I set the screen background to yellow and the font background black. (sorry my english is not the best)

`tft.setTextWrap(true, true);
  String fileName = "Cambria-Bold-32";
  tft.loadFont(fileName);
  tft.setTextColor(TFT_WHITE, TFT_BLACK);
  tft.setCursor(230, 14);
  tft.println(temp);`

[image]

Fredy12335, 2019-12-04:
I forget I use ILI9341 (SPI) with Wemos D1 mini pro, (ESP8266).

Fredy12335, 2019-12-04:
I found this solution in Font_Demo_1 
`tft.fillRect (50, 90, 60, 40, TFT_BLACK); // Overprint with a filled rectangle
 tft.setTextColor(TFT_GREEN, TFT_BLACK);
 tft.setCursor(50, 90);`
But it caused flickering. Is there any easier way?

Bodmer, 2019-12-04:
Look at the other Font_Demo_2/3/4 examples and you will find a solution. Thesketch  header comments provide help too.

Fredy12335, 2019-12-04:
Thanks i'm sorry to bother you with such a basic thing. But I think we really need a good documentation. In fact everything is in it the examples, but it's often not easy to find what I'm looking for. Thanks again your jobs and time.

closed 2019-12-04 by Fredy12335

Bodmer, 2019-12-04:
No problem. documentation is now high on the TO DO list see #415