Bodmer/TFT_eSPI #489: Possible speed-up of anti-aliased text drawing

delhoume, 2019-12-05:
The core of the loop for this is 
drawPixel(x + cx, y + cy, alphaBlend(pixel, fg, bg));

In the general case fg and bg are fixed by setTextColor() and will not change during drawGlyph()

Could this be used to speed-up the alphaBlend function, for example with a lookup table ?
Or pre-computing some terms that will never vary, like fgR, fgG fgB bgR bgG and bgB.
Or maybe use pre-multiplied alpha (https://microsoft.github.io/Win2D/html/PremultipliedAlpha.htm) ? 

Any speed-up would be appreciable I think.

delhoume, 2019-12-05:
I found this formula on Google that gives good visual results 👍 
```
/**
 * Fast RGB565 pixel blending
 * @param fg      The foreground color in uint16_t RGB565 format
 * @param bg      The background color in uint16_t RGB565 format
 * @param alpha   The alpha in range 0-255
 **/
uint16_t alphaBlendRGB565( uint8_t alpha, uint32_t fg, uint32_t bg){
    alpha = ( alpha + 4 ) >> 3;
    bg = (bg | (bg << 16)) & 0b00000111111000001111100000011111;
    fg = (fg | (fg << 16)) & 0b00000111111000001111100000011111;
    uint32_t result = ((((fg - bg) * alpha) >> 5) + bg) & 0b00000111111000001111100000011111;
    return (uint16_t)((result >> 16) | result);
}
```

Bodmer, 2019-12-05:
Thanks for the suggestion. It is one that I have previously tried. It is about 10% faster than the method used but is makes negligible difference to smooth font rendering speed as that is dominated by other factors. The bad news is that the faster algorithm throws away the extra resolution in the green channel and this increases the banding in colour gradients. The eye is more sensitive to green, so the extra resolution of that colour is beneficial. Of course for antialiased fonts this loss of resolution would not be noticeable, but I use the same alphaBlend function for other purposes.

Bodmer, 2019-12-05:
Forgot to mention, if you wish to see how much speed could be gained for an infinitely fast alphaBlend algorithm, just replace the function with the foreground colour.

closed 2019-12-05 by Bodmer

Bodmer, 2019-12-06:
Thought on this further and implemented it like this to retain the green resolution:
```
    alpha = ( alpha + 2 ) >> 2;
    uint32_t bg = bgc & 0b0000011111100000;
    uint32_t fg = fgc & 0b0000011111100000;
    uint32_t result = ((((fg - bg) * alpha) >> 6) + bg) & 0b0000011111100000;

    bg = bgc & 0b1111100000011111;
    fg = fgc & 0b1111100000011111;
    result |= ((((fg - bg) * alpha) >> 6) + bg) & 0b1111100000011111;
    return result;
```
This runs ~7% faster than the original but will have negligible effect to the smooth font rendering.

Bodmer, 2019-12-06:
Tests reveal a flaw in the rounding. So now switched back to the original algorithm!

Bodmer, 2023-01-15:
Have added more smooth graphics which use alpha blending so I looked at this again. Here is the new faster algorithm that retains the green resoloution. This provided a few percent gain mainly with sprites:
```
uint16_t TFT_eSPI::alphaBlend(uint8_t alpha, uint16_t fgc, uint16_t bgc)
{
  // Split out and blend 5 bit red and blue channels
  uint32_t rxb = bgc & 0xF81F;
  rxb += ((fgc & 0xF81F) - rxb) * (alpha >> 2) >> 6;
  // Split out and blend 6 bit green channel
  uint32_t xgx = bgc & 0x07E0;
  xgx += ((fgc & 0x07E0) - xgx) * alpha >> 8;
  // Recombine channels
  return (rxb & 0xF81F) | (xgx & 0x07E0);
}
```