Bodmer/TFT_eSPI #490: International font ( Greek ) support

karthan, 2019-12-07:
Hi Bodmer
After our last communication on October 22, I followed your suggestions and downloaded your Adafruit-GFX-Library  and compiled the fontconvert.exe following your instructions.
I created 2 font files: First with Latin characters from 32 to 126, second from 902 to 974 ( only greek characters).
Question 1: Is it possible to merge these 2 files into one?
What i really need is to create a .h font file which will contain following characters:
From position  32 to 126 Latin characters.
From position 134 to 206 the greek characters which are in position 902 to 974 in ttf font.
Question 2: Is it possible to modify the file fontconvert.c to accept two more arguments (first2 last2) and having so the ability to create a single font file containing both character sets?
Or somewhere in fontconvert.c an additional  check  excluding the creation of characters in a given range?
Thank you
karthan

Bodmer, 2019-12-09:
The Adafruit_GFX font format requires a contiguous code point value character set so the library software would need to be adapted to support non-contiguous code blocks.

You could adapt fontconvert.c to concatenate the code blocks into the range 32 to 206 with 127-133 set to be empty characters, but this is not somenting I can help you to do.

The smooth font format does support non-contiguous code blocks or randomly ordered characters.

karthan, 2019-12-10:
Hi
Thank you for your response.
A modified fontconvert.c with first=32 and last=974 which will 
1. create fonts 32 to 133
2. Skip creating fonts 134 to 901
3. Create fonts 902 to 974 and placing them in position 134 to 206 creating a single font file.
What do you think about this approach?

karthan, 2019-12-14:
Hi
Another question:
Is there a file size limit for fonts files in the library?

Bodmer, 2019-12-15:
Yes, that approach would create a contiguous font that could be rendered. However you would need to adapt the UTF-8 decoded point code to map the range at 902 to 134, for example add a line before[ this one](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.cpp#L4049) such as:
```
if (decoderBuffer > 901) decoderBuffer -= (902 - 134);
```
The size is limited by the FLASH memory size.

karthan, 2019-12-16:
Thank you for your response.
I will try it.

closed 2019-12-16 by Bodmer