Bodmer/TFT_eSPI #491: How to use ST7789 1.14" 135x240

giga300, 2019-12-07:
Hello,

I want use a 1.14" (RGB) 135x240 IPS ST7789V 4 Wire SPI Port with NodeMCU. I found this issue https://github.com/Bodmer/TFT_eSPI/issues/379 therefore this library is normaly compatible with my screen.

My screen have GND, VCC, SCL, SDA, RES, DC, CS and BLK pins and I don't have any ideas to how to use it.

Thanks you

Bodmer, 2019-12-09:
Edit User_Setuo_select.h to call up Setup135_ST7789.h, read the comments for help.

Edit the Setup135_ST7789.h file in the User_Setups folder to use the following pins:

// ESP8266 and displays with chip select
#define TFT_CS   PIN_D8  // Chip select control pin D8
#define TFT_DC   PIN_D3  // Data Command control pin
#define TFT_RST  PIN_D4  // Reset pin (could connect to NodeMCU RST, see next line)

Connect NodeMCU to display as follows:

// Display GND       to NodeMCU pin GND (0V)
// Display VCC       to NodeMCU 5V or 3.3V// Display LED       to NodeMCU pin VIN (or 5V, see below)
// Display SCL       to NodeMCU pin D5
// Display SDA       to NodeMCU pin D7
// Display RESET     to NodeMCU pin D4
// Display DC        to NodeMCU pin D3
// Display CS        to NodeMCU pin D8
// Display BLK       leave disconnected (can drive low to turn off backlight)

giga300, 2019-12-10:
Thanks for info, my display is not black now but there are problems again. I follow this tutorial to show a image to my display https://www.hackster.io/148468/display-images-on-a-st7789-screen-fbb0d7

This is my code:

```c
#include <TFT_eSPI.h>

TFT_eSPI tft = TFT_eSPI();

void setup() {
  Serial.begin(115200);
  Serial.print("ST7789 TFT Bitmap Test");

  tft.begin();
  tft.setSwapBytes(true);

  tft.fillScreen(TFT_BLACK);
  tft.pushImage(0,0,135,240,mercy); // mercy is a LCD converted image
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

And I get (I want show this https://picsum.photos/id/400/135/240):

https://i.imgur.com/AdsQjx7.jpg

You have any solutions ?

Bodmer, 2019-12-10:
Try the examples that come with yhe library first to mske sure the connections and configuration are OK.

giga300, 2019-12-14:
It doesn't work with example, always sidebar on the left

Bodmer, 2019-12-15:
Please post the converted image file and sketch as a zip file for me to test. To post a zip file you just need to drop it on the Git text window. IT looks like the converted image may need a screen rotation to be set and swap bytes changed to false, viz:  tft.setSwapBytes(false); this is because of the endianess of the images 16 bit data being different.

Some library examples will not work well as they are written for much larger displays.  TFT_Flash_Bitmap should work OK as the initially drawn three icons are centered on any screen size.

Bodmer, 2020-01-11:
Closing as without feedback this will not be solved.

closed 2020-01-11 by Bodmer

OptrisBeginner, 2020-09-07:
I have a TFT LCD "EA TFT028-AI" which it has ST7789v controller. I am using BF706 processor, that supports TFT, RGB interface. In my code, there are some configurations for the LCD.
How can I use ST7789 controller in my project?, How can I use initialization example for the controller in my project?
If someone can please help me. 
Thank you.