Bodmer/TFT_eSPI #492: FS.h missing?

wmacturk, 2019-12-09:
This is (from what I'm told) THE library to use for ILI9486, like the 3.5" TFT touchscreen I got. However, the touchscreen examples included all #include FS.h, a header I don't have which wasn't included in the .zip. Where do I get it?

Bodmer, 2019-12-09:
FS.h is part of the Arduino IDE ESP32 and ESP8266 board packages, currently those are the only processors supported by the library. Check you have the appropriate board package loaded.

I am currently updating the library to support other processors, in particular the STM32, but it will be a few weeks before that is ready for release.

closed 2019-12-15 by Bodmer

Akrai, 2021-04-23:
Hi! So if i'm currently using an STM32, can I use the touch functionality of the TFT?

Bodmer, 2021-04-23:
Yes with some caveats. Yes, support is built in the library if the display is SPI and the touch controller is XPT2046.

Akrai, 2021-04-24:
Okay but how do I get the FS.h library? Or is it not needed?

Bodmer, 2021-04-24:
The FS.h library is for ESP32 and ESP8266 processors so that Filing Systems such as SPIFFS or LittleFS can be used. It is not needed for many examples and it is not needed for STM32 processors (which do not have SPIFFS or LittleFS support).

Tiramisumisu, 2022-07-03:
I wanted to use my font ,so I #define SMOOTH_FONT, but I missing FS.h. My board is Repberry RP2040. If I don't #define SMOOTH_FONT I can't use loadfont function.

Bodmer, 2022-07-03:
You need to load Earle Philhower's board package to use LittleFS on the RP2040.

Tiramisumisu, 2022-07-03:
> 

I have load Earle Philhower's board,but how to install LittleFS? I download https://github.com/littlefs-project/littlefs/releases and add to my project, it also have error with #include <LittleFS.h>

Bodmer, 2022-07-03:
LittleFS is built into Earle's board package so an extra library is not needed and may cause conflicts.

ariefazzmi, 2024-09-16:
How can we calibrate touchscreen using blue-pill STM32 if we can't access the FS.h file?