Bodmer/TFT_eSPI #493: Vertical Scroll for ST7789 1.14" 135x240

Jorgen-VikingGod, 2019-12-09:
I wanted to try the [Matrix Example](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/320%20x%20240/TFT_Matrix/TFT_Matrix.ino) on the ST7789 1.14" 135x240 - but it seams not to work correctly. (see some random pixel data...

I updated this two methods:

```
void setupScrollArea(uint16_t TFA, uint16_t BFA) {
  // Vertical scroll definition
  tft.writecommand(ST7789_VSCRDEF);
  tft.writedata(TFA >> 8);
  tft.writedata(TFA);
  tft.writedata((240 - TFA - BFA) >> 8);
  tft.writedata(240 - TFA - BFA);
  tft.writedata(BFA >> 8);
  tft.writedata(BFA);
}

void scrollAddress(uint16_t VSP) {
  // Vertical scrolling start address
  tft.writecommand(0x37); // ILI9341_VSCRSADD);
  tft.writedata(VSP >> 8);
  tft.writedata(VSP);
}
```

The similar define of `ILI9341_VSCRSADD` is missing in the **ST7789** driver header;
so I used the same value (0x37) the **ILI9341** driver uses.

For information - this is my `scroll_slow` method:
```
int scroll_slow(int lines, int wait) {
  int yTemp = yStart;
  for (int i = 0; i < lines; i++) {
    yStart++;
    if (yStart == 240 - BOT_FIXED_AREA)
      yStart = TOP_FIXED_AREA;
    scrollAddress(yStart);
    delay(wait);
  }
  return yTemp;
}
```

and here the content of `loop`:

```
// First fill the screen with random streaks of characters
  for (int j = 0; j < 400; j += TEXT_HEIGHT) {
    for (int i = 0; i < 40; i++) {
      // Rapid fade initially brightness values
      if (pos[i] > 20)
        pos[i] -= 3;
      // Slow fade later
      if (pos[i] > 0)
        pos[i] -= 1;
      // ~1 in 20 probability of a new character
      if ((random(20) == 1) && (j < 200))
        pos[i] = 63;
      // Set the green character brightness
      tft.setTextColor(pos[i] << 5, TFT_BLACK);
      // Draw white character
      if (pos[i] == 63)
        tft.setTextColor(TFT_WHITE, TFT_BLACK);
      // Draw the character
      xPos += tft.drawChar(random(32, 128), xPos, yDraw, 1);
    }
    // Scroll, 14ms per pixel line
    yDraw = scroll_slow(TEXT_HEIGHT, 14);
    xPos = 0;
  }

  // Now scroll smoothly forever
  while (1) {
    yield();
    yDraw = scroll_slow(240, 5);
  } // Scroll 320 lines, 5ms per line
```

Bodmer, 2019-12-09:
I suspect the probem is that the ST7789 drivers full screen area capability is 240x320 (width x height). The 135x240 screen is a window in this total capability, the top left corner of the screen window is positioned at x,y = 52,40 so the scroll window top y coord must therefore be set to 40 and the bottom to 280.

Jorgen-VikingGod, 2019-12-13:
@Bodmer thx for the hint.
But using this offset did not work correctly - got some black lines instead.
I reset the example code again and use the 135x240 just with the settings of the 240x320 display and it worked fine out of the box.

closed 2019-12-13 by Jorgen-VikingGod

ghostoy, 2020-07-28:
I also experienced the same issue with TTGO T-Display (240x135). The problem is addressed in the restriction section of `VSCRDEF` command on the datasheet, which states
> The condition is TFA+VSA+BFA = 320, otherwise Scrolling mode is undefined. 
Therefore, we should always assume the display memory as 320 x 240, and customize TFA and BFA to constraint the scrolling area.

Also I found in the builtin driver of ST7789 that the drawing methods are shifted by [(52, 40) pixels](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ST7789_Rotation.h#L8-L12). So `yDraw` should be `yStart - 40`.

Here is what I did:
* define TFA and BFA as 40, which is not visible on the display
* subtract 40 (or TFA) from return value of `scroll_slow` for new `yDraw`
* initialize `yDraw` as `240 - TEXT_HEIGHT`