Bodmer/TFT_eSPI #496: Issue jpeg arrary does not render correctly with version >1.3.8 TFT_eSPI

jeckert12, 2019-12-18:
Symptom 
I upgraded Arduino ide to 1.8.10 and ESP8266 to 2.6.3. At that I could no longer compiled with the version of TFT_eSPI I had from 12/24/2018, So I pull the latest down from website.
new version complied but now Array no longer render correctly.

To get working again I tried downgrading ESP8266 to 2.4.2 still didn't render correctly. Then I down graded ide to 1.8.9. Still didn't work. I had saved the old version to tft_eSPI and copied that back over.
After that it worked. I upgraded IDE 1.8.10 again a still worked with older TFT_eSPI.

  So With IDE at 1.8.10 esp8266 at 2.4.2 and tft_eSPI at version from 12/24/2018( I don't see version number in it) drawArrayJpeg works fine. 
  but with 1.4.20 ti didn't. 

With further testing I found the last version that worked for me was 1.3.8.  Anything newer then that and the array didn't render correctly.
[image]
[image]

Bodmer, 2019-12-18:
Some time ago the Adafruit changed the parameters required by setAddrWindow().

TFT_eSPI was updated to be compatible with the new Adafruit-GFX parameters and thus some old sketches need to be updated. Post the sketch and I will tell you what to change.

jeckert12, 2019-12-19:
Prefect, I replaced the jpegRender() I had been using with renderJPEG() from a newer example and it worked.   Thank you, your libraries greatly simplified and sped up my early attempts at downloading and displaying jpegs from a webserver. 

I included my files anyway incase someone is interested in how to download and display jpeg from a webserver.  

[ESP-JEPG-V3_2.zip](https://github.com/Bodmer/TFT_eSPI/files/3983453/ESP-JEPG-V3_2.zip)

Bodmer, 2019-12-19:
Great.
If you have a working setup then there is no need to change it but it may be of interest that I have recently added a new jpeg decoder based on Tiny Jpeg that runs ~50% faster on 32 bit processors.There is an[ example here](https://github.com/Bodmer/TJpg_Decoder/tree/master/examples/Web_Jpg) that downloads jpegs from the web and saves them in SPIFFS before rendering.

I see you have saved to an array first which is a good idea for images that change often.

closed 2019-12-19 by Bodmer