Bodmer/TFT_eSPI #497: Deep sleep current with ST7789V

dmenne, 2019-12-18:
Great library, everything works fine out of the box. Thanks for the huge amount of work you put into this. 
I have a minor problem in "non-working" (i.e. deep sleep) mode. TTGO ST7789V: I cannot switch back the display to full sleep (or simply did not find the right function to call).

Deep sleeps draws almost no current when the display was not initialized. Fine.
```
void setup(){
  delay(10000); // 38 mA
  esp_deep_sleep_start(); // 0.3mA, great!
}
```

But once display is initialized, I cannot get back to low power:

void setup(){
  delay(10000); //40 mA
  tft.init();
  tft.fillScreen(TFT_BLACK);
  delay(10000); // 60 mA
  digitalWrite(TFT_BL, 0); // 47 mA
  delay(10000);
  esp_deep_sleep_start(); // 7 mA
}

Any idea how to get back to 0.3 mA in deep sleep?

Bodmer, 2019-12-18:
There is a command to put the display into low power mode but there is not feature in the library to do this. You can send commands to the display from the sketch. This should work:

tft.writecommand(0x10); // Sleep
delay(5); // Delay for shutdown time before another command can be sent

tft.writecommand(0x11); // Wake display
delay(120); // Delay for pwer supplies to stabilise

dmenne, 2019-12-19:
Super! Thanks, works.

closed 2019-12-19 by dmenne

michapixel, 2020-02-24:
```
#define TFT_BL 4
digitalWrite(TFT_BL, LOW); // turns off backlight
```

maybe that helps.

michapixel, 2020-02-24:
```
void TFT_powerUp() {
  tft.init();
  tft.fillScreen(TFT_BLACK);
  tft.setRotation(0); // porttrait 0, landscape: 1
}

void TFT_sleep() {
  Serial.println("Setting TFT (again) to deep-sleep ");
  //tft.writecommand(0x10); // Sleep (backlight still on ...)
  digitalWrite(TFT_BL, LOW);
  //tft_inited = false;
  delay(5); // needed!
}

void TFT_wake() {
  tft.writecommand(0x11); // WAKEUP
  delay(120); // needed! PWR neeeds to stabilize!
  digitalWrite(TFT_BL, HIGH);
}
```

to be complete, and it did the job for me, although i didn't measure power consumption, may  it needs `tft.writecommand(0x10);` too?

dmenne, 2020-02-24:
See my first example, where I measured this using 0 instead of LOW

michapixel, 2020-02-24:
But you only set the ESP to deep-sleep, not the TFT, right? 
Might i ask if you can measure again with my code?

```
#include <TFT_eSPI.h> // Graphics and font library for ST7735 driver chip
#define TFT_BL 4 // Display backlight control pin
TFT_eSPI tft = TFT_eSPI(); // Invoke library

void TFT_powerUp() {
  Serial.println("Powering up TFT");
  tft.init();
  // clear screen:
  tft.fillScreen(TFT_BLACK);
  tft.setRotation(0); // porttrait 0, landscape: 1
}

void TFT_sleep() {
  // clear screen:
  tft.fillScreen(TFT_BLACK);
  //
  Serial.println("Setting TFT (again) to deep-sleep ");
  tft.writecommand(0x10); // SLEEP (backlight still on ...)
  digitalWrite(TFT_BL, LOW); // backlight OFF
  delay(5); // needed!
}

void TFT_wake() {
  tft.writecommand(0x11); // WAKEUP
  delay(120); // needed! PWR neeeds to stabilize!
  digitalWrite(TFT_BL, HIGH); // backlight ON
}

void wakeup_reason(){
  esp_sleep_wakeup_cause_t wakeup_reason;
  wakeup_reason = esp_sleep_get_wakeup_cause();
  switch(wakeup_reason)
  {
    //case ESP_SLEEP_WAKEUP_EXT0 : Serial.println("Wakeup caused by external signal using RTC_IO"); break;
    //case ESP_SLEEP_WAKEUP_EXT1 : Serial.println("Wakeup caused by external signal using RTC_CNTL"); break;
    case ESP_SLEEP_WAKEUP_TIMER : 
      Serial.println("> Wakeup caused by timer"); 
      TFT_wake();
    break;
    //
    //case ESP_SLEEP_WAKEUP_TOUCHPAD : Serial.println("Wakeup caused by touchpad"); break;
    //case ESP_SLEEP_WAKEUP_ULP : Serial.println("Wakeup caused by ULP program"); break;
    default : 
    Serial.printf("Wakeup was not caused by deep sleep: %d\n", wakeup_reason); 
    break;
  }
}

void setup() {
  Serial.begin(115200);
  Serial.println("ESP Booted up ... (setup)");
  wakeup_reason();
  // always power up? maybe just tft.init() if wakeup on timer ... 
  TFT_powerUp();
  //
  delay(1000);
  Serial.println("Writing to TFT ...");
  tft.setTextColor(TFT_WHITE, TFT_BLACK);
  tft.setTextFont(2);
  tft.println("TFT here ...");
  delay(10000);
  TFT_sleep();
  // enable wakeup after 2 secs:
  esp_sleep_enable_timer_wakeup(5000 * 1000); // microseconds
  // send ESP to sleep
  esp_deep_sleep_start();
  // nothings gonna run after that!
}

  
void loop() {
  // put your main code here, to run repeatedly:
}
```
BTW:
https://randomnerdtutorials.com/esp32-timer-wake-up-deep-sleep/

michapixel, 2020-02-24:
On the other hand:

My experiments show, that if i set the display to sleep and shutoff the backlight, the screen-content is still held in memory ( i guess by the TFT itself?)
That may cause the higher consumption, or could it?

dmenne, 2020-02-24:
Sorry, I cannot easily measure it now since the device is build in. I will check again when I revise it. 

I have an USB Stick to measure current, but it is not accurate enough for low currents. I measured  with a battery connected by splicing the power wire.

michapixel, 2020-02-24:
I guess these aren't good enough. 
Maybe i can give some insight, i am using a battery too, if i can find one for mine, any recommendations?
That would be inaccurate too, but we know how long battery should last, without the display, so it's rough, but good enough for me.

dmenne, 2020-02-24:
I am not totally happy because the 650mAh Battery only lasts for a week when not used. So something must be wrong - waiting for next hardware round to go into details

sej7278, 2021-04-04:
seeing similar issues with ST7735, 1000mAh lipo lasts under 2 weeks, the esp32 is only turned on for 5-15secs a day using the reset button then goes to sleep - i get the white flash mentioned in #715 too:

```
void gotoSleep()
{
    tft.fillScreen(TFT_BLACK);
    tft.writecommand(ST7735_SLPIN);
    digitalWrite(TFT_BL, LOW);
    delay(5);
    gpio_hold_en(GPIO_NUM_4);
    gpio_deep_sleep_hold_en();
    esp_deep_sleep_start();
}

void setup()
{
    // init screen
    tft.init();
    tft.fillScreen(TFT_BLACK);
    tft.setRotation(3);
    tft.writecommand(ST7735_SLPOUT);
    delay(120);
    pinMode(TFT_BL, OUTPUT);
    digitalWrite(TFT_BL, HIGH);

    // rest of code ....
}

void loop() {}
```

can anyone recommend a display that actually goes to sleep and uses uA's (could do without having to keep a pin held low to keep the backlight off too).