Bodmer/TFT_eSPI #498: Touch XPT2046 dont work. No touch

arnold-b, 2019-12-20:
Hello, 

there i don´t find the Problem. For now i changed your touch and using 
https://github.com/PaulStoffregen/XPT2046_Touchscreen

But it is destoyed on every update. May you can help to find my Problem.

Bodmer, 2019-12-21:
[Run this sketch](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Test%20and%20diagnostics/Read_User_Setup) to verify the settings are being picked up correctly. Post the serial output here.

Then [run this sketch](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Test%20and%20diagnostics/Test_Touch_Controller), touch the screen and verify the values output to the serial port change. Post some of the readings here.

Bodmer, 2020-01-11:
Closing as without feedback this will not get solved.

closed 2020-01-11 by Bodmer

hrauch, 2020-04-12:
Hello,
I do have the same problem, but I am using an ILI9341 chip.

These are my changes in User_Setup.h:

// ###### EDIT THE PIN NUMBERS IN THE LINES FOLLOWING TO SUIT YOUR ESP32 SETUP   ######

// For ESP32 Dev board (only tested with ILI9341 display)
// The hardware SPI can be mapped to any pins

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   15  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   4  // Reset pin (could connect to RST pin)
//#define TFT_RST  -1  // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST

#define TOUCH_CS 21     // Chip select pin (T_CS) of touch screen

Here you can see the output of your read_info-sketch:

[code]
TFT_eSPI ver = 2.2.0
Processor    = ESP50
Frequency    = 240 MHz
Transactions = Yes 
Interface    = SPI 
Display driver = 9341
Display width  = 240 
Display height = 320 

MOSI    = D-1 (GPIO 23)
MISO    = D-1 (GPIO 19)
SCK     = D-1 (GPIO 18)
TFT_CS   = D8 (GPIO 15)
TFT_DC   = D4 (GPIO 2)
TFT_RST  = D2 (GPIO 4)

TOUCH_CS = D-1 (GPIO 21)

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 27.0 MHz 
Touch SPI frequency   = 2.5 MHz 
[/code]

I hope you can give me a hint for solving my touch-problem.
Hans

arnold-b, 2020-04-12:
Hello, 

i change the code in eSpi to use XPT2046_Touchscreen

Since this i dont have problem

 

#ifdef TOUCH_CS

  #include <XPT2046_Touchscreen.h>

  #ifdef TIRQ_PIN

  XPT2046_Touchscreen ts(TOUCH_CS, TIRQ_PIN);  // Param 2 - Touch IRQ Pin - interrupt enabled polling

#else

XPT2046_Touchscreen ts(TOUCH_CS, 255);  // Param 2 - Touch IRQ Pin - interrupt enabled polling

 

 

 

uint8_t TFT_eSPI::getTouch(uint16_t *x, uint16_t *y, uint16_t threshold){

  uint16_t x_tmp, y_tmp;

  uint8_t valid = 0;

  

  #ifdef TIRQ_PIN 

   if (ts.tirqTouched()) { xx

  #endif

    if (ts.touched()) {

    TS_Point p = ts.getPoint();

  

  if (threshold<20) threshold = 20;

  if (_pressTime > millis()) threshold=20;

uint8_t n = 5;

  x_tmp = p.x;

  y_tmp = p.y;

/* Serial.print(", x = ");

      Serial.print(p.x);

      Serial.print(", y = ");

      Serial.print(p.y);

      

      Serial.println();

                */  

  /*

  while (n--)

  {

    if (validTouch(&x_tmp, &y_tmp, threshold)) valid++;;

  }

*/

/* if (valid<1) { _pressTime = 0; return false; }

  

  _pressTime = millis() + 50;

*/

  convertRawXY(&x_tmp, &y_tmp);

 

  if (x_tmp >= _width || y_tmp >= _height) return valid;

 

  _pressX = x_tmp;

  _pressY = y_tmp;

/*  Serial.print(", tmpx = ");

      Serial.print(x_tmp);

      Serial.print(", tmpy = ");

      Serial.print(y_tmp);

      

      Serial.println();

  */

  valid = 1;

  *x = _pressX;

  *y = _pressY;

  //Serial.printf("Get Touch %d %d  %d\n",_pressX,_pressY,valid);

  

                }

                #ifdef TIRQ_PIN 

   }

   #endif

  return valid; 

}

hrauch, 2020-04-13:
Thank you Arnold for your hint.
In the end this info
https://github.com/Bodmer/TFT_eSPI/issues/498#issuecomment-612622537
solved my problem.
May be it should be in the documentation ...
Hans

arnold-b, 2020-04-13:
I´m happy that i could help you

Patman2862, 2023-12-27:
Hi Guys, would you please be able to let me know where I need to add Arnolds code?

I am having the same problem as hrauch and Arnolds code seemed to work, but I cant figure out where I need to add it.
Is it in the main code or the TFT_eSPI Cpp file?

Any help would be great, thank you.

arnold-b, 2023-12-28:
Sorry that i can not help just now. I m far away from my developmentsVon meinem iPhone gesendet