Bodmer/TFT_eSPI #499: ST7796S  compatibility.

peppeve, 2019-12-20:
Is possible to have compatibility on esp32 with [this ](http://www.lcdwiki.com/MHS-4.0inch_Display-B)it use the ST7796S i think very similar to ili9488. I have ordered 2 pieces and i think i get next week.
Thank's.

Bodmer, 2019-12-21:
The screen appears to have a special serial to parallel IC fitted to the PCB, thus is likely to have a bespoke interface protocol.

Do you have information on the protocol used to communicate and setup the screen?

peppeve, 2019-12-21:
sorry but i have no idea. When i get it i try to tell you more.

Bodmer, 2019-12-21:
It looks interesting as it can be clocked fast if the spec is to be beleived, but it may be necessary to reverse engineer the protocol...

peppeve, 2019-12-21:
btw st7796s support this Interface:
- Parallel 8080-series MCU Interface (8-bit, 9-bit, 16-bit, and 18-bit) 
- 16/18 RGB Interface (VSYNCX, HSYNCX, DOTCLK, ENABLE, DB[17:0]) 
- Serial Peripheral Interface (SPI Interface) 
- MIPI

peppeve, 2019-12-21:
[There](https://github.com/lcdwiki/LCD-show/blob/master/MHS40-show) there is the configuration file used on the raspberry if it can help.

peppeve, 2020-01-02:
@Bodmer, some news? My dispaly is underway, i hope to receive soon.

Bodmer, 2020-01-07:
The good news is that I have one of these displays now as I think it will be impossible to get it running otherwise.

The bad news (for you) is that paid work and family are taking up my time, but I may be able to look at it again towards the end of January.

Bodmer, 2020-01-09:
I have had a go at getting this display to work, but no success.  This maybe because of an incorrect assumption about the way the interface chip on the back of the board works, I suspect it is just a serial to parallel converter. To eliminate the ST7789S driver as the cause [I have bought one of these](https://www.aliexpress.com/item/4000224451632.html?spm=a2g0s.9042311.0.0.5a3f4c4dlOshqp), but it will take a few weeks to arrive.

peppeve, 2020-01-10:
My displays are under delivery today, tell me how i can help you.

peppeve, 2020-01-10:
I get the displays but the big square chip is without any sign. :(

Bodmer, 2020-01-10:
Good news. I have a working version on the ESP32. Performance is very good for a SPI display as it is working fine at 80MHz SPI clock rate (which is the maximum SPI clock rate for an ESP32).

```
Benchmark,                Time (microseconds)
Screen fill,              180522
Text,                     13573
Lines,                    165921
Horiz/Vert Lines,         16287
Rectangles (outline),     9354
Rectangles (filled),      440321
Circles (filled),         102082
Circles (outline),        101753
Triangles (outline),      34522
Triangles (filled),       164049
Rounded rects (outline),  43086
Rounded rects (filled),   490614
Total = 1762084us
Total = 1.7621s
```
I anticipate I will be able to push the changes to Github hopefully within 24hrs after testing with the ESP8266.

peppeve, 2020-01-10:
:) very happy.

Hans007a, 2020-01-10:
I'm very happy too! :)

Bodmer, 2020-01-11:
Display support now added to the Master branch.

ESP32 SPI can clock the display at 80MHz.

ESP8266 can clock the display at 40MHz maximum.

Please raise any problems as a NEW issue.

simonchen007, 2020-04-16:
it seems not works in st7796s currently.
just a white screen.

Bodmer, 2020-04-16:
The setups and testing has been performed on a RPi MHS 4.0 display.

If I guess that you are using an ESP32 and an SPI base display that is not for a RPi then try this setup and wiring:

```
#define ST7796_DRIVER

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   15
#define TFT_DC    2
#define TFT_RST   4

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

#define SPI_FREQUENCY  20000000

``` 

I am not able to advise further without knowing more about your hardware and software setup.

simonchen007, 2020-04-17:
I tried your configuration,still a white blank screen.
here is the main code I using:
```c
/*
 * Blink
 * Turns on an LED on for one second,
 * then off for one second, repeatedly.
 */

#include <Arduino.h>
/*

  This program provides cartesian type graph function

  Revisions
  rev     date        author      description
  1       12-24-2015  kasprzak    initial creation

  Updated by Bodmer to be an example for the library here:
  https://github.com/Bodmer/TFT_eSPI
  
*/

#include <TFT_eSPI.h> // Hardware-specific library
#include <SPI.h>

TFT_eSPI tft = TFT_eSPI();                   // Invoke custom library with default width and height

#define LTBLUE    0xB6DF
#define LTTEAL    0xBF5F
#define LTGREEN   0xBFF7
#define LTCYAN    0xC7FF
#define LTRED     0xFD34
#define LTMAGENTA 0xFD5F
#define LTYELLOW  0xFFF8
#define LTORANGE  0xFE73
#define LTPINK    0xFDDF
#define LTPURPLE  0xCCFF
#define LTGREY    0xE71C

#define BLUE      0x001F
#define TEAL      0x0438
#define GREEN     0x07E0
#define CYAN      0x07FF
#define RED       0xF800
#define MAGENTA   0xF81F
#define YELLOW    0xFFE0
#define ORANGE    0xFC00
#define PINK      0xF81F
#define PURPLE    0x8010
#define GREY      0xC618
#define WHITE     0xFFFF
#define BLACK     0x0000

#define DKBLUE    0x000D
#define DKTEAL    0x020C
#define DKGREEN   0x03E0
#define DKCYAN    0x03EF
#define DKRED     0x6000
#define DKMAGENTA 0x8008
#define DKYELLOW  0x8400
#define DKORANGE  0x8200
#define DKPINK    0x9009
#define DKPURPLE  0x4010
#define DKGREY    0x4A49

// these are the only external variables used by the graph function
// it's a flag to draw the coordinate system only on the first call to the Graph() function
// and will mimize flicker
// also create some variables to store the old x and y, if you draw 2 graphs on the same display
[... 536 lines cut ...]
    pinMode(LED,OUTPUT);
    LED_SET;
    LCD_CS_SET;
    LCD_RST_SET;
    LCD_RS_SET;
}

void setup() 
{
    spi_init();
    pin_init();
    Lcd_Init();
}

void loop() 
{   
   Lcd_Clear(RED); 
 // delay(0); 
  Lcd_Clear(GREEN);
 // delay(0);  
   Lcd_Clear(BLUE); 
 // delay(100); 
  // }
}
```

Bodmer, 2020-04-17:
Hi, I do not have an SPI ST7796S display so I cannot test. My ST7796 board is 8 bit parallel.

I suspect that in SPI mode the ST7796 may be like the ILI9488 and only accept 18 bit color configuration.  You can test this for me by [finding this line](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ST7796_Init.h#L28), and changing it to:

```
writedata(0x66);    //Control interface color format set to 18
```
Recompile and upload your sketch. The display should look different, it will have missing pixels but at the moment this is a test and we can correct the missing pixels later when we know this color mode is the problem.

simonchen007, 2020-04-17:
no, it still not works with your way.
but however, I make it works just now.
I change with that way,
change:
```c
	writecommand(0xB6); //Display Function Control
	writedata(0x20);    //RGB interface for RAM access, RGB transfer in DE mode  
						          //display data path in memory
						          //Gate outputs in non-display area in Normal scan 
						          //Source output in non-display area =V63

	writedata(0x02);    //Source Output Scan from S1 to S960, Gate Output scan from G1 to G480, scan cycle=2
        writedata(0x3B);    //LCD Drive Line=8*(59+1)
```
with:
```c
	writecommand(0xB6); //Display Function Control
	// writedata(0x20);    //RGB interface for RAM access, RGB transfer in DE mode  
						          //display data path in memory
						          //Gate outputs in non-display area in Normal scan 
						          //Source output in non-display area =V63

	writedata(0x02);    //Source Output Scan from S1 to S960, Gate Output scan from G1 to G480, scan cycle=2
	// writedata(0x3B);    //LCD Drive Line=8*(59+1)
```
it works now.
but another problem here, 
I can not change SPI_FREQUENCY as 80000000, once it changed as 80000000, it will also white display.

And I also have the parallel version of ST7796s, can you please show me the way about connection between esp32 to display ?
I will test it tomorrow.

prenticedavid, 2020-04-17:
I have the Red SPI ST7796S display.   It "works" with the TFT_eSPI init code if you change the first parameter for the 0xB6 register e.g. from
    0xB6, 3, 0x20, 0x02, 0x3B,  // Display Function Control [80 02 3B]
to
    0xB6, 3, 0x80, 0x02, 0x3B,  // Display Function Control [80 02 3B]
i.e. the power up default.

The ST7796S can read GRAM in 565 mode (16-bit) as well as write GRAM in 565 mode.
Obviously read and write use 24-bit in 666 mode.
So this is a great relief compared to  the ILI9488 which can only use 666 mode with the SPI interface.

Most commands work fine.

However,  there are some very weird behaviours e.g. when you write to MADCTRL (0x36)
readPixel() seems to require monster delays.
I get intermitient weird behaviour with an SD card.

I was hoping that you (Bodmer) possessed a native SPI display.    And we could resolve these weirdos.

David.

Bodmer, 2020-04-17:
Thanks.

 I have updated the init code for the ST7796 to that specified by David.

I do not have an SPI ST7796 to test. I will have a look to see if I can find one for sale. I see no obvious reason for the slow reads.

A SPI clock rate of 27MHz or 40MHz maximum is suggested. The 80MHz rate is only for the [MHS 4.0 RPi display](http://www.lcdwiki.com/File:MHS4001-02.jpg) which has a dedicated high speed serial to parallel chip.

simonchen007, 2020-04-17:
I got the data sheet just now.
B6 = BYPASS | RCM | RM | 0 | PTG[1:0] | PT[1:0]
RM Interface for RAM access
0 System Interface
1 RGB interface

0x20 = 0010 0000
so, I do not works in the SPI mode.
In spi version, we need ??0? ????
so, 0x80 or 0x00 also okay.

@Bodmer ,
Could  you please show me the way about connection between esp32 to display of the ST7796s parallel version ?

simonchen007, 2020-04-17:
> Thanks.
>
> I have updated the init code for the ST7796 to that specified by David.
>
> I do not have an SPI ST7796 to test. I will have a look to see if I can find one for sale. I see no obvious reason [...]

can you please drop me a email ?
simon@yeacreate.com
I can support you a SPI version of 4.0 inch st7796s display module.

prenticedavid, 2020-04-17:
@simonchen007,
I have never seen a ST7796S with 8080-8 interface.
Bodmer supports the "8080-16 driven by shift register" e.g. RaspberryPi Hat

I would expect the ST7796S to work exactly the same with 8080-8, 8080-16, SPI, ...
Obviously some Manufacturer Registers only apply to the RGB interface.

If you have some connection with LCDWIKI,  it would be useful to have a GitHub link.
The current lcdwiki GitHub does not respond to any Pull Requests or contain the ST7796S support.

@Bodmer ,
40MHz is the practical limit for native SPI.

David.

simonchen007, 2020-04-17:
> @simonchen007,
> I have never seen a ST7796S with 8080-8 interface.
> Bodmer supports the "8080-16 driven by shift register" e.g. RaspberryPi Hat
>
> I would expect the ST7796S to work exactly the same [...]

They just use github as a net disk. so, few respond.
if you need some documents, please let me know.

Bodmer, 2020-04-17:
This is the [8 bit parallel ST7796S board](https://www.aliexpress.com/item/4000127049804.html?spm=a2g0o.productlist.0.0.18f63a977XpY7p&algo_pvid=cb5ebb02-1be5-4cdc-b1c1-f8b2fbd9ce6f&algo_expid=cb5ebb02-1be5-4cdc-b1c1-f8b2fbd9ce6f-36&btsid=0ab6f8ad15871341459362977e2252&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_) that I have.

It fits on the ESP32 UNO format board as shown in the [TFT_eSPI Github ReadMe](https://github.com/Bodmer/TFT_eSPI#8-bit-parallel-support). Extra links have to be added to the board as shown. The pin connections are then in the [setup file here](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup14_ILI9341_Parallel.h#L6-L24), to use that setup change the driver from ILI9431 to ST7796.

@ simonchen007 : If you wish you can [contact me](https://github.com/Bodmer/TFT_eSPI/blob/master/library.json#L15).

prenticedavid, 2020-04-17:
Having any central Repository is wise.   LCDWIKI don't have to use GitHub (but GitHub would be fine)

Correction.   I see that LCDWIKI have a 8/16 bit parallel board with 34-pin header.
Arduino 16-bit shields are for Mega2560 / Due.
Or there are 40-pin boards that use 40-pin Arduino Adapter Shields.
The ESP32 has not got enough GPIO for 8080-16 but it can drive 8080-8.

Documents.  Some Mcufriend Uno Shields have appeared with ID = 0x3229.
I would dearly like to find a datasheet.  The Manufacturer or part number might help me to find a datasheet.   

David.

Bodmer, 2020-04-17:
I think this is solved now so closing, but feel free to carry on conversations.

closed 2020-04-17 by Bodmer

Triangulix, 2022-12-10:
I can confirm that my recently bought red no-name display board with ST7796 only works if the ST7796_Init.h is changed as described by [prenticedavid](https://github.com/prenticedavid).

Triangulix, 2022-12-10:
@Bodmer Do you still have no display with ST7796? I would donate one if you like.