Bodmer/TFT_eSPI #500: Font sizes in pixels?

md88x, 2019-12-24:
Hello everyone,

I am displaying a clock and different data from a weather station on a TFT Lcd. 
To organize my screen, which is the font size in pixels according also the text size?

For instance if I want to print a "0" if the number is <10 which sapcing I need to respect?

 >if (second()<10){
  >  tft.drawChar('0', **x coordinate**, 100, 6);
   > tft.drawNumber(second(), **x coordinate+?**, 100, 6);
  >  }
  >  else
   > tft.drawNumber(second(), **x coordinate**, 100, 6); 

Thanks in advance for the help and the beatiful library.

Bodmer, 2019-12-24:
[Look at this code](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/320%20x%20240/TFT_Clock_Digital/TFT_Clock_Digital.ino#L94-L100), th functions used for printing the characters return the width of the printed string.

md88x, 2019-12-26:
Thank you,

your suggestion is very helpful. Is something similar available also for the y coordinate?

Another question: how do I print the "°" symbol?

Bodmer, 2019-12-27:
fontHeight(f); will return the height of the currently selected free font, where f is 1 for GLCD or free font and 2,3,6,7,8 for the build in fonts.

The grave ( ` ) diacritical mark will show as a degree ( ° ) symbol in fonts 2 and 4, e.g.
```
tft.drawString("`", 0, 0, 2);
tft.drawString("`", 0, 50, 4);
```

closed 2019-12-27 by Bodmer