Bodmer/TFT_eSPI #502: How to config  for 3.5" ILI9486  and TTGO-T Cell (T18)  ?

TridentTD, 2019-12-27:
[image]

I have tried to config **TFT 3.5" 480x320  ILI9486** from
https://www.aliexpress.com/item/33005785723.html?spm=a2g0s.9042311.0.0.158a4c4dors4ko
to **ESP32 TTGO-T Cell.**

The connections as the upper picture.
And config at `User_Setup.h` like the following code

```c
#define ILI9486_DRIVER

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS    5  // Chip select control pin
#define TFT_DC   22  // Data Command control pin
#define TFT_RST   4  // Reset pin (could connect to RST pin)

#define SPI_FREQUENCY  27000000 // Actually sets it to 26.67MHz = 80/3
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000

```

When I test by `example > 480x320 > TFT_Rainbow480`,
the result can't show anything.

How to config for **TFT 3.5" 480x320  ILI9486**?
Thank you very much.

Bodmer, 2019-12-27:
The heading for the advert says it is an ILI9486 but the electrical specs say ILI9488 which is very different.

Try ILI9488_DRIVER

Unfortunately there are many adverts from low cost vendors that quote the wrong driver.

Note the comment associated with this driver in the original setup file. It could be the case that the SD card slot uses the MISO signal:
**WARNING: Do not connect ILI9488 display SDO to MISO if other devices share the SPI bus (TFT SDO does NOT tristate when CS is high)**

TridentTD, 2019-12-28:
It's worked!  
Driver is LI9488 

Thank you very much for advise and warning

Bodmer, 2020-01-11:
Thanks for letting me know.

closed 2020-01-11 by Bodmer