Bodmer/TFT_eSPI #504: Hot to print empty space

md88x, 2019-12-29:
Hello again,

how do I print an empty space?

tft.drawChar(' ', xpos, 100, 4);      
doesn't work for me.

Many thanks in advance

Bodmer, 2019-12-29:
You will not see a space unless you have set the background colour:
tft.setTextColor(TFT_GREEN, TFT_BLUE);

This will make a space appear as a blue rectangle. Characters will show as green over a blue background.

md88x, 2019-12-30:
Thank you for your reply.
I cheched my sketch but does not work for me.

I report the code lines regarding the issue:

tft.setTextColor(TFT_WHITE, TFT_BLACK);

  xpos = 20;
  tft.drawString ("IN", xpos, 80, 2);
  if (temperature >= 0)
      xpos += tft.drawChar('+', xpos, 100, 4);
  if ((temperature <10) && (temperature >-10))
      xpos += tft.drawChar(' ', xpos, 100, 4);
  xpos += tft.drawFloat(temperature, 1, xpos, 100, 4);

  tft.drawString ("OUT", 196, 80, 2);
  xpos = 240-xpos;
  if (temperature >= 0)
    xpos += tft.drawChar('+', xpos, 100, 4);
  if ((data.t <10) && (data.t >-10))
    xpos += tft.drawChar(' ', xpos, 100, 4);
  xpos += tft.drawFloat(data.t, 1, xpos, 100, 4);

If I force data.t or temperature to be 0, on my screen I see +0.0 instead of the desired + 0.0. If i replace the empty space with any other letter (e.g. character 'a'), I correctly visualize +a0.0.

Bodmer, 2020-01-02:
I tried your test code and it worked fine. Bear in mind that Font 4 is a proportional font and that different characters have different width. A space character is quite narrow and is smaller that the width of a digit.

The functions tft.setTextPadding(...), tft.drawFloat(..) and setTextDatum(..) are intended to be used to position text and over-write old values. Example:

```
  tft.fillScreen(TFT_NAVY);

  tft.setTextColor(TFT_WHITE, TFT_BLACK);

  while (1) {

    float temperature = random(400) / 10.0 - 20.0;

    int16_t xpos = 20; // Draw all relative to this datum

    tft.setTextDatum(TL_DATUM); // Top left datum

    tft.drawString ("IN", xpos, 80, 2);

    int16_t w = tft.textWidth("88.8", 4); // Width of maximum length string)

    tft.setTextPadding( w ); // Set blanking width to string length

    tft.setTextDatum(TR_DATUM); // Set datum to top right

    tft.drawFloat(abs(temperature), 1, xpos + w, 100, 4); // Draw value without sign

    tft.setTextPadding(tft.textWidth("+", 4)); // Set padding width to cover +

    // Draw sign
    if (temperature >= 0) tft.drawString("+", xpos, 100, 4);
    else  tft.drawString("-", xpos, 100, 4);

    tft.setTextPadding(0); // Reset padding to 0 for future drawing

    delay(1000);
  }

```

md88x, 2020-01-03:
Thankyou Bodmer. 
Could you please explain to better understand and to improve myself, setTextPadding and setTextDatum?

From what I am understanding also with your examples, setting the datum I can print relative to the point of the string/numbers from left to right, centered or left to right (top left, middle left, etc.). If I set  for instance top right (TR_DATUM) my data is printed so that the last character finishes in my xpos? 
With tft.setTextPadding, I set the space to be left empty to align the elements?

I hope I haven't messed up.

Thanks in advance

md88x, 2020-01-04:
With your sugestion and a little bit of trial and error I was able to resolve  and correctly layout my data.

Thanks!

closed 2020-01-04 by md88x