Bodmer/TFT_eSPI #505: ReadRect() reads wrong pixel value?

TheHexaCube, 2019-12-29:
Hello, it seems that there is some difference between using readPixel() and readRect(). 

Using readPixel to return a pixel value at a coordinate determined by touch, I get the correct color value, ie. 0x2148 actually is 0x2148. 
When using readRect() however, I got 0x9042 instead, for instance.
It seems to be grabbing the correct pixel positions, but storing them with wrong pixel values. 

After some additional testing it seems to read some pixel values just fine, like pure red, green blue. 

I'm sure I'm doing something wrong, but I can't quite figure out what exactly so far.

Bodmer, 2019-12-29:
It should work as expected but there are a number of reasons why the rectangle read may be failing.

Can you tell me more about your hardware setup, processor ES{32 or ESP8266), display driver, inteface (SPI or parallel). If SPI what clock rates are set for transmit and receive?

If you can also provie a simple sketch that illustrates the problem then I will use it for tests.

TheHexaCube, 2019-12-29:
I'm using an ESP32 using a 480x320 TFT display with an ILI9488 driver over 4-wire SPI.
Writes at 27MHz, Reads I tested at different speeds, but even at 500kHz it gives this issue, so I doubt it's a problem. 

I'll write a little sketch that illustrates the problem, but it might take me a bit.

Bodmer, 2019-12-29:
I just read your original post again. You use the word "touch" which implies you may have a touch controller connected. Is the touch controller wired to the same SPI interface as the TFT?

If so there is another issue as the ILI9488 does not tristate the SDO output which means there will be a bus conflict when you read the touch controller. This is not the cause of the read pixel problem but will need to be circumvented.

TheHexaCube, 2019-12-29:
There is touch involved, yes, using a capacitive FT6236 controller, however it uses I2C and not SPI.

I've written some test code now that writes a bunch of differently colored squares on the screen, reads out the pixels using both readPixel() and readRect() and compares them to the intended color.

Afterwards it also writes the 'wrong' readRect() colors on the screen to show the difference.

If it turns out to be simply me using bad code, please excuse me wasting your time, as my coding skills are still pretty 'amateurish'. 

I've also attached two images of the colors I get on my particular screen for anyone not wanting to run the code.

Using readPixel() and working as intended;
[image]

Using readRect() resulting in false colors, almost seems like color channels are mixed up?
[image]

```
#include <Arduino.h>

#include <SPI.h>
#include <TFT_eSPI.h> 

TFT_eSPI tft = TFT_eSPI();       

uint16_t buffer[4096];
// change size here
int8_t square_size = 64;

// Define colors to be tested here, currently using all predefined colors
uint16_t test_palette[] = {TFT_BLACK, TFT_NAVY, TFT_DARKGREEN, TFT_DARKCYAN, TFT_MAROON, TFT_PURPLE, TFT_OLIVE, TFT_LIGHTGREY, TFT_DARKGREY, TFT_BLUE, TFT_GREEN, TFT_CYAN, TFT_RED, TFT_MAGENTA, TFT_YELLOW, TFT_WHITE, TFT_ORANGE, TFT_GREENYELLOW, TFT_PINK};
uint16_t result_palette[sizeof(test_palette)/2];
int16_t h;
int16_t w;

int x = 0;
int y = 0;

void setup() {

  
  Serial.begin(9600);
  Serial.println("TFT Test");
  
  delay(100);
  tft.init();
  

  h = tft.height();
  w = tft.width();

  tft.setRotation(1);

  tft.fillScreen(TFT_BLACK);

  //Iterates through all squares, prints out actual displayed value and value reported by 
  //both readPixel() and readRect() aswell as checking wether or not they're the same

  Serial.println("\n\nGenerating squares...");
  for (int i=0; i<sizeof(test_palette)/2; i++)
  {
    
    int rect_pos_x = (i%(h/square_size))*square_size;
    int rect_pos_y = square_size*(i / (h/square_size));

    tft.fillRect(rect_pos_x, rect_pos_y, square_size, square_size, test_palette[i]);

    Serial.print("Color at square ");
    Serial.print(i);
    Serial.print(": \t0x");
    Serial.println(test_palette[i], HEX);

    
    uint16_t readback = tft.readPixel(rect_pos_x,rect_pos_y);
    Serial.print("readPixel(): \t\t0x");
    Serial.print(readback, HEX);    
    if(readback == test_palette[i])
    {
      Serial.println(" --- PASS");
    }
    else
    {
      Serial.println(" --- FAIL");
    }

    
    tft.readRect(rect_pos_x, rect_pos_y, 1, 1, buffer);
    readback = buffer[0];

    Serial.print("readRect(): \t\t0x");
    Serial.print(readback, HEX);  
    result_palette[i] = readback;
    if(readback == test_palette[i])
    {
      Serial.println(" --- PASS");
    }
    else
    {
      Serial.println(" --- FAIL");
    }
    Serial.println("--------------------------------------------");
    Serial.println("Done.");
  }

  delay(5000);
  Serial.println("Displaying colors from readRect()");

  for (int i=0; i<sizeof(result_palette)/2; i++)
  {
    
    int rect_pos_x = (i%(h/square_size))*square_size;
    int rect_pos_y = square_size*(i / (h/square_size));

    tft.fillRect(rect_pos_x, rect_pos_y, square_size, square_size, result_palette[i]);

  }
}

void loop() { 
}

```

TheHexaCube, 2020-01-01:
I'm sorry and I don't mean to be impolite nor pushy @Bodmer - 
I'm still having the issue, does your tag 'Not a TFT_eSPI issue' mean it's a my code being wrong? Could you perhaps explain what I'm doing wrong?

Bodmer, 2020-01-01:
Ah, I read it too quickly and missed the "If" in your line _"If it turns out to be simply me using bad code please excuse me wasting your time...

I will re-open and consider further.

Bodmer, 2020-01-01:
OK, I understand the problem.

readRect() was originally intended to be used with pushRect() to move pixel blocks on the screen. This means the [endianess](https://en.wikipedia.org/wiki/Endianness) of the values does not matter. So read Rect() 16 bit values have swapped bytes.

So for example to correct the byte swap you would use:

```
    tft.readRect(rect_pos_x, rect_pos_y, 1, 1, buffer);
    readback = buffer[0] <<8 | buffer[0] >> 8;
```

TheHexaCube, 2020-01-01:
That seems to partially work, however it looks like the 'shading' information is lost somehow? I'm not quite sure how to call it, but the images should show what I mean. 
[image]

Bodmer, 2020-01-01:
Can you [find this line in the TFT_eSPI.h file](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.cpp#L909) then change to:

```
#if !defined (ILI9488_DRIVER_XXX)
```
and save the altered file. Then try your test again.

TheHexaCube, 2020-01-01:
The line you mentioned was in TFT_eSPI.cpp but I found it, it works now. 
Thank you very much!

Bodmer, 2020-01-02:
Well spotted! You seem to have a peculiar ILI9488 or perhaps it is a different controller. Can you provide a link to the sellers web page. Some sellers quote the wrong controller and actually the display has an ILI9481 or ILI9486.  I do not have an SPI ILI9488 so am unable to test.

TheHexaCube, 2020-01-02:
https://www.buydisplay.com/lcd-3-5-inch-320x480-tft-display-module-optl-touch-screen-w-breakout-board

That's the display/board I bought.

Bodmer, 2020-01-02:
BuyDisplay are a reliable supplier so I trust their description. It looks like I misunderstood the datasheet in SPI mode, thanks.

closed 2020-01-02 by Bodmer