Bodmer/TFT_eSPI #506: Sprite rotation too slow

2dom, 2019-12-29:
First of all - thanks for the great library!

I am working on a project that requires lots of sprite rotations and I found the current code a bit slow. I gave it a shot and managed to improve the speed a quite bit. Main difference - fixed point arithmetic and optimized SPI payload. The implementation does, however, not support transparency. Let me know what you think!

Best Dom

```
/***************************************************************************************
** Function name:           pushFastRotated
** Description:             Push a rotated copy of the Sprite to TFT screen 
*************************************************************************************x*/

bool TFT_eSprite::pushFastRotated(int16_t angle)
{
  if ( !_created ) return false;

  // Trig values for the rotation
  float radAngle = -angle * 0.0174532925; // Convert degrees to radians
  float sinraf = sin(radAngle);
  float cosraf = cos(radAngle);

  int32_t sinra = sinraf*1024;
  int32_t cosra = cosraf*1024;

  // Bounding box parameters
  int16_t min_x;
  int16_t min_y;
  int16_t max_x;
  int16_t max_y;

  // Get the bounding box of this rotated source Sprite relative to Sprite pivot
  getRotatedBounds(sinraf, cosraf, width(), height(), _xpivot, _ypivot, &min_x, &min_y, &max_x, &max_y);

  // Move bounding box so source Sprite pivot coincides with TFT pivot
  min_x += _tft->_xpivot;
  max_x += _tft->_xpivot;
  min_y += _tft->_ypivot;
  max_y += _tft->_ypivot;

  // Test only to show bounding box on TFT
  // _tft->drawRect(min_x, min_y, max_x - min_x + 1, max_y - min_y + 1, TFT_GREEN);

  // Return if bounding box is outside of TFT area
  if (min_x > _tft->width()) return true;
  if (min_y > _tft->height()) return true;
  if (max_x < 0) return true;
  if (max_y < 0) return true;

  // Clip bounding box to be within TFT area
  if (min_x < 0) min_x = 0;
  if (min_y < 0) min_y = 0;
  if (max_x > _tft->width()) max_x = _tft->width();
  if (max_y > _tft->height()) max_y = _tft->height();

  uint32_t iheight = height()<<10;

  uint32_t iwidth = width()<<10;

  uint16_t sline_buffer[width()];
  // Scan destination bounding box and fetch transformed pixels from source Sprite
  for (int32_t y = min_y; y <= max_y; y++) {
    int32_t yt = y - _tft->_ypivot;
    uint16_t byte_count=0;
    uint16_t x_start=0;
    uint16_t x_end=0;
    int32_t cyt =  _ypivot*1024 + cosra * yt; 
    int32_t cxt =  _xpivot*1024- sinra * yt; 
    
    bool row_drawn = false;
    for (int32_t x = min_x; x <= max_x; x++) {
      int32_t xt = x - _tft->_xpivot;
      int32_t ys = (sinra * xt + cyt);
      if (ys < 0 || ys >= iheight)
        continue;
      int32_t xs = (cosra * xt + cxt);
      if (xs < 0 || xs >= iwidth)
      {
        if (row_drawn)
          break;
        else
          continue;
      }

      xs=xs>>10; // This is allowed as we know that xs is >0
      ys=ys>>10; // This is allowed as we know that ys is >0
      row_drawn=true;
      if (x_start==0)
        x_start=x;
      x_end=x;
      uint16_t rp = readPixel(xs, ys);
      sline_buffer[byte_count]=rp; 
      //_tft->drawPixel(xt, yt, rp);
      byte_count++;
    }
    _tft->pushImage(x_start, y,x_end-x_start,1,sline_buffer);
  }

  return true;
}

```

Bodmer, 2019-12-30:
I have tried fixed point in the past but the problem is that very small numbers from the trig values are multiplied by relatively large ones (coordinates) so the rotated image lacks fidelity due to duplicated pixels. As an example try [Rotated_Sprite_2 example with this line changed to](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Sprite/Rotated_Sprite_2/Rotated_Sprite_2.ino#L90):

```
    needle.pushFastRotated(angle);
```
The white pivot point pixel gets duplicated at certain angles and the fast function does not render correctly over the full 360 angular range. I suspect this is a sign change issue.

I think the lower fideleity might be OK in some applications but the function does need to work at all angles.

2dom, 2019-12-30:
I see...thanks for the explanation. 

Might be helpful to offer it as an alternative where fidelity does not matter that much whilst pointing out the limitations.

2dom, 2019-12-30:
Another potential improvement that I haven't explored is to determine the edges of the rotated sprite from the the edges of the bounding box to restrict the X/Y iterations to only the rotated slide not the entire bounding box. This should in theory improve performance for angles (45deg) where bounding box area is largest.

Bodmer, 2019-12-30:
This is a screen shot for a "needle" sprite from the Rotated_Sprite_2 example, the needle is plotted at 5 deg. increments from 0-30 degrees. On the left after the existing slow function, on the right after the fast function which is not working correctly.
[image]

So I think step 1 is for you to get the new fast function to work at all angles.

Bodmer, 2019-12-30:
I have had a quick look and the bug is in the sline_buffer handling.

2dom, 2019-12-31:
I'll have a look ... Did not have an issue so far

2dom, 2019-12-31:
Yeah ... sline_buffer needs to be bigger to cover the diagonal. Also swapped the bytes around. So it seems to work fine now with

```
  // The needle graphic has a black border so if the angle increment is small
  // (6 degrees or less in this case) it wipes the last needle position when
  // it is rotated and hence it clears the swept area to black
  for (int16_t angle = 0; angle <= 360; angle += 3)
  {
    needle.pushFastRotated(angle); // Plot direct to TFT at specifed angle
    delay(50);
    yield();                   // Avoid a watchdog time-out
  }
```

in the example. 

New code:

```

/***************************************************************************************
** Function name:           pushFastRotated
** Description:             Push a rotated copy of the Sprite to TFT screen 
*************************************************************************************x*/

bool TFT_eSprite::pushFastRotated(int16_t angle)
{
  if ( !_created ) return false;

  // Trig values for the rotation
  float radAngle = -angle * 0.0174532925; // Convert degrees to radians
  float sinraf = sin(radAngle);
  float cosraf = cos(radAngle);

  int32_t sinra = sinraf*1024;
  int32_t cosra = cosraf*1024;

  // Bounding box parameters
  int16_t min_x;
  int16_t min_y;
  int16_t max_x;
  int16_t max_y;

  // Get the bounding box of this rotated source Sprite relative to Sprite pivot
  getRotatedBounds(sinraf, cosraf, width(), height(), _xpivot, _ypivot, &min_x, &min_y, &max_x, &max_y);

  // Move bounding box so source Sprite pivot coincides with TFT pivot
  min_x += _tft->_xpivot;
  max_x += _tft->_xpivot;
  min_y += _tft->_ypivot;
  max_y += _tft->_ypivot;

  // Test only to show bounding box on TFT
  // _tft->drawRect(min_x, min_y, max_x - min_x + 1, max_y - min_y + 1, TFT_GREEN);

  // Return if bounding box is outside of TFT area
  if (min_x > _tft->width()) return true;
  if (min_y > _tft->height()) return true;
  if (max_x < 0) return true;
  if (max_y < 0) return true;

  // Clip bounding box to be within TFT area
  if (min_x < 0) min_x = 0;
  if (min_y < 0) min_y = 0;
  if (max_x > _tft->width()) max_x = _tft->width();
  if (max_y > _tft->height()) max_y = _tft->height();

  uint32_t iheight = height()<<10;

  uint32_t iwidth = width()<<10;

  uint16_t sline_buffer[width()+height()];
  // Scan destination bounding box and fetch transformed pixels from source Sprite
  for (int32_t y = min_y; y <= max_y; y++) {
    int32_t yt = y - _tft->_ypivot;
    uint16_t byte_count=0;
    uint16_t x_start=0;
    uint16_t x_end=0;
    int32_t cyt =  _ypivot*1024 + cosra * yt; 
    int32_t cxt =  _xpivot*1024- sinra * yt; 
    
    bool row_drawn = false;
    for (int32_t x = min_x; x <= max_x; x++) {
      int32_t xt = x - _tft->_xpivot;
      int32_t ys = (sinra * xt + cyt);
      if (ys < 0 || ys >= iheight)
        continue;
      int32_t xs = (cosra * xt + cxt);
      if (xs < 0 || xs >= iwidth)
      {
        if (row_drawn)
          break;
        else
          continue;
      }

      xs=xs>>10; // This is allowed as we know that xs is >0
      ys=ys>>10; // This is allowed as we know that ys is >0
      row_drawn=true;
      if (x_start==0)
        x_start=x;
      x_end=x;
      uint16_t rp = readPixel(xs, ys);
      rp = rp<<8 | rp>>8;
      sline_buffer[byte_count]=rp; 
      
      //_tft->drawPixel(xt, yt, rp);
      byte_count++;
    }
    _tft->pushImage(x_start, y,x_end-x_start,1,sline_buffer);
  }

  return true;
}

```

Bodmer, 2019-12-31:
I have had a further look at the function and got it working and the performance improvement is good.

I found 3 bugs, some are subtle and may not show in a limited test scenario:
```
uint16_t sline_buffer[width()];
```
1. This line buffer is too small, a rotated sprite has a line longer than the width (i.e. it can be the diagonal).  On an ESP32 this will cause a crash, the ESP8266 is more tolerant and other variables may get corrupted causing odd behavior. Corrected line is:
```
uint16_t sline_buffer[max_y - min_y + 1];
```
2. A boundary case bug:
```
      if (x_start==0)  x_start=x;
```
The problem is that x_start = 0 is a valid start coordinate but on the next loop x_start will get set to 1 which is wrong, (should have stayed 0).

Corrected line is to have a boolean flag set to true at the start of the y loop e.g. "start_line", then use:
```
if (start_line) { x_start = x; start_line = false; }
```
Then x_start can then never be set twice.

3. Line length calculation wrong (1 pixel short) and not needed:
```
_tft->pushImage(x_start, y,x_end-x_start,1,sline_buffer);
```
Can be simplified to:
```
_tft->pushImage(x_start, y, byte_count, 1, sline_buffer);
```
This means that the x_end variable is not needed.

I have also added transparent handling and swapped back to column drawing as per original floating point function. The reason for this is the I use rotated sprites in landscape mode and you get significantly less "tearing effect" during screen updates if the rendering runs along the TFT refresh direction.

I have added the fixed point equivalent of the Round() function and this appears to eliminate the duplicated pixels which was the main reason I abandoned fixed point maths. 

So here is the final code:
```
/***************************************************************************************
** Function name:           pushRotated2
** Description:             Push a rotated Sprite to TFT screen using integer maths 
*************************************************************************************x*/

bool TFT_eSprite::pushRotated2(int16_t angle, int32_t transp)
{
  if ( !_created ) return false;

  // Trig values for the rotation
  float radAngle = -angle * 0.0174532925; // Convert degrees to radians
  float sinraf = sin(radAngle);
  float cosraf = cos(radAngle);

  int32_t sinra = sinraf * 1024;
  int32_t cosra = cosraf * 1024;

  // Bounding box parameters
  int16_t min_x;
  int16_t min_y;
  int16_t max_x;
  int16_t max_y;

  // Get the bounding box of this rotated source Sprite relative to Sprite pivot
  getRotatedBounds(sinraf, cosraf, width(), height(), _xpivot, _ypivot, &min_x, &min_y, &max_x, &max_y);

  // Move bounding box so source Sprite pivot coincides with TFT pivot
  min_x += _tft->_xpivot;
  max_x += _tft->_xpivot;
  min_y += _tft->_ypivot;
  max_y += _tft->_ypivot;

  // Test only to show bounding box on TFT
  // _tft->drawRect(min_x, min_y, max_x - min_x + 1, max_y - min_y + 1, TFT_GREEN);

  // Return if bounding box is outside of TFT area
  if (min_x > _tft->width()) return true;
  if (min_y > _tft->height()) return true;
  if (max_x < 0) return true;
  if (max_y < 0) return true;

  // Clip bounding box to be within TFT area
  if (min_x < 0) min_x = 0;
  if (min_y < 0) min_y = 0;
  if (max_x > _tft->width())  max_x = _tft->width();
  if (max_y > _tft->height()) max_y = _tft->height();

  uint32_t iheight = height()<<10;

  uint32_t iwidth = width()<<10;

  uint16_t sline_buffer[max_y - min_y + 1];

  // Scan destination bounding box and fetch transformed pixels from source Sprite
  for (int32_t x = min_x; x <= max_x; x++) {
    int32_t xt = x - _tft->_xpivot;
    int32_t cxt = cosra * xt + _xpivot * 1024;
    int32_t sxt = sinra * xt + _ypivot * 1024;
    bool column_drawn = false;
    uint16_t pixel_count=0;
    uint16_t y_start=0;
    bool start_line = true;
    for (int32_t y = min_y; y <= max_y; y++) {
      int32_t yt = y - _tft->_ypivot;
      int32_t xs = cxt - sinra * yt;
      // Do not calculate ys unless xs is in bounds
      if (xs >= 0 && xs < iwidth)
      {
        int32_t ys = sxt + cosra * yt;
        // Check if ys is in bounds
        if (ys >= 0 && ys < iheight) {
          int32_t rp = readPixel(((xs>>9) + (xs < 0 ? -1 : 1))>>1, ((ys>>9) + (ys < 0 ? -1 : 1))>>1);
          if (transp >= 0 ) {
            if (rp != transp) _tft->drawPixel(x, y, rp);
          }
          else {
            if (start_line) { y_start = y; start_line = false; }
            sline_buffer[pixel_count++] = rp>>8 | rp<<8;
          }
          column_drawn = true;
        }
      }
      else if (column_drawn) y = max_y; // Skip remaining column pixels
    }
    if (pixel_count) _tft->pushImage(x, y_start, 1, pixel_count, sline_buffer);
  }

  return true;
}
```

Testing in the Rotated_Sprite_2 example to draw the "needle" shows a 2x speed improvement by using a line buffer, and 2.8x speed improvement using integer maths.

Since the fixed point rounding has eliminated the pixel jitter I will replace the original slower function if further testing shows no issues.

So thanks for encouraging me to look at this again, the 5.7x performance improvement is very worthwhile.

2dom, 2019-12-31:
Thanks for the feed-back!
Looking forward to seeing the improvements in the next release :)

Happy new year to you and your family.

Dom

2dom, 2020-01-01:
Just a note ... pushing the image vertically costs about 2x in performance. Together with the improvements before we have about 10x better performance. This might help with visible "tearing effect" during screen updates?

Bodmer, 2020-01-01:
Performance will change depending on the aspect ratio of the sprite. I am guessing your tests used a wide sprite with small height rotated through small angles. A "tall" sprite will draw much faster with column rendering due to the fewer setAddrWindow commands to the TFT.

One way around this would be to make the function adapt to the aspect ratio of the rotated sprite.

If you share your performance test sketch then I will able able to see where you are losing performance. In my tests (incremental 360 rotation) there is zero difference between column or row rendering.

2dom, 2020-01-01:
I use example 2 with the following setup()

```
void setup()   {
  Serial.begin(9600); // Debug only

  tft.begin();
  tft.setRotation(0);
  tft.setPivot(120,120);

  // Clear TFT screen
  tft.fillScreen(TFT_NAVY);

  // Create the dial Sprite and dial (this temporarily hijacks the use of the needle Sprite)
  createDialScale(-120, 120, 30);   // create scale (start angle, end angle, increment angle)
  drawEmptyDial("Label", 12345);    // draw the centre of dial in the Sprite

  dial.pushSprite(110, 0);          // push a copy of the dial to the screen so we can see it

  // Create the needle Sprite
  createNeedle();                // draw the needle graphic
  needle.pushSprite(95, 7);    // push a copy of the needle to the screen so we can see it

  unsigned long start_time=millis();
  for (int16_t angle = 0; angle <= 360; angle += 1) 
   needle.pushFastRotated(angle); // Sprite is used to make scale markers
   // Avoid a watchdog time-out
  unsigned long delta=millis()-start_time;
  tft.drawString("My implementation: " + String(delta),0,200);

  delay(3000);
  start_time=millis();
  for (int16_t angle = 0; angle <= 360; angle += 1) 
    needle.pushRotated2(angle); // Sprite is used to make scale markers
   // Avoid a watchdog time-out
  delta=millis()-start_time;
  tft.drawString("Bodmers implementation: " + String(delta),0,200);
  delay(3000);
}
```

2dom, 2020-01-01:
Row rendering code:

```
bool TFT_eSprite::pushFastRotated(int16_t angle, int32_t transp)
{
  if ( !_created ) return false;

  // Trig values for the rotation
  float radAngle = -angle * 0.0174532925; // Convert degrees to radians
  float sinraf = sin(radAngle);
  float cosraf = cos(radAngle);

  int32_t sinra = sinraf * 1024;
  int32_t cosra = cosraf * 1024;

  // Bounding box parameters
  int16_t min_x;
  int16_t min_y;
  int16_t max_x;
  int16_t max_y;

  // Get the bounding box of this rotated source Sprite relative to Sprite pivot
  getRotatedBounds(sinraf, cosraf, width(), height(), _xpivot, _ypivot, &min_x, &min_y, &max_x, &max_y);

  // Move bounding box so source Sprite pivot coincides with TFT pivot
  min_x += _tft->_xpivot;
  max_x += _tft->_xpivot;
  min_y += _tft->_ypivot;
  max_y += _tft->_ypivot;

  // Test only to show bounding box on TFT
  // _tft->drawRect(min_x, min_y, max_x - min_x + 1, max_y - min_y + 1, TFT_GREEN);

  // Return if bounding box is outside of TFT area
  if (min_x > _tft->width()) return true;
  if (min_y > _tft->height()) return true;
  if (max_x < 0) return true;
  if (max_y < 0) return true;

  // Clip bounding box to be within TFT area
  if (min_x < 0) min_x = 0;
  if (min_y < 0) min_y = 0;
  if (max_x > _tft->width())  max_x = _tft->width();
  if (max_y > _tft->height()) max_y = _tft->height();

  uint32_t iheight = height()<<10;

  uint32_t iwidth = width()<<10;

  uint16_t sline_buffer[max_x - min_x + 1];

  // Scan destination bounding box and fetch transformed pixels from source Sprite
  for (int32_t y = min_y; y <= max_y; y++) {
    int32_t yt = y - _tft->_ypivot;
    
    int32_t cxt = - sinra * yt + _xpivot * 1024;
    int32_t sxt = + cosra * yt + _ypivot * 1024;
    bool row_drawn = false;
    uint16_t pixel_count=0;
    uint16_t x_start=0;
    bool start_line = true;
    for (int32_t x = min_x; x <= max_x; x++) {
      int32_t xt = x - _tft->_xpivot;  
      int32_t xs = cxt + cosra * xt ;
      // Do not calculate ys unless xs is in bounds
      if (xs >= 0 && xs < iwidth)
      {
        int32_t ys = sxt + sinra * xt;
        // Check if ys is in bounds
        if (ys >= 0 && ys < iheight) {
          int32_t rp = readPixel((xs + (xs < 0 ? -1<<9 : 1<<9))>>10, (ys + (ys < 0 ? -1<<9 : 1<<9))>>10);
          
          if (transp >= 0 ) {
            if (rp != transp) _tft->drawPixel(x, y, rp);
          }
          else {
            if (start_line) { x_start = x; start_line = false; }
            sline_buffer[pixel_count++] = rp>>8 | rp<<8;
          }
          row_drawn = true;
        }
      }
      else if (row_drawn) x = max_x; // Skip remaining column pixels
    }
    if (pixel_count) _tft->pushImage(x_start, y, pixel_count,1,sline_buffer);
  }

  return true;
}

```

2dom, 2020-01-01:
This gives me 439ms for 1st and 840ms for 2nd impementation.

Bodmer, 2020-01-01:
Interesting. The results I get using micros()/1000.0 for timing and an infinite test loop with ESP8266 at 160MHz, ILI9341 display, 40MHz SPI:
```
My implementation:      429.82
Bodmers implementation: 429.30
My implementation:      429.82
Bodmers implementation: 429.30
My implementation:      429.83
Bodmers implementation: 429.58
My implementation:      429.54
Bodmers implementation: 429.19
My implementation:      429.83
Bodmers implementation: 429.58
My implementation:      429.83
Bodmers implementation: 429.60
My implementation:      429.82
Bodmers implementation: 429.28
```

The code is now built into my master library copy now is as follows:
```
// pushRotated support - round a fixed point arithmetic integer value V scaled by S
#define roundFP(V,S) (V + (V < 0 ? -1<<(S-1) : 1<<(S-1)))

/***************************************************************************************
** Function name:           pushRotated
** Description:             Push rotated Sprite to TFT screen (fixed point maths)
*************************************************************************************x*/
bool TFT_eSprite::pushRotated(int16_t angle, int32_t transp)
{
  if ( !_created ) return false;

  // Trig values for the rotation
  float radAngle = -angle * 0.0174532925; // Convert degrees to radians
  float sinraf = sin(radAngle);
  float cosraf = cos(radAngle);

  int32_t sinra = sinraf * 1024;
  int32_t cosra = cosraf * 1024;

  // Bounding box parameters
  int16_t min_x;
  int16_t min_y;
  int16_t max_x;
  int16_t max_y;

  // Get the bounding box of this rotated source Sprite relative to Sprite pivot
  getRotatedBounds(sinraf, cosraf, width(), height(), _xpivot, _ypivot, &min_x, &min_y, &max_x, &max_y);

  // Move bounding box so source Sprite pivot coincides with TFT pivot
  min_x += _tft->_xpivot;
  max_x += _tft->_xpivot;
  min_y += _tft->_ypivot;
  max_y += _tft->_ypivot;

  // Test only to show bounding box on TFT
  // _tft->drawRect(min_x, min_y, max_x - min_x + 1, max_y - min_y + 1, TFT_GREEN);

  // Return if bounding box is outside of TFT area
  if (min_x > _tft->width()) return true;
  if (min_y > _tft->height()) return true;
  if (max_x < 0) return true;
  if (max_y < 0) return true;

  // Clip bounding box to be within TFT area
  if (min_x < 0) min_x = 0;
  if (min_y < 0) min_y = 0;
  if (max_x > _tft->width())  max_x = _tft->width();
  if (max_y > _tft->height()) max_y = _tft->height();

  uint32_t iheight = height()<<10;
[... 98 lines cut ...]
    float sxt = sinra * xt + _ypivot * 1024;
    bool column_drawn = false;
    for (int32_t y = min_y; y <= max_y; y++)
    {
      int32_t yt = y - spr->_ypivot;
      int32_t xs = cxt - sinra * yt;
      // Do not calculate ys unless xs is in bounds
      if (xs >= 0 && xs < iwidth)
      {
        int32_t ys = sxt + cosra * yt;
        // Check if ys is in bounds
        if (ys >= 0 && ys < iheight)
        {
          int32_t rp = readPixel( roundFP(xs, 10)>>10, roundFP(ys, 10)>>10 );
          if (rp != transp) spr->drawPixel(x, y, rp);
          column_drawn = true;
        }
      }
      else if (column_drawn) y = max_y; // Skip the remaining pixels below the Sprite
    }
  }

  return true;
}
```

Bodmer, 2020-01-01:
Just thinking .. I suspect you picked up my code from a Github email, but I edited it to correct with some uint64_t variables added for testing. Github would not send the edited copy, so the code would have to be pulled from edited Github web page. 64 bit variables are slower.

Bodmer, 2020-01-02:
Further tests look good for fidelity. I have moved some calculations outside the loop to give a small performance boost test time now 398ms instead of 429ms.
Performance profile tests indicate the algorithm time is ~178ms, pixelRead uses ~200ms of this time and rendering ~20ms. Since the algorithm already does bounds checks and rotation is for 16 bit pixels only I think the readPixel performance could be improved significantly but it is a law of diminishing returns.

Closing this not as this will be in the next update which will happen sometime in January I expect.

closed 2020-01-02 by Bodmer