Bodmer/TFT_eSPI #507: ST7735 1.44" 128x128 weird results

keejayzee, 2019-12-30:
I got this ST7735 working nicely with Adafruit's library. Using TFT_eSPI produces weird results using the examples with Arduino IDE with Setup7_ST7735_128x128.h

Any suggestions ? May this be a bug or perhaps an unsupported board ?

. Please, see attached pics.

**BOARD:**
[image]

**LocalCustomFonts example:**
[image]

**DrawXBitmap example:**
[image]

**UncannyEyes example:**
[image]

keejayzee, 2019-12-30:
Not sure if I'm violating some github rule, but it is exactly this:
https://www.aliexpress.com/item/32947890530.html?spm=a2g0s.9042311.0.0.49d24c4djwx4hT

keejayzee, 2019-12-30:
Output from Read_User_Setup:

```
TFT_eSPI ver = 1.4.21
Processor    = ESP8266
Frequency    = 80 MHz
Voltage      = 3.30 V
Transactions = No 
Interface    = SPI 
SPI overlap  = No 

Display driver = 7735
Display width  = 128 
Display height = 128 

R0 y offset = 32 
TFT_CS   = D8 (GPIO 15)
TFT_DC   = D1 (GPIO 5)
TFT_RST  = D0 (GPIO 16)

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 27.0 MHz 
```

Bodmer, 2020-01-01:
The problem is that the ST7735 controller supports displays up to 132x160. You have a 128x128 display and manufacturers map this to different positions in the 132x160 area available. So you have yet another variant of the possible mapping.

As the colours look OK, you will have to experiemtn to determine the offsets your display needs. To do this use the a setup like this:

#define ST7735_DRIVER

#define ST7735_GREENTAB

In rotation 0 try to establish where the corner pixels are, I can then create add this display variant to those available.

Bodmer, 2020-01-01:
Other settings to try are:

// #define ST7735_GREENTAB2
// #define ST7735_GREENTAB3

keejayzee, 2020-01-01:
It's apparently working now. Changing the define ST7735_GREENTAB128 for ST7735_GREENTAB3
as you said did the trick, so I guess it's a 128x128 GRENTAB3 variant. If you want me to do some more tests to check if this is really a new model or whatever, let met know.

```
TFT_eSPI ver = 1.4.21
Processor    = ESP8266
Frequency    = 80 MHz
Voltage      = 3.30 V
Transactions = No 
Interface    = SPI 
SPI overlap  = No 

Display driver = 7735
Display width  = 128 
Display height = 128 

R0 x offset = 2 
R0 y offset = 3 
TFT_CS   = D8 (GPIO 15)
TFT_DC   = D1 (GPIO 5)
TFT_RST  = D0 (GPIO 16)

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 27.0 MHz 

```
`

Bodmer, 2020-01-01:
OK, as it is a square display you will probably not need to use setRotation() for a different orientation but if you did you would get the odd behavior again.

To check if the offsets are indeed OK try:
tft.fillScreen(TFT~_BLACK);
tft.drawRect(0,0,128,128,TFT_GREEN);

You should see a green rectangle with no missing sides.

keejayzee, 2020-01-01:
This is the result.

[image]

Bodmer, 2020-01-01:
Looks good. So I will close.

closed 2020-01-01 by Bodmer