Bodmer/TFT_eSPI #509: M5Stack compile errors in Arduino ide (fonts...)

MooonWalker, 2019-12-31:
Hi,

first of all, thank you for your great effort.
I am trying to compile a test ino but getting tons of errors:

My sketch starts like this:
"#include <M5Stack.h>
#include <SPI.h>
#include <TFT_eSPI.h>

#define TFT_GREY 0x5AEB

TFT_eSPI tft = TFT_eSPI();       // Invoke custom library

void setup() ..."

The errors:......................
Arduino: 1.8.10 (Windows 10), Board: "M5Stack-Core-ESP32, QIO, 80MHz, Default, 921600, None"

In file included from C:\Users\almod\Documents\Arduino\libraries\M5Stack\src/Fonts/Font16.h:1:0,

                 from C:\Users\almod\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:66,

                 from C:\Users\almod\Documents\Arduino\tft_teszt\tft_teszt.ino:3:

C:\Users\almod\Documents\Arduino\libraries\M5Stack\src/Fonts/Font16.c:8:42: error: redefinition of 'const unsigned char widtbl_f16 [96]'

 PROGMEM const unsigned char widtbl_f16[96] =         // character width table

                                          ^

In file included from C:\Users\almod\Documents\Arduino\libraries\M5Stack\src/Fonts/Font16.h:1:0,

                 from C:\Users\almod\Documents\Arduino\libraries\M5Stack\src/utility/In_eSPI.h:66,

                 from C:\Users\almod\Documents\Arduino\libraries\M5Stack\src/M5Display.h:7,

                 from C:\Users\almod\Documents\Arduino\libraries\M5Stack\src/M5Stack.h:108,

                 from C:\Users\almod\Documents\Arduino\tft_teszt\tft_teszt.ino:1:

C:\Users\almod\Documents\Arduino\libraries\M5Stack\src/Fonts/Font16.c:8:29: note: 'const unsigned char widtbl_f16 [96]' previously defined here

 PROGMEM const unsigned char widtbl_f16[96] =         // character width table

                             ^

In file included from C:\Users\almod\Documents\Arduino\libraries\M5Stack\src/Fonts/Font16.h:1:0,

                 from C:\Users\almod\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:66,

                 from C:\Users\almod\Documents\Arduino\tft_teszt\tft_teszt.ino:3:

C:\Users\almod\Documents\Arduino\libraries\M5Stack\src/Fonts/Font16.c:26:42: error: redefinition of 'const unsigned char chr_f16_20 [16]'

 PROGMEM const unsigned char chr_f16_20[16] =         // 1 unsigned char per row

                                          ^

In file included from C:\Users\almod\Documents\Arduino\libraries\M5Stack\src/Fonts/Font16.h:1:0,

                 from C:\Users\almod\Documents\Arduino\libraries\M5Stack\src/utility/In_eSPI.h:66,

                 from C:\Users\almod\Documents\Arduino\libraries\M5Stack\src/M5Display.h:7,

                 from C:\Users\almod\Documents\Arduino\libraries\M5Stack\src/M5Stack.h:108,

                 from C:\Users\almod\Documents\Arduino\tft_teszt\tft_teszt.ino:1:

C:\Users\almod\Documents\Arduino\libraries\M5Stack\src/Fonts/Font16.c:26:29: note: 'const unsigned char chr_f16_20 [16]' previously defined here
[... 352 lines cut ...]

In file included from C:\Users\almod\Documents\Arduino\libraries\M5Stack\src/Fonts/Font16.h:1:0,

                 from C:\Users\almod\Documents\Arduino\libraries\M5Stack\src/utility/In_eSPI.h:66,

                 from C:\Users\almod\Documents\Arduino\libraries\M5Stack\src/M5Display.h:7,

                 from C:\Users\almod\Documents\Arduino\libraries\M5Stack\src/M5Stack.h:108,

                 from C:\Users\almod\Documents\Arduino\tft_teszt\tft_teszt.ino:1:

C:\Users\almod\Documents\Arduino\libraries\M5Stack\src/Fonts/Font16.c:104:29: note: 'const unsigned char chr_f16_2D [16]' previously defined here

 PROGMEM const unsigned char chr_f16_2D[16] =         // 1 unsigned char per row

                             ^

In file included from C:\Users\almod\Documents\Arduino\libraries\M5Stack\src/Fonts/Font16.h:1:0,

                 from C:\Users\almod\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:66,

Could you help in this please?

Thy
Attila

Bodmer, 2020-01-01:
The M5Stack library has a renamed copy of TFT_eSPI inside it called In_eSPI so you need to either use one library or the other, not both. I do not provide support for the M5Stack library

closed 2020-01-01 by Bodmer

MooonWalker, 2020-01-01:
> 
> 
> The M5Stack library has a renamed copy of TFT_eSPI inside it called In_eSPI so you need to either use one library or the other, not both. I do not provide support for the M5Stack library

Hi Bodmer, 
thank you for your reply.
I do not need M5Stack library support just wondered why it is not working. Maybe it would worth to mention this situation in the user settings header files.

My goal was to attach a 2nd external lcd to the M5Stack core using your library. So it seems i need to use In_eSPI.

Case can be closed.

Thank you 
Br
Attila