Bodmer/TFT_eSPI #510: STM32F103 support in the new STM32 beta branch

MiLeG, 2020-01-02:
Greetings,

I am working on a project using the super-cheap STM32F103 board and I was so glad to have seen you working on an STM32 adaptation for your library.
However, the STM32 branch isn't compatible with the STMF1xxx series and that's what I'd like to know more about.
Is there a hardware issue preventing you from writing compatible code for it?
If not, then were you planning on making it compatible with the STM32F1xxx variants?

Anyway, kudos for making such an awesome library!

Bodmer, 2020-01-03:
[Try commenting out this line](https://github.com/Bodmer/TFT_eSPI/blob/STM32/User_Setups/Setup_STM32_ILI9341.h#L17). The library will then use generic drive code that should work on any Arduino. The library will run slow with the [generic driver](https://github.com/Bodmer/TFT_eSPI/blob/STM32/Processors/TFT_eSPI_Generic.c) as there is no attempt at optimisation.

MiLeG, 2020-01-04:
Commenting the line still results in an SPI class error:
`lib\TFT_eSPI\TFT_eSPI.cpp:81:9: error: 'class SPIClass' has no member named 'setFrequency'`

I am compiling the code with the latest platform and "genericSTM32F103C8" board.
The core for the STM32F1xxx series seems to have some code specifics I'm not that determined to delve into. Will probably drop the whole idea of an STM32F103 LCD project.

Thanks for the help, though. I'll be keeping track of how this STM32 branch will be going, hopefully an STM32F1xxx version comes around someday.

Bodmer, 2020-01-04:
It looks like you are using a different STM32 core package or an old one that does not set SPI_HAS_TRANSACTION

[This is the package](https://github.com/stm32duino/Arduino_Core_STM32) I use.

If you include this in your setup file I suspect it will work.

#if !defined (SPI_HAS_TRANSACTION)
  #define SPI_HAS_TRANSACTION
#endif

Bodmer, 2020-01-04:
You should use [this setup as the starting point](https://github.com/Bodmer/TFT_eSPI/blob/STM32/User_Setups/Setup_STM32_ILI9341.h) and add the #if from the last post.

Bodmer, 2020-01-04:
I have ordered an STM32 "blue pill" to play with!

I was given an old "LeafLabs" STM32103 board but it does not seem to get recognised by the IDE, so I assume it needs a special driver or is broken. Looks like leaflabs are no longer trading and so docs are old.

Bodmer, 2020-01-04:
If you are not reading from the TFT you can comment out lines 81 and 103 in TFT_eSPI.cpp

Bodmer, 2020-01-06:
My "Blue pill" arrived today so I connected it up to an ILI9341 display and used the [STM32 setup](https://github.com/Bodmer/TFT_eSPI/blob/STM32/User_Setups/Setup_STM32_ILI9341.h) edited to use the following pin connections:
```
#define TFT_CS   A1 // Chip select control pin to TFT CS
#define TFT_DC   A0 // Data Command control pin to TFT DC (may be labelled RS = Register Select)
#define TFT_RST  A2 // Reset pin to TFT RST (or RESET)
```
Also wired power and LED plus:
A5 -> TFT SCK
A7 -> TFT SDI (MOSI)
The simple sketches (e.g. graphicstest) compiled OK and ran fine with good performance.

Some ESP32/ESP8266 sketches are too big for the RAM an FLASH size to run on a Blue Pill.

Make sure you have the[ "official" STM Arduino board support](https://github.com/stm32duino/Arduino_Core_STM32) package loaded.

If you are still having problems then post back. There is pixel aliasing with the camera and TFT in this picture but it captures the test setup.

[image]

Bodmer, 2020-01-11:
Closing as I suspect you are not using the [STM board package](https://github.com/stm32duino/Arduino_Core_STM32).

closed 2020-01-11 by Bodmer

holsteinjohan, 2020-02-08:
Hi,
 I use a ILI9488 and get some errors see below, is there anything I can do? to get this working.

Best regards,
Johan Holstein

In file included from /home/holstein/Arduino/libraries/TFT_eSPI-master/TFT_eSPI.h:759:0,
                 from /home/holstein/Arduino/libraries/TFT_eSPI-master/examples/480 x 320/UTFT_Demo_480x320/UTFT_Demo_480x320.ino:25:
/home/holstein/Arduino/libraries/TFT_eSPI-master/Extensions/Smooth_font.h:7:38: error: 'fs' has not been declared
   void     loadFont(String fontName, fs::FS &ffs);
                                      ^
/home/holstein/Arduino/libraries/TFT_eSPI-master/Extensions/Smooth_font.h:7:45: error: expected ',' or '...' before '&' token
   void     loadFont(String fontName, fs::FS &ffs);
                                             ^
/home/holstein/Arduino/libraries/TFT_eSPI-master/Extensions/Smooth_font.h:40:3: error: 'fs' does not name a type
   fs::File fontFile;
   ^
/home/holstein/Arduino/libraries/TFT_eSPI-master/Extensions/Smooth_font.h:47:3: error: 'fs' does not name a type
   fs::FS   &fontFS = SPIFFS;
   ^
exit status 1
Error compiling for board Generic STM32F103C series.

Bodmer, 2020-02-08:
Yes, comment out the smooth font enable line in the setup file. See the [comment here](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup29_ILI9341_STM32.h#L53)

holsteinjohan, 2020-02-08:
Still some errors...
Best,
Johan

/home/holstein/Arduino/libraries/TFT_eSPI-master/TFT_eSPI.cpp: In member function 'int16_t TFT_eSPI::drawNumber(long int, int32_t, int32_t)':
/home/holstein/Arduino/libraries/TFT_eSPI-master/TFT_eSPI.cpp:3883:25: error: 'ltoa' was not declared in this scope
   ltoa(long_num, str, 10);
                         ^

Bodmer, 2020-02-08:
Have you loaded the "official" STM32 Arduino board package?
If so, close the Arduino IDE and open again.
Roger Clark's version of the STM32 board package will not work.

hamed-ta, 2020-04-28:
Hi,

I have the same problem 

`lib\TFT_eSPI\TFT_eSPI.cpp:81:9: error: 'class SPIClass' has no member named 'setFrequency'`

I'm using PlatformIo, it install the packages which are needed so it doesn't compile, how can I add "official" STM32 Arduino board package in platfomIo for STM32F1 ?

Bodmer, 2020-04-28:
I do not use PlatformIO. Maybe try the Arduino IDE or [read this](https://docs.platformio.org/en/latest/platforms/ststm32.html) which suggests you can use the official board package if your configuration options are correct.

hamed-ta, 2020-04-29:
it's my configs in PlatformIO

```
[env:genericSTM32F103RC]
platform = ststm32
#board = genericSTM32F103RC
board = bluepill_f103c6
board_build.mcu = stm32f103rct6
framework = arduino
upload_protocol = stlink
board_build.f_cpu = 72000000L

build_flags =
  -Os
  -DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG
  -DUSER_SETUP_LOADED=1
  -DST7789_DRIVER=1
  -DTFT_WIDTH=240
  -DTFT_HEIGHT=240
  -DTFT_CS=-1
  -DTFT_DC=PA2
  -DTFT_RST=PA3
  -DLOAD_GLCD=1
  -DSPI_FREQUENCY=32000000
  -DSTM32=1
```

my board is a custom board with STM32F103RC but it only compiles when I use bluepill_f103c6 as my board in the config which doesn't work on my board as the MCU is diffrent so and if I use genericSTM32F103RC as board it doesn't compile see

<Details><Summery>Compile Output</Summery>

```
Processing genericSTM32F103RC (platform: ststm32; board: genericSTM32F103RC; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 6.1.0 > STM32F103RC (48k RAM. 256k Flash)    
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32-maple 2.10000.200103 (1.0.0)        
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 31 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TFT_eSPI> 2.2.2
|   |-- <SPI> 1.0   
Building in release mode
Compiling .pio\build\genericSTM32F103RC\lib399\TFT_eSPI_ID1559\TFT_eSPI.cpp.o
In file included from C:\Users\Hamed\.platformio\lib\TFT_eSPI_ID1559\TFT_eSPI.cpp:24:0:
C:\Users\Hamed\.platformio\lib\TFT_eSPI_ID1559\Processors/TFT_eSPI_STM32.c:16:3: error: 'SPI_HandleTypeDef' does not name a type
   SPI_HandleTypeDef spiHal;
   ^~~~~~~~~~~~~~~~~
In file included from C:\Users\Hamed\.platformio\lib\TFT_eSPI_ID1559\TFT_eSPI.cpp:24:0:
C:\Users\Hamed\.platformio\lib\TFT_eSPI_ID1559\Processors/TFT_eSPI_STM32.c: In member function 'void TFT_eSPI::pushBlock(uint16_t, uint32_t)':
C:\Users\Hamed\.platformio\lib\TFT_eSPI_ID1559\Processors/TFT_eSPI_STM32.c:327:23: error: 'spiHal' was not declared in this scope
     HAL_SPI_Transmit(&spiHal, (uint8_t*)col, len<<1, HAL_MAX_DELAY);
                       ^~~~~~
C:\Users\Hamed\.platformio\lib\TFT_eSPI_ID1559\Processors/TFT_eSPI_STM32.c:327:23: note: suggested alternative: 'spi'
     HAL_SPI_Transmit(&spiHal, (uint8_t*)col, len<<1, HAL_MAX_DELAY);
                       ^~~~~~
                       spi
C:\Users\Hamed\.platformio\lib\TFT_eSPI_ID1559\Processors/TFT_eSPI_STM32.c:327:54: error: 'HAL_MAX_DELAY' was not declared in this scope
     HAL_SPI_Transmit(&spiHal, (uint8_t*)col, len<<1, HAL_MAX_DELAY);
                                                      ^~~~~~~~~~~~~
C:\Users\Hamed\.platformio\lib\TFT_eSPI_ID1559\Processors/TFT_eSPI_STM32.c:327:54: note: suggested alternative: 'TFT_INIT_DELAY'
     HAL_SPI_Transmit(&spiHal, (uint8_t*)col, len<<1, HAL_MAX_DELAY);
[... 802 lines cut ...]
           ^~~~
In file included from C:\Users\Hamed\.platformio\lib\TFT_eSPI_ID1559\TFT_eSPI.h:28:0,
                 from C:\Users\Hamed\.platformio\lib\TFT_eSPI_ID1559\TFT_eSPI.cpp:17:
C:\Users\Hamed\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\SPI\src/SPI.h: At global scope:
C:\Users\Hamed\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\SPI\src/SPI.h:171:15: warning: '_spi3_this' defined but not used [-Wunused-variable]
 static void (*_spi3_this);
               ^~~~~~~~~~
C:\Users\Hamed\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\SPI\src/SPI.h:168:15: warning: '_spi2_this' defined but not used [-Wunused-variable]
 static void (*_spi2_this);
               ^~~~~~~~~~
C:\Users\Hamed\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\SPI\src/SPI.h:165:15: warning: '_spi1_this' defined but not used [-Wunused-variable]
 static void (*_spi1_this);
               ^~~~~~~~~~
C:\Users\Hamed\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\SPI\src/SPI.h:163:16: warning: 'ff' defined but not used [-Wunused-variable]
 static uint8_t ff = 0XFF;
                ^~
*** [.pio\build\genericSTM32F103RC\lib399\TFT_eSPI_ID1559\TFT_eSPI.cpp.o] Error 1
====================================================================================== [FAILED] Took 4.14 seconds ======================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

```

</details>

hamed-ta, 2020-04-29:
it the board and I could get it work with the adafruit ST7789 library but for the Smoth font capability I need to use TFT_eSPI in PlatformIo.

[image]

hamed-ta, 2020-04-29:
as you can see below for test I installed the Arduino IDE, loaded the latest STM32 Core and added the TFT_eSPI lib and example sketch it compiles and programms but doesn't work on the board, what's the problem ?

[image]
[image]
[image]
[image]

Bodmer, 2020-04-29:
You MUST edit the User_Setup.h file INSIDE the library folder. The Arduino IDE does NOT allow a library to see #define's in a sketch.

hamed-ta, 2020-04-30:
I did, but it doesn't work again !

[image]

Bodmer, 2020-05-01:
I have connected one of those displays to a STM32F103 blue pill board and it does not work as you have discovered.  If I change the processor to an ESP32 or ESP8266 it does work so I am not sure what is happening with the STM32.

I suspect that the display is not being configured properly for some reason by the STM32 processor even though the exact same configuration works with the ESP boards.  I have checked with an oscilloscope and the signals on the STM32 are toggling as expected.

The display does not use a chip select (CS) line, this means that the synchonism of the byte boundaries can be lost if there are spurious clocks, this is a guess at the reason the display does not work with the STM32 processor.

I did discover a bug, the TFT_CS pin must either not be defined at all, or it must be set to a valid pin number. If it is set to -1 then the processor will crash when the library instance is created. This appears ultimately to be a bug in the STM32 board package. I will update the library to get around that bug but it does not fix the problem with this display.

I have spent a few hours trying to determine why this display does not work with the STM32, further investigation will have to wait.

Unfortunately I have no solution for you at the moment.

Bodmer, 2020-05-01:
Update:
If I comment out the **#define STM32** in the setup file then the basic Arduino SPI calls are used for a generic processor. The display then works, but very slowly because the standard Arduino SPI calls are not optimised for the processor.

I will think further on this but at least you can start to test some code.

Bodmer, 2020-05-01:
I have identified a small bit of code that appears to be causing an issue just for this particular ST7789 chip. If you use the #define STM32 in your setup file and [replace these three lines](https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_STM32.h#L834-L836) with this one:
```
#define tft_Write_8(C)   spi.transfer(C)
```
, then it works with better performance than the generic driver. No idea why the HAL SPI driver behaves differently for a 1 byte transfer.

hamed-ta, 2020-05-01:
> I have connected one of those displays to a STM32F103 blue pill board and it does not work as you have discovered. If I change the processor to an ESP32 or ESP8266 it does work so I am not sure what [...]

yes as you said, the processor crashes and I have no signals on any pins even on reset or DC pins, but just by commenting out the TFT_CS pin definition, the processor starts working and I have signals on the pins but the LCD doesn't work.

but after commenting out the STM32 definition as well, as you said the LCD works but really slow and also.

in this type of LCD the CS pin is directly connected to VCC.

hamed-ta, 2020-05-01:
> I have identified a small bit of code that appears to be causing an issue just for this particular ST7789 chip. If you use the #define STM32 in your setup file and [replace these three [...]

it didn't make change but by changing the lines below to what you said, now it's working well 

`////////////////////////////////////////////////////////////////////////////////////////
// Macros for all other SPI displays
////////////////////////////////////////////////////////////////////////////////////////
#else

  #define tft_Write_8(C)   spi.transfer(C)`

hamed-ta, 2020-05-01:
the only thing that is needed, is a 240x240 pixels graphics test.

Bodmer, 2020-05-01:
The 240x320 graphics test should work as it adapts to screen size.

hamed-ta, 2020-05-02:
the problem also solved in PlatformIO see [here](https://community.platformio.org/t/how-to-force-platformio-to-use-the-latest-arduino-core-stm32-in-ststm32-platform/13388/6 )

dkrausz, 2022-04-18:
I am trying to use this library with STM32F103 and display SPI St7796, but every time i recived this message from arduino IDE..

C:\Users\danil\OneDrive\Documentos\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'void TFT_eSPI::begin_tft_read()':
C:\Users\danil\OneDrive\Documentos\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:140:9: error: 'class SPIClass' has no member named 'setFrequency'
     spi.setFrequency(SPI_READ_FREQUENCY);
this is my user setup.
`//                            USER DEFINED SETTINGS
//   Set driver type, fonts to be loaded, pins used and SPI control method etc
//
//   See the User_Setup_Select.h file if you wish to be able to define multiple
//   setups and then easily select which setup file is used by the compiler.
//
//   If this file is edited correctly then all the library example sketches should
//   run without the need to make any more changes for a particular hardware setup!
//   Note that some sketches are designed for a particular TFT pixel width/height

// ##################################################################################
//
// Section 1. Call up the right driver file and any options for it
//
// ##################################################################################

// Define STM32 to invoke optimised processor support (only for STM32)
#define STM32

// Defining the STM32 board allows the library to optimise the performance
// for UNO compatible "MCUfriend" style shields
//#define NUCLEO_64_TFT
//#define NUCLEO_144_TFT

// STM32 8 bit parallel only:
// If STN32 Port A or B pins 0-7 are used for 8 bit parallel data bus bits 0-7
// then this will improve rendering performance by a factor of ~8x
//#define STM_PORTA_DATA_BUS
//#define STM_PORTA_DATA_BUS

// Tell the library to use 8 bit parallel mode (otherwise SPI is assumed)
//#define TFT_PARALLEL_8_BIT

// Display type -  only define if RPi display
//#define RPI_DISPLAY_TYPE // 20MHz maximum SPI

// Only define one driver, the other ones must be commented out
//#define ILI9341_DRIVER
//#define ST7735_DRIVER      // Define additional parameters below for this display
//#define ILI9163_DRIVER     // Define additional parameters below for this display
//#define S6D02A1_DRIVER
//#define RPI_ILI9486_DRIVER // 20MHz maximum SPI
//#define HX8357D_DRIVER
//#define ILI9481_DRIVER
//#define ILI9486_DRIVER
//#define ILI9488_DRIVER     // WARNING: Do not connect ILI9488 display SDO to MISO if other devices share the SPI bus (TFT SDO does NOT tristate when CS is high)
//#define ST7789_DRIVER      // Full configuration option, define additional parameters below for this display
//#define ST7789_2_DRIVER    // Minimal configuration option, define additional parameters below for this display
//#define R61581_DRIVER
//#define RM68140_DRIVER
#define ST7796_DRIVER
//#define SSD1963_480_DRIVER
//#define SSD1963_800_DRIVER
//#define SSD1963_800ALT_DRIVER
//#define ILI9225_DRIVER
//#define GC9A01_DRIVER

// Some displays support SPI reads via the MISO pin, other displays have a single
// bi-directional SDA pin and the library will try to read this via the MOSI line.
// To use the SDA line for reading data from the TFT uncomment the following line:

// #define TFT_SDA_READ      // This option is for ESP32 ONLY, tested with ST7789 and GC9A01 display only

// For ST7735, ST7789 and ILI9341 ONLY, define the colour order IF the blue and red are swapped on your display
[... 254 lines cut ...]
// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  12000000 

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY  2500000 

// The ESP32 has 2 free SPI ports i.e. VSPI and HSPI, the VSPI is the default.
// If the VSPI port is in use and pins are not accessible (e.g. TTGO T-Beam)
// then uncomment the following line:
//#define USE_HSPI_PORT

// Comment out the following #define if "SPI Transactions" do not need to be
// supported. When commented out the code size will be smaller and sketches will
// run slightly faster, so leave it commented out unless you need it!

// Transaction support is needed to work with SD library but not needed with TFT_SdFat
// Transaction support is required if other SPI devices are connected.

// Transactions are automatically enabled by the library for an ESP32 (to use HAL mutex)
// so changing it here has no effect

// #define SUPPORT_TRANSACTIONS
#if !defined (SPI_HAS_TRANSACTION)
#define SPI_HAS_TRANSACTION
#endif`

Bodmer, 2022-04-18:
It sounds like you do not have the STM32 Arduino board support package loaded? See here for instructions:
https://github.com/stm32duino/Arduino_Core_STM32

dkrausz, 2022-04-18:
looks like its works, i was using a difent package, but i still not able to program.. i was using STlink but i didint find this upload method. Thanks for helping.

dkrausz, 2022-04-18:
Now it works! but i have some issues, only one exemple worked, all the other exemples said it does not have flash enought to compile. i am using a bluepil.  i try this code below, and when i put the line lcd.setTextColor(TFT_YELLOW); the counter dosent work, it shows me 0.08, after 0.88 then 8.88... if i dont choose the color the counter works fine... 

`#include<SPI.h>

#include<TFT_eSPI.h>

TFT_eSPI lcd = TFT_eSPI();

float i=0.00;

void setup() {
  lcd.init();
  lcd.setRotation(0);

  lcd.fillScreen(TFT_BLACK);
  lcd.setTextColor(TFT_YELLOW);
  lcd.setCursor(1,1);
  lcd.setTextFont(7);   

}

void loop() {
  
   lcd.setCursor(1,1);
    lcd.print(i,2);
delay(1000);
i+=0.03;
}`

Bodmer, 2022-04-18:
Glad you have it working!
The processor has limited Flash and RAM so you will need to be careful and disable any fonts you do not use in the setup file.

To over-write previous characters you need to specify a background colour. I suggest you read the tutorial here:
https://learn.adafruit.com/adafruit-gfx-graphics-library/graphics-primitives#characters-and-text-2002798-24

dkrausz, 2022-04-19:
It;s like an adafruit library... I thought weird cause with no color
select works fine

algavrik, 2022-05-14:
Has anyone succeed initializing ILI9481 display controller w/ 5-pins SPI? Read through all suggestions here, tried

#define STM32
#define ILI9481_DRIVER

#define TFT_SDI PA7
#define TFT_SCK PA5
#define TFT_DC   PA0
#define TFT_CS PA1
#define TFT_RST  PA2

#define SPI_FREQUENCY    15000000

and that configuration:

#define STM32
#define ILI9481_DRIVER
#define TFT_SPI_PORT 1
#define TFT_MOSI PA7
#define TFT_SCLK PA5
#define TFT_DC   PA2
#define TFT_CS PA4
#define TFT_RST  PA3

#define SPI_FREQUENCY    15000000

With and w/o STM32 definition, MOSI and SCK pins. No errors compilation using STM32duino ver 2.2.0. Blue pill.
Display works fine with TFT_eSPI compiled for EX8266.

Bodmer, 2022-05-14:
Yes, this has been tested but that was quite a while ago.

Example setup file here (read the comments):
https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup32_ILI9341_STM32F103.h

Comment out line 12
Uncomment line 13
The ILI9481 will need a lower SPI clock frequency, change line 53 to #define SPI_FREQUENCY  10000000

algavrik, 2022-05-14:
Got the picture! Thank you.