Bodmer/TFT_eSPI #511: Can not compile , after the upgrade

cagabit, 2020-01-03:
Hi,
I am sure this is a minor issue or even  non-lib related but could not solve it for sometime and stuck, any direction will be usefull.

I have a project using TFT_eSPI library with many other libs, after the arduino upgrade I started getting some errors, and getting similar ones with even example of the TFT lib, so in order to isolate the problem here is what I did:

- uninstall arduino, clean all the folders, setup fresh v1.8.10
- define additional board manager URL and from board manager install  esp8266 v2.6.3
- from lib manager install TFT_eSPI v1.4.20
- open known simplest example from TFT lib, "On_Off_Button"
- to give the TFT lib min settings and set not to load user_setup.h file

`#define TFT_CS   PIN_D0  
#define TFT_DC   PIN_D2  
#define TFT_RST  -1   
#define TOUCH_CS PIN_D3     // Chip select pin (T_CS) of touch screen
#define ILI9341_DRIVER

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts`

**and getting this error;**

`c:/users/caga/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: sketch\On_Off_Button.ino.cpp.o: in function `greenBtn()':

C:\Users\Caga\AppData\Local\Temp\arduino_modified_sketch_160301/On_Off_Button.ino:220: undefined reference to `TFT_eSPI::getTouch(unsigned short*, unsigned short*, unsigned short)'

c:/users/caga/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Caga\AppData\Local\Temp\arduino_modified_sketch_160301/On_Off_Button.ino:222: undefined reference to `TFT_eSPI::getTouch(unsigned short*, unsigned short*, unsigned short)'

c:/users/caga/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: sketch\On_Off_Button.ino.cpp.o:(.text._Z15touch_calibratev+0x20): undefined reference to `TFT_eSPI::setTouch(unsigned short*)'

c:/users/caga/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: sketch\On_Off_Button.ino.cpp.o:(.text._Z15touch_calibratev+0x24): undefined reference to `TFT_eSPI::calibrateTouch(unsigned short*, unsigned int, unsigned int, unsigned char)'

Multiple libraries were found for "SPI.h"
c:/users/caga/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: sketch\On_Off_Button.ino.cpp.o: in function `touch_calibrate()':

 Used: C:\Users\Caga\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\SPI
Multiple libraries were found for "TFT_eSPI.h"
 Used: D:\Users\caga\Documents\Arduino\libraries\TFT_eSPI
C:\Users\Caga\AppData\Local\Temp\arduino_modified_sketch_160301/On_Off_Button.ino:131: undefined reference to `TFT_eSPI::setTouch(unsigned short*)'

c:/users/caga/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Caga\AppData\Local\Temp\arduino_modified_sketch_160301/On_Off_Button.ino:152: undefined reference to `TFT_eSPI::calibrateTouch(unsigned short*, unsigned int, unsigned int, unsigned char)'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Generic ESP8266 Module.
`

My main project is also getting similar unrelated errors and if I comment out the display functions it compiles ok.

I am forgetting to do something but what ??

Bodmer, 2020-01-03:
The error messages occur when TOUCH_CS is not defined. Although you have set this is a file somewhere the User_Setup_Select.h file is not pointing to it.

Assuming you start with a new uneditted clean install of TFT_eSPI you just need to [uncomment this one line](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L135) and the example will compile. Do not make any other changes to the library until you have tried this.

You can [run this diagnostic example](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Test%20and%20diagnostics/Read_User_Setup) to see if the correct setup file and settings are being picked up by the compiler.

cagabit, 2020-01-04:
Hi Bodmer,
Thank you for your guidence.
Tested as you wrote , as suspected no problems ! Then I wonder what is the problem in my test setup?
As far as I can understand the problem is with the 

> #define USER_SETUP_LOADED //for not loading internal USER_SETTINGS

which actually prevents the original User_Setup.h file to be read but by testing preprocessor message command I saw that this define is not seen, and the compiler reads the User_Setup.h file.
In my opinion ( if not missing something about the preprocessor commands) this is related with arduino, so all is good.

Thank you again for this and the great lib you created.

closed 2020-01-04 by cagabit

Bodmer, 2020-01-04:
USER_SETUP_LOADED is there to allow users of the PlatformIO IDE to include the setup information in a project file. It does not affect Arduino IDE users.
 If you edit the User_Setup_select.h file the do make sure onky one setup file is called and all others are commented out.

Glad to hear you have it working.

cagabit, 2020-01-05:
Was trying to call the lib without changing any lines in it, it is another to track when the version changes.
The plan was to create a *.h file in project which has the needed settings lines ( like touch, pins etc) and call the lib.
I read your notes about platform.io, but it is a standart .h file and preprocessor commands, there is #ifndef at the start of the User_Setup_Select.h file and normally if I define it it should not enter.
Nevermind, Its ok with user_setup files you created.

Thank you.