Bodmer/TFT_eSPI #513: Flickering when pushSprite on a other sprite

Tinkerpete, 2020-01-05:
First: Your library is very good and i have made a lot of little ESP32 programs with it. But i have a big problem and have tried a few days for solve it. But now i give up. Please test this code, after a while you can see, that the upper rectangle ocasionally flicker. What can i do for avoid this?

#include <TFT_eSPI.h>

#define SCREENWIDTH 135
#define SCREENHEIGHT 240

TFT_eSPI tft = TFT_eSPI(SCREENWIDTH, SCREENHEIGHT);
TFT_eSprite upperSprite  = TFT_eSprite(&tft);
TFT_eSprite lowerSprite = TFT_eSprite(&tft);

void setup() {
  tft.init();
  tft.setRotation(0);
  tft.fillRect(0, 0, SCREENWIDTH, SCREENHEIGHT, TFT_BLACK);
  upperSprite.createSprite(SCREENWIDTH, SCREENHEIGHT / 2);
  lowerSprite.createSprite(52, 52);
  lowerSprite.fillRect(1, 1, 50, 50, TFT_RED);
}

void loop() {
  int y = 0;
  for (int x = 0; x < SCREENWIDTH; x++) {
    upperSprite.pushSprite(0, 0);
    lowerSprite.pushSprite(x, y);
    lowerSprite.pushSprite(x, y + 120);
    y++;
    delay (333);
  }
}

Bodmer, 2020-01-06:
It will flicker as you are drawing a red sprite, then drawing a black sprite on top before drawing the red sprite again 1 pixel away. So pixels are toggled black-red-black etc. Why are you drawing the upperSprite at all?

You can use pushImage with a transparent colour,  but you must use a pointer to the sprite. For a 16 bit sprite:
uint16_t* ptr = lowerSprite.createSprite(52, 52);
... draw in sprite.. with part being a colour you want transparent
tft.pushImage(x ,y, w, h, ptr, tranparent_color);

What are you trying to achieve?

Tinkerpete, 2020-01-06:
Thank you for your answer!

I did not understand what you mean with black-red-black? The 1-pixel frame is for overwriting the red color when moving the red rectangle and when i not use the sprite. My understanding is, when i insert a sprite to the tft with pushSprite, there is no flickering. That works fine. I had think, that this is also no problem, when i insert a second sprite over the area from the first sprite. This seems not true.

In my original program the black sprite is a star sky with stars with animated different intensities. Over this star sky i want to paint a moving moon image, stored in a const array, either with pushImage or with pushSprite every few milliseconds. The moon ist stored like this:

const uint16_t PROGMEM Moon [] = {0x0000, 0x0000, 0x0000, 0x0000, 0x0000, ...

Because the moon is round, i want see stars through the edges from the moon. Therefore i need transparency, either with pushimage or pushsprite. I think the normal way should be transparency with pushsprite, but this did not work. When i did NOT use pushSprite, and paint the moon with pushImage directly in the sky, there is no flickering, but also no transparency. When i use pushSprite the moon after pushSprite the sky, there is flickering.

When i try your suggestion:

uint16_t* ptr = moonSprite.createSprite(51, 51)

i get this;

'invalid conversion from 'void*' to 'uint16_t* {aka short unsigned int*}' [-fpermissive]'

Can be, that i'm not good enough in c++.

Tinkerpete, 2020-01-06:
I work with the Arduino IDE. Therefore i had the 'conversion error'! I must cast it like:

uint16_t* ptr = (uint16_t*)lowerSprite.createSprite(52, 52);

I have changed the program to this:

#include <TFT_eSPI.h>

#define SCREENWIDTH 135
#define SCREENHEIGHT 240

TFT_eSPI tft = TFT_eSPI(SCREENWIDTH, SCREENHEIGHT);
TFT_eSprite upperSprite  = TFT_eSprite(&tft);
TFT_eSprite lowerSprite = TFT_eSprite(&tft);

uint16_t* ptr;

void setup() {
  tft.init();
  tft.setRotation(0);
  tft.fillRect(0, 0, SCREENWIDTH, SCREENHEIGHT, TFT_BLACK);
  upperSprite.createSprite(SCREENWIDTH, SCREENHEIGHT / 2);
  ptr = (uint16_t*)lowerSprite.createSprite(52, 52);
  lowerSprite.fillRect(1, 1, 50, 50, TFT_RED);
}

void loop() {
  int y = 0;
  for (int x = 0; x < SCREENWIDTH; x++) {
    upperSprite.pushSprite(0, 0);
    tft.pushImage(x ,y, 52, 52, ptr, TFT_BLACK);
    lowerSprite.pushSprite(x, y + 120);
    y++;
    delay (333);
  }
}

But there is still flickering by the top rectangle.

Bodmer, 2020-01-06:
Yes, I ran your sketch. My explanation is correct. I have slowed down your sketch with delays so you can see what is happening. 

```
#include <TFT_eSPI.h>

#define SCREENWIDTH 135
#define SCREENHEIGHT 240

TFT_eSPI tft = TFT_eSPI(SCREENWIDTH, SCREENHEIGHT);
TFT_eSprite upperSprite = TFT_eSprite(&tft);
TFT_eSprite lowerSprite = TFT_eSprite(&tft);

void setup() {
  tft.init();
  tft.setRotation(0);
  tft.fillRect(0, 0, SCREENWIDTH, SCREENHEIGHT, TFT_BLACK);
  upperSprite.createSprite(SCREENWIDTH, SCREENHEIGHT / 2);
  upperSprite.fillSprite(TFT_GREEN);
  lowerSprite.createSprite(52, 52);
  lowerSprite.fillRect(1, 1, 50, 50, TFT_RED);
}

void loop() {
  int y = 0;
  for (int x = 0; x < SCREENWIDTH; x++) {
    upperSprite.pushSprite(0, 0);
    delay(200);
    lowerSprite.pushSprite(x, y);
    delay(200);
    lowerSprite.pushSprite(x, y + 120);
    delay(200);
    y++;
    delay (333);
  }
}
```
The uppersprite is filled with green just so you can see it.

The flickering appears inconsistent in the fast loop because the redraw is aliasing with the screen refresh rate built into the TFT. Using sprites is only flicker free if you create a composite sprite containing all the graphics in the memory before pushing to screen. The screen does not have image layers that you can sprite to individually.

The Sprite class does not have a function to plot the moon image in it with transparency, this is not something I have needed (but could be added and another option is to add a mask to a Sprite).

Now I understand what you are trying to achieve I suggest the following approach:
1. Use a single sprite to just contain the moon image.
2. Draw the stars on the screen (in fixed positions)
3. In the animation loop:
   a) Clear the sprite to black
   b) Draw the moon in the sprite with black for the sky
   c) Draw the stars in the sprite, only if the sprite pixel is black (sprite.readPixel(x,y) == 0).
      The star coordinates in the sprite must be shifted based on where the sprite will be plotted relative
      to 0,0 on the screen.
   d) Plot the moon sprite in the new position on the screen

To plot the stars create a function that accepts the coordinate offset and plots ALL the stars. Use it to plot to the screen with offset 0,0 then use it to plot to the sprite with an offset. If the sprite is to be plotted at 20,50 on the screen, then use offsets -20, -50 when plotting to the sprite. It is OK to attempt to plot outside of the sprite. See point 3 c) above. The library will just ignore writes outside of the sprite area.

Tinkerpete, 2020-01-06:
Many thanks for the detailed information! I hope I understand everything. My idea of your library was that each sprite has its own memory area and is copied to or into screen memory (tft) flicker-free with 'pushSprite'. The second 'pushSprite' would only have to overwrite the information in screen memory that is covered by this sprite? If this is the case, I still don't understand why this works with the first pushSprite and flickers with the second pushSprite when overwriting it?

It seems that the display works completely different in terms of hardware than I had imagined so far.

I have understood your suggestion and maybe I will implement it that way. However, any usability of the library in this area will be lost. If you want to make it easier for the user, there should be a 'pushImage' with transparency. As far as I can tell, most of the code is already available in the library.

It's very nice of you to take so much time for my problem!

Bodmer, 2020-01-06:
OK, will close but post again it you need help. Thanks for the feedback.

closed 2020-01-06 by Bodmer

JasXSL, 2022-04-11:
Just leaving a comment here because I have the same issue, and I'm not sure about a workaround. I'm using 3 layers stacked:

1. A background sprite with a tiling pattern. Rendering this as a non sprite takes seconds.
2. A rotating sprite. 
3. A large text on top of the sprite which covers most of the screen length.

Am I right in assuming the only way to make it not flicker each time I want to update a rotation is to make essentially the whole screen a sprite, and pushing to that?