Bodmer/TFT_eSPI #514: SD Card issue on ILI9341 3.2 inch TFT

stephensaid, 2020-01-05:
First and foremost, I have tried your library and the graphic test is extremely fast. Thank you.
I need to use the SD card on the TFT and this is where I got stuck.

[image]
[image]

With the above wiring, I have successfully run both graphics tests, which I may say are unbelievably fast!

I am now trying to get the SD card working using EPS32_Smooth_Font_SD using the same setup file I used for the other tests. I loaded a newly formatted SD card FAT32 with the font file as instructed in the comments and I receive the following message:

23:54:52.013 -> rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
23:54:52.013 -> configsip: 0, SPIWP:0xee
23:54:52.013 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
23:54:52.013 -> mode:DIO, clock div:1
23:54:52.013 -> load:0x3fff0018,len:4
23:54:52.013 -> load:0x3fff001c,len:1216
23:54:52.013 -> ho 0 tail 12 room 4
23:54:52.013 -> load:0x40078000,len:9720
23:54:52.013 -> ho 0 tail 12 room 4
23:54:52.013 -> load:0x40080400,len:6352
23:54:52.013 -> entry 0x400806b8
23:54:52.283 -> Card Mount Failed
23:54:52.283 -> Font file Final-Frontier-28 not found!
23:54:52.761 -> 480

Any help would be greatly appreciated.
Once again, thank you for this great library. I'm also looking forward to using your fonts.

Bodmer, 2020-01-06:
First you need to make sure the SD card interface is working. Run the SD library examples with the TFT_CS line from the display connected to 3.3V (to disable it). Check the card can be read.

Next try the sketch again with the SPI clock rate lower, say 20MHz.

You could put the font in SPIFFS, it is much easier.

stephensaid, 2020-01-13:
Thanks, Bodmer. Took your advice and used SPIFFS.
Much better.

closed 2020-01-14 by Bodmer