Bodmer/TFT_eSPI #515: White screen on ESP32 with RPi 3.2 ( ili9341) Touch Display

baleksey, 2020-01-09:
I've managed to run TFT_eSPI library on my setup but noticed strange behavior: 
If I draw anything very fast to the display it turns white very fast. For example, in [Pie Chart](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/320%20x%20240/TFT_Pie_Chart/TFT_Pie_Chart.ino) example if I add "delay(5);" (more then 5!) in the fillSegment() at he end of for loop it shows animation for levers seconds then turn white, but without that delay - it never draws anything on screen and show white from the beginning.

When I connect MISO and MOSI together as mentioned in[ this issue](https://github.com/Bodmer/TFT_eSPI/issues/310#issuecomment-464908466) it can work without "delay" pretty long but will turn white sooner or later as well.

It looks like I have very similar situation to [this issue](https://github.com/Bodmer/TFT_eSPI/issues/452) but I don't think it's purely hardware problem as I can "fix" it by connection MISO and MOSI pins from previous issue. Is there any change to have a fix for that?

For reference:

- [My esp32](https://docs.zerynth.com/latest/official/board.zerynth.doit_esp32/docs/index.html)
- [RPi 3.2 Display](http://www.lcdwiki.com/3.2inch_RPi_Display)

Bodmer, 2020-01-09:
Firstly, do not connect MISO and MOSI to try and cure this problem!

I have not seen that display variant before. Are you sure it is an ILI9341 display?

TFT_Pie_Chart example already has a delay of 4000 ms so it is not clear why adding 5 ms changes the fault signature.

If you have made other modifications to the sketch then post a copy so I can see what might be happening. Do any other examples fail? Bear in mind you have to regularly call yield() or delay(n) at least every 0.5s to stop the ESP32 watchdog from timing out.

baleksey, 2020-01-09:
I've removed those 4000ms delay from PieChart to test speed etc. But it doesn't make any difference as I have the same problem almost in any example which do a lot of redraws in a one cycle. 

I've started with a simplest [example of ellipse](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/320%20x%20240/TFT_Ellipse/TFT_Ellipse.ino) and I'm still cant render it as expected, just white screen from the beginning. I thought I just have broken display, or wrong driver/setup until I commented loop() body and just draw static images in the setup - and it works fine. 

I'm pretty sure it's ILI9341. Anyway I've tested another drives from your list and only ILI9341 works for me. 
The problem also doesn't relate to SPI frequency.

In my own example I just draw very fast animated (along X axes) square and int works perfect, but If I add to the same cycle, lets say circle and another square redraw - it fails to white pretty quick.

Disconnection of MISO and MOSI doesn't help, anyway I need touch capability which works good in Keypad_240x320 (but after X touches, it slowly whitens all).

Generally it looks like some memory leak or overflow when display just cant handle many redraws in once cycle.

By the way, almost all sprites examples I tested - works perfect. Whitening happens only with drawing primitives. And it happens to ALL examples I tested. Some whitens instantly and others (like keypad) after some insignificant time.

Bodmer, 2020-01-09:
OK, from your description this could be a power supply problem - intensive graphics demands a lot of power or a timing problem introduced by what looks like a buffer chip. on the the back of the display.

How are you powering the display. I hope you have NOT connected 3.3V AND 5V ... otherwise one or more regulators may start to overheat and shut down.  Should be wired[ like this ](https://github.com/Bodmer/TFT_eSPI/blob/master/Tools/RPi_TFT_Connections.png)as the 3.3V from the processor board will struggle to provide the power.

If power is OK that leaves timing and I can propose changes to test that but I am now busy for the next 24 hrs.

baleksey, 2020-01-09:
I think power is OK in this situation, it's a 5V external source. Furthermore I can't even make this display to work with 3.3V (external), it just doesn't turn on at all. (it seems like a bug or so) 

Yes it's  XPT2046 touch controller. Does it make any sense to you now? :)

Bodmer, 2020-01-10:
OK. I suspect data corruption. Wires to the display need to be short (<100mm). If you are using thin wires then a 3rd or 4th GND connection may help. Typically this is provided by a ground plane on a PCB.

Another option is to force the slower DC and CS timing by defining BOTH display types like this in your setup file:
```
#define RPI_ILI9486_DRIVER
#define ILI9341_DRIVER

This will invoke slower timing (e.g. [for CS here](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.h#L217-L218)) but load the ILI9341 driver.

Bodmer, 2020-01-10:
Had a look at the display manual, it does say ILI9341.

The suggests you must connect 5V AND 3.3V to the display. A Raspberry Pi would do this so perhaps that is what is needed. This is not something I have seen before (usually it is OR) but it is possible that is what is needed as there are so many display variants and why else would the data sheet say 3.3V power INPUT if it was not needed.

The display has a regulator onboard but I suspect this does not power the display controller and the SPI signals are providing the power.... when the SPI signals are mostly low (CS low and sending black (black is all logic 0)) the display chip will not get enough power to stay alive and the configuration will be lost. Perhaps adding a delay of 5ms allows the CS data pin to bring up the power again. Perhaps you have a duff/damaged 3,3V regulator on your board. Do you have a multimeter to check the display regulator output voltage (5V input, 3.3V output)?

Failing that then I have run out of ideas. Some displays do have Chinese ILI9341 emulating clone chips but they tend to work OK, so I doubt that is the issue.

baleksey, 2020-01-10:
Ok.. I've accidentally fixed the problem and feel very stupid but happy. So, I just connected ESP32 ground with my external source ground and everything works perfect now. I don't know why it was so important but maybe it should be noted somewhere to those who have the same problem. 

Bodmer thank you so much for your help and time (and for excellent library!), and sorry for such a silly mistake.

Bodmer, 2020-01-11:
Thanks for letting me know, good luck with your project.

closed 2020-01-11 by Bodmer