Bodmer/TFT_eSPI #517: Color map to reduce memory usage for sprites (implemented)

kamorris, 2020-01-12:
I've extended the sprite class to add support for 4 bits per pixel sprites, where the colors map into a 16 bit color map (with up to 16 entries).  There are three benefits: 

- it reduces the memory required by sprites by 50% over 8 bit per pixel sprites, 
- it allows more fine-grained colors (if I want 10 different greens, I can do that), and 
- it allows you to quickly change the appearance of a sprite by changing the color map and pushing the sprite again.  

The changes are not extensive, but I see you are refactoring and are not accepting pull requests right now.  I'll be happy to send a pull request when you are ready (my changes are based off 1.4.20, and I see the code is now at TFT_ESPI_VERSION 1.5.0, so I have some refactoring to do).

I have a working Example using it (of course).  Pixels, lines, circles, fonts all work.

  ```
  spr.setColorDepth(4);
  spr.createSprite(WIDTH, HEIGHT);
  spr.fillSprite(5);     // entry into color map; with the map below this will be TFT_PURPLE
  
  uint16_t cmap[16];
  cmap[0] = TFT_BLACK;
  cmap[1] = TFT_NAVY;
  cmap[2] = TFT_DARKGREEN;
  cmap[3] = TFT_DARKCYAN;
  cmap[4] = TFT_MAROON;
  cmap[5] = TFT_PURPLE;
  cmap[6] = TFT_OLIVE;
  cmap[7] = TFT_LIGHTGREY;
  cmap[8] = TFT_DARKGREY;
  cmap[9] = TFT_BLUE;
  cmap[10] = TFT_GREEN;
  cmap[11] = TFT_CYAN;
  cmap[12] = TFT_RED;
  cmap[13] = TFT_MAGENTA;
  cmap[14] = TFT_YELLOW;
  cmap[15] = TFT_WHITE;

  spr.setColorMap(cmap, 16);

  spr.pushSprite(x, y);

```
I needed to add one new method to the main TFT_eSPI class:
```
/***************************************************************************************
** Function name:           pushImage - for sprites using a 4 bit colour map
** Description:             plot sprite with color map.
***************************************************************************************/
void  TFT_eSPI::pushImage(int32_t x0, int32_t y0, int32_t w, int32_t h, uint8_t  *data, uint16_t *cmap)
```

Bodmer, 2020-01-12:
Yes, I think this addition would be good. The Sprite class has not changed much apart from the rotate function so the changes you have made should port over easily.

Two functions will be needed in TFT_eSPI.cpp, the second handling a tranparent colour.

[These are the existing 1+8bpp functions](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.h#L407-L408). So 4 bit support can be added to those. 8 bit colours can also be paletted (BMP format with colour table). 1bpp is already effectively paletted. The colour map pointer can be added:
```
  void     pushImage(int32_t x0, int32_t y0, int32_t w, int32_t h, uint8_t  *data, bool bpp8 = true, uint16_t *cmap);
  void     pushImage(int32_t x0, int32_t y0, int32_t w, int32_t h, uint8_t  *data, uint8_t  transparent, bool bpp8 = true, uint16_t *cmap);
```
If bpp = false and cmap!=nullptr then 4bpp can be assumed.

If you fork TFT_eSPI and make the changes then I will test and check them.

kamorris, 2020-01-13:
Thanks for the suggestions.  I am doing some tidying up and will send a pull request when ready.

Bodmer, 2020-01-13:
OK, I will close this and wait for the pull request.

closed 2020-01-13 by Bodmer