Bodmer/TFT_eSPI #520: ESP32 with ILI9341 White screen

stephensaid, 2020-01-13:
Hi, 

This screen was working but not anymore. I have everything wired up as in the image below and the following is my setup.

[code]
TFT_eSPI ver = 1.4.21
Processor    = ESP32
Frequency    = 240 MHz
Transactions = Yes 
Interface    = SPI 
Display driver = 9341
Display width  = 240 
Display height = 320 

TFT_CS   = D5 (GPIO 5)
TFT_DC   = D21 (GPIO 21)
Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 27.0 MHz 
[/code]

[image]

Cannot figure out the problem. I tried different SPI freq (20, 27 and 40), different CS pins, CS to ground, 3.3V and 5V.

I am running the example sketch for 320x240 - TFT_graphictest_one_lib which worked in earlier on but not anymore!

I've also tried using Board Setup in my Arduino IDE:  NodeMU-32S, DOIT ESP32 DevKit V1 and  ESP32 Dev Module.

Thanks in advance.

Bodmer, 2020-01-14:
What has changed? Have you loaded a new version of the library?

stephensaid, 2020-01-14:
Nope. Library is the same. I can try re downloading it and starting from scratch.

Bodmer, 2020-01-14:
Start with the pin [default allocations in the Setup.h file](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L167-L172). It is best to start with the TFT Reset pin being driven by a hardware pin as the guarantees the TFT gets reset during initialisation.

Apart from that, check your wiring to make sure pin connections have not got muddled up.

Post a picture of your wiring. Beware the some breadboards have a break in the blue and red power bus lines that catches people out.

stephensaid, 2020-01-15:
Removed old library and reinstalled v 1.4.20 

This is my setup file after removing all commented lines (for readability purposes and to reproduce here)
```
#define ILI9341_DRIVER

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   15  
#define TFT_DC    2  
#define TFT_RST   4  

#define LOAD_GLCD   
#define LOAD_FONT2  
#define LOAD_FONT4  
#define LOAD_FONT6  
#define LOAD_FONT7  
#define LOAD_FONT8  
#define LOAD_GFXFF  

#define SMOOTH_FONT

#define SPI_FREQUENCY  27000000 

#define SPI_READ_FREQUENCY  20000000

#define SPI_TOUCH_FREQUENCY  2500000
```
The following is my wiring. The breadboard is fine. I also checked continuation from TFT pins to ESP32 connectors.

[image]

[image]

This is my TFT screen: http://www.lcdwiki.com/3.2inch_SPI_Module_ILI9341_SKU:MSP3218

This is my ESP32 board: https://circuits4you.com/2018/12/31/esp32-devkit-esp32-wroom-gpio-pinout/

Pin 8 (LED) is wired to 3.3V directly.

Should D2 be High all the time? The onboard LED remains on even after Serial outputs Done! although the loop is still going on?

Bodmer, 2020-01-16:
The wiring looks right. D2 should be hign mist of the time. It goes low briefly to send a command then is set high again.
Rune the [Read_User_Setup](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Test%20and%20diagnostics/Read_User_Setup) diagnostic sketch and check the correct settings are being picked up by the compiler.

Petros144, 2020-01-16:
Sometimes the Duplont cables break, check, if possible replace all cables just to be shure.
I had this problem and it the past. Drives you nuts!

stephensaid, 2020-01-16:
@Bodmer, thanks for taking the time to look into this. Settings read by the Read_User_Setup below and they seem to be correct!

[code]
TFT_eSPI ver = 1.4.20
Processor    = ESP32
Frequency    = 240 MHz
Transactions = Yes 
Interface    = SPI 
Display driver = 9341
Display width  = 240 
Display height = 320 

MOSI    = D23 (GPIO 23)
MISO    = D19 (GPIO 19)
SCK     = D18 (GPIO 18)
TFT_CS   = D15 (GPIO 15)
TFT_DC   = D2 (GPIO 2)
TFT_RST  = D4 (GPIO 4)

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 27.0 MHz 
[/code]

@Petros144 , I have checked the wiring for continuity and all seem solid. Thanks for the input.

Petros144, 2020-01-16:
Can you make us a Photo of the back of the TFT? there is a Solder bridge J1 sometimes. If you brige this you can run vcc with 3,3 V straight. maybe this is the problem.

Bodmer, 2020-01-17:
As Petros144 has indicated, these screens typically are fitted with a regulator that accepts a 5V input and produces 3.3V. If a 3.3V input is provided the output will be less than 3.3V due to losses in the regulator, however this is normally sufficient.  A link on the board  allows the regulator to be bypassed. Given that you had a working system and with the checks uou have made, I suspect a damaged screen. This damage may have occured while expanding your setup, by for example plugging in the screen into the breadboard shifted by 1 hole which drives 3.3V into the GND pin and reverse biases the TFT. Of course this is just speculation.
I would replace the screen, wires and possibly the ESP32 as well because I know the setup should work.
Unfortunately I have now run out of ideas to help you further.

stephensaid, 2020-01-17:
Thanks for all your replies. I really appreciate and understand.
I have a new TFT ordered since my first post, just in case!

Here are the photos of the TFT from behind.

[image]
[image]

TonyKenny666, 2020-01-18:
HI, I'm having a similar issue with a ESP8266 and waveshare 4inch touch TFT.  it did work last year and I have a short video of it. However, now it's not working. I thought i had the same code, but I've tried settings just in case.
If I connect it to the Mega2560 with some code i wrote last year, it also works, but, this library and the example code just gives me a white screen. Tried all the above hints.
I've tried powering from a separate 5v supply instead of just the USB and I get the same result.

Bodmer, 2020-01-18:
@Petros144 Your display has a 5V to 3.3V regulator so you could try connecting the TFT Vcc input to VIN (5V) on you ESP32 board. However typically these display do run, albeit dimly, from 3.3V on Vcc.

As I cannot help further and cannot replicate the issue I will close.

@TonyKenny666: Raise a new issue as you are not using an ESP32 + ILI9341 display.

closed 2020-01-18 by Bodmer

Petros144, 2020-01-19:
Brige J1 and it should work but only 3,3V VCC imputs