Bodmer/TFT_eSPI #523: Compiling for teensy 4.0 causes build errors

TonyKenny666, 2020-01-20:
Compiling the sample code causes build errors under platform IO. I've tried this on mac and Ubuntu 18.  Trying the waveshare demo code and that compiles fine for the teensy which is why I'm wondering if it's a combination of the ttensy 4.0 platform and the TFT library.
If I switch back to another platform it compiles.

My platformio.ini file contents:

[env:teensy40]
platform = teensy
board = teensy40
framework = arduino
lib_deps = TFT_eSPI
build_flags =
 -Idir /Users/myname/.platformio/lib/TFT_eSPI_ID1559
 -Ldir /Users/myname.platformio/lib/TFT_eSPI_ID1559
upload_protocol = teensy-gui

I had to add the build flags because without those it was trying to add that path to the current project path instead of using it as an absolute. Again, that didn't happen until i tried to compile for teensy 4.0

Bodmer, 2020-01-20:
Unfortunately I do not use PlatformIO or have access to a Teensy so I am only going to be able to help if you quote TFT_eSPI library specific compiler error messages.

The library is targetted at ESP8266, ESP32 and STM processors. Although there is generic processor support there will be some sketches that will not work, in particular any sketches that use SPIIFS.

The Teensy seems to be well supported with optimised libraries so you are likely to get better performance out of those.

TonyKenny666, 2020-01-20:
thanks for the swift reply, all understood. Let's close this issue unless I get a library specific error message.

closed 2020-01-20 by TonyKenny666