Bodmer/TFT_eSPI #524: Support for vertical wrap around

Derek-K, 2020-01-23:
Feature request related to "Vertical Scrolling Definition" (VSCRDEF) & "Vertical Scroll Start Address of RAM" (VSCSAD) which is a built-in hardware scrolling feature (e.g. ST7789)

The ability for many draw functions (mainly drawChar) when the middle of the character reaches the bottom edge of the drawing area, a flag/function call, will let the internal draw routine to continue from the top (or a specific row) of the drawing area.

For example, a character is 20 pixel tall, after drawing 8 rows of pixel, it reaches the bottom of the drawing area. If a flag is set, the internal routine will continue to draw the rest of the 12 rows of pixel from the top. (or start from a specific row, useful when TFA - Top Fixed Area is not 0)

Perhaps a call to define a boundary for drawing to wrap around (on all 4 sides) would work too.

Bodmer, 2020-01-24:
I understand the problem, essentially there is a coordinate discontinuity that moves as the scroll area is moved. To resolve this would be more complex than it first appears as the library would have to work out the split point by reading the TFT registers and then draw the character twice.

Given the intrusive nature of the required code changes to the library and an anticipated low usage of the feature for the majority of library users I think it is better to manage this in a users sketch (using for example a Sprite to simplify character splitting) or as a new capability to the extension library TFT_eFEX.

closed 2020-01-24 by Bodmer

Derek-K, 2020-01-24:
Any suggestion on how to just print the "bottom half" of a string/characters? I can't set cursor position x<0...

I can think of using a sprite, then do a vertical scroll. But that's the original problem I have, not enough memory, hence trying to use the memory on LCD controller.

Bodmer, 2020-01-24:
There is no simple way to print the bottom half of characters other than via a Sprite. In the [TFT_Terminal](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/320%20x%20240/TFT_Terminal) example I avoided the need to do this by ensuring the scroll window height was an integral number of character lines and then ensuring no character straddled the coordinate discontinuity. Maybe you could try the same approach.

Bodmer, 2020-01-24:
This is an example with a Sprite. It uses a 1 bit per pixel (2 colour) Sprite so in this case the Sprite uses 780 bytes:
```
#include <TFT_eSPI.h>

TFT_eSPI tft = TFT_eSPI();

TFT_eSprite spr = TFT_eSprite(&tft); // Sprite object spr

//==========================================================================================
void setup() {
  Serial.begin(115200);

  tft.init();
  tft.fillScreen(TFT_BLACK);

  // Create a sprite for the screen
  spr.setColorDepth(1); // 1 bit per pixel
  spr.createSprite(240, tft.fontHeight(4)); // 240 x 26 = 780 bytes
  spr.setBitmapColor(TFT_GREEN, TFT_BLACK); // Colors for 1bpp Sprite foreground/background
  spr.setTextColor(TFT_WHITE);              // Any colour but black for text in 1bpp Sprite
}

//==========================================================================================
void loop() {
  // Top half
  spr.fillSprite(TFT_BLACK);
  spr.drawString("Split text test 12345", 0, tft.fontHeight(4) / 2, 4);
  spr.pushSprite(0, 40);

  // Bottom half
  spr.fillSprite(TFT_BLACK);
  spr.drawString("Split text test 12345", 0, -tft.fontHeight(4) / 2, 4);
  spr.pushSprite(0, 40 + tft.fontHeight(4) + 8); // Gap = 8 pixels

  while (1) yield();
}
//==========================================================================================
```

Derek-K, 2020-01-25:
I will take a look later this weekend. Does it work for smooth fonts? (i.e. support .vlw fonts file?)

Bodmer, 2020-01-25:
Smooth fonts could use the same technique but you would need to increase the Sprite color depth to 8 bits at least to get the anit-aliasing to work.

Derek-K, 2020-01-25:
> I will take a look later this weekend. Does it work for smooth fonts? (i.e. support .vlw fonts file?)

Thanks, it works. Didn't know Sprite support drawString with y<0. And for smooth font, I just use spr.setColorDepth(16) as per one of the examples.

With a bit of tweak and optimization, I only need to define a sprite to hold the 'bottom half' when needed (i.e. crossing the bottom boundary)