Bodmer/TFT_eSPI #525: Minor fix in BMP_functions.ino

Derek-K, 2020-01-23:
Suggest to save the SwapBytes state and restore it upon exit in example code.

When used with Sprite class, it causes incorrect font color.

```
void drawBmp(const char *filename, int16_t x, int16_t y) {
  bool SwapByteOrig=tft.getSwapBytes();
```
.
.
.
```
  bmpFS.close();
  tft.setSwapBytes(SwapByteOrig);
}
```

Bodmer, 2020-01-24:
This is already been incorporated into sketches [such as this one](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Generic/TFT_SPIFFS_BMP/BMP_functions.ino#L37).  I guess I missed this in one example, exactly which example are you referring to?

Derek-K, 2020-01-24:
Oh, my bad. I was using the sample code from 1.4.16 library... :sweat_smile:

BTW, I modified that function a bit so it works with #524 (at least for loading bitmaps from SPIFFS)  :grin:

closed 2020-01-24 by Derek-K