Bodmer/TFT_eSPI #530: An LCD with ILI9488 driver is not working

HumamHelfawi, 2020-01-25:
Hi Bodmer,
Thanks again for the great work and the great library!

I am using the library with ESP32, this is my setup file:
```
#define ILI9488_DRIVER
#define ESP32_PARALLEL
#define TFT_CS   -1  // Chip select control pin (library pulls permanently low
#define TFT_DC   26 // Expander // Data Command control pin - use a pin in the range 0-31
#define TFT_RST  -1  // Reset pin, toggles on startup
#define TFT_WR   4  // Expander // Write strobe control pin - use a pin in the range 0-31
#define TFT_RD    -1  // Read strobe control pin  - use a pin in the range 0-31
#define TFT_D0   14  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   12  // so a single register write sets/clears all bits.
#define TFT_D2    13  // Pins can be randomly assigned, this does not affect
#define TFT_D3   15  // TFT screen update performance.
#define TFT_D4   2
#define TFT_D5   0
#define TFT_D6   21
#define TFT_D7   22
```

P.S. TFT_CS is wired directly to GND, TFT_RST is wired directly to ESP32_EN

The library works well with an parallel 16-pins ILI9486 (I double link the DB[8-15] to DB[0-7]). Now I bring an 8-pins ILI9488 LCD to test on it. However, it does not work. I just got white screen. I double checked the wiring, power and tried different screen copies to make sure that it is not broken with no luck.

Do you have any suggestion about what the problem could be? Any hint that may help in debugging the problem?

Thank you very much for your time and effort!

Bodmer, 2020-01-25:
I would use the setup file "Setup16_ILI9488_Parallel.h" to start with as that is working fine with my display.

Connecting TFT_CS to GND is fine but then the TFT controller is susceptible to garbage commands at startup so it is best to always use a hardwired reset pin from the ESP32.

If you do not connect the TFT_RD pin to the ESP32 then you must tie this line high to ensure the TFT does not drive the bus.
 I see the word "expander" in your description, this implies you have additional circuitry in the signal line. if this is indeed the case then it may upset the signal timing and data written to the TFT will be then be corrupted.

HumamHelfawi, 2020-01-25:
Thank you very much for your reply!

Regarding exapnder, it is an old thing and I forget to delete the comment. No expander is involved here.

I will try your suggestion.

Thanks again!

Bodmer, 2020-02-06:
Post back if you still have the issue and I will open again.

closed 2020-02-06 by Bodmer

HumamHelfawi, 2020-03-03:
Sorry for late reply. I though it may be hardware problem and spent time debugging the circuit with no luck.

I noticed sth weird regarding the TFT_CS on my old working screen and I suspect it is the reason that the new screen is not working. When connecting TFT_CS to GND directly, it works. If I connect TFT_CS to one of the GPIOs (27 exactly) and edit user_setup accordingly, it is not wokring!
I tried even to call:
```
  pinMode(27, OUTPUT);
  digitalWrite(27, LOW);
```
Before anything in the setup() function with no luck. The same circuit and same firmware won't work if TFT_CS is attached to an GPIO but will work if I connect it directly to GND! Any though about this behavior? 

P.S. The new (problematic) screen  CS pin is soldered to one of ESP32 GPIOs and can pull it now and test with GND connection to ensure that it is the real problem.
Please accept my deepest appreciation