Bodmer/TFT_eSPI #531: white screen on 8266

TonyKenny666, 2020-01-26:
HI again,
Trying this again on an 8266 (also tried EDP32-12E with same problems)

Hardware is the waveshare XPT 2046.  I did have it running last year must misplaced the code.
The DIP switches on the device are switched to SPI and I have this working with the waveshare demo code, but prefer Bodmer's library.

setup file contains   (also tried the 9431 driver 
#define ILI9486_DRIVER
#define TFT_MISO 12
#define TFT_MOSI 13
#define TFT_SCLK 14
#define TFT_CS   15  // Chip select control pin
#define TFT_DC    5  // Data Command control pin
#define TFT_RST   4  // Reset pin (could connect to RST pin)
#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

// Comment out the #define below to stop the SPIFFS filing system and smooth font code being loaded
// this will save ~20kbytes of FLASH
#define SMOOTH_FONT
 #define SPI_FREQUENCY  40000000
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000

I've used the code that outputs those settings to the serial monitor and confirmed that these are seen on the device. I've confirmed the pin connections over and over, they are printed on the back of the board so were easy to make sure I had the right pins.

All i see is a white screen. I've added some serial println statements so i can see that the loop()  method in looping, but nothing appearing on the screen.
Short of buying an oscilloscope and seeing whats happening on the pins, I'm a bit stuck.

Please can you advise what I might have missed?
Thanks in advance!

Bodmer, 2020-01-26:
Post a link to the vendors product page.

Is it a RPi board?

TonyKenny666, 2020-01-26:
Thank you, I managed to find the board in my amazon orders, it's a product from MakerHawk, 

https://www.makerhawk.com/products/makerhawk-2pcs-esp8266-module-esp8266-esp-12f-development-board-nodemcu-lua-wifi-internet-development-board-new-version-cp2102-chip

Bodmer, 2020-01-26:
Also I need a link to the TFT board you have as well.

TonyKenny666, 2020-01-26:
Sorry, please see this link:

https://www.waveshare.com/wiki/4inch_TFT_Touch_Shield

TonyKenny666, 2020-01-26:
Maybe pause this one for a few days, I've gone back to the waveshare library, also reverted to using the Arduinio IDE instead of platformio and I'm seeing a loop of  rst cause:4 boot mode(3,6)  So I suspect there problem is somewhere else and until I resolve that, I don't think the issue I'm seeing is in your library. I'll get back to you ASAP with an answer.

TonyKenny666, 2020-01-26:
Sorry, I'm back again, it looks like the waveshare library i was trying has a bug causing the reset. When i use your library, it runs, no reset, I see the expected output on the serial monitor. 

I'm also trying a nodeMCU labelled 'Amica' on the board.  https://iotbytes.wordpress.com/nodemcu-pinout/
Comments on this link mention that these boards operate on 3.3v but the screen requires 5v (which is obtain from Vin), so it may be that the logic levels are just not high enough as suggested with the forum post here:  https://forum.arduino.cc/index.php?topic=528368.15

Curious how I had this working a year ago with the NodeMCU amica device, seen here:  https://www.instagram.com/p/BtbIjQ8B38U

My plan now is to try pull up/down resistors to adapt the logic level and purchase an oscilloscope to confirm this. It's been 25 years since I did electronics, so a little refreshing is going to be needed :)

For these reasons, sorry again, I think this is not library related but is hardware, although still wondering how it worked last year with the same device.

TonyKenny666, 2020-01-26:
OK, got it, I apologise for the trouble, I got it working!
I managed to find old photos I took when it was working and thankfully google drive kept copies of the old files so i was able to get the config.  Turns out is IS an RPi screen and I just needed the RPi 9486 driver.
Please tell me if you accept donations towards your work on this and I'll happily send something!

closed 2020-01-26 by TonyKenny666

Bodmer, 2020-01-26:
Glad to hear it is working, no payment necessary!

TonyKenny666, 2020-01-26:
Indeed, but it's nice to show appreciation to a fellow developer and we're going to be using your library a lot for some upcoming open source projects. If you're ever in Leicester, then maybe a coffee or a beer to show our gratitude?