Bodmer/TFT_eSPI #532: Error Compiling for Board ESP32 Dev Module

TSJim, 2020-01-28:
I am using a generic ESP32 dev board with a 2.4" TFT SPI display.  This board and display work fine with the Adafruit libraries.

However, with the TFT_eSPI library, I'm getting the following compile errors:

`Arduino: 1.8.11 (Mac OS X), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/jim/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/jim/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/jim/Documents/Arduino/libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -vid-pid=0000_0000 -ide-version=10811 -build-path /var/folders/67/blc1rtx10xs0q3mzzmczns9m0000gn/T/arduino_build_966187 -warnings=all -build-cache /var/folders/67/blc1rtx10xs0q3mzzmczns9m0000gn/T/arduino_cache_15738 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=/Users/jim/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/Users/jim/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/Users/jim/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=/Users/jim/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=/Users/jim/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=/Users/jim/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1 -verbose /Users/jim/Documents/Arduino/libraries/TFT_eSPI/examples/320 x 240/TFT_graphicstest_one_lib/TFT_graphicstest_one_lib.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/jim/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/jim/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/jim/Documents/Arduino/libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -vid-pid=0000_0000 -ide-version=10811 -build-path /var/folders/67/blc1rtx10xs0q3mzzmczns9m0000gn/T/arduino_build_966187 -warnings=all -build-cache /var/folders/67/blc1rtx10xs0q3mzzmczns9m0000gn/T/arduino_cache_15738 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=/Users/jim/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/Users/jim/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/Users/jim/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=/Users/jim/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=/Users/jim/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=/Users/jim/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1 -verbose /Users/jim/Documents/Arduino/libraries/TFT_eSPI/examples/320 x 240/TFT_graphicstest_one_lib/TFT_graphicstest_one_lib.ino
Using board 'esp32' from platform in folder: /Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4
Using core 'esp32' from platform in folder: /Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4
Detecting libraries used...
/Users/jim/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++ -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/config -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/app_trace -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/app_update -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/asio -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/bootloader_support -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/bt -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/coap -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/console -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/driver -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp-tls -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp32 -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_adc_cal -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_event -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_http_client -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_http_server -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_https_ota -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_ringbuf -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/ethernet -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/expat -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/fatfs -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/freemodbus -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/freertos -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/heap -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/idf_test -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/jsmn -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/json -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/libsodium -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/log -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/lwip -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mbedtls -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mdns -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/micro-ecc -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mqtt -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/newlib -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/nghttp -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/nvs_flash -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/openssl -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/protobuf-c -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/protocomm -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/pthread -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/sdmmc -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/smartconfig_ack -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/soc -I/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/spi_flash -I/Users/jim/Library/Arduino15/ [... cut ...]
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

// #define SPI_FREQUENCY  27000000
#define SPI_FREQUENCY  40000000
// #define SPI_FREQUENCY  80000000

#define SPI_READ_FREQUENCY  20000000

#define SPI_TOUCH_FREQUENCY  2500000

// #define SUPPORT_TRANSACTIONS
`

Any idea what I'm doing wrong?

Thanks!

Bodmer, 2020-01-29:
You are not defining the pins correctly for your setup:

Use this format:
```
#define ILI9341_DRIVER

//#define TFT_INVERSION_OFF

#define TFT_MISO 19 // (leave TFT SDO disconnected if other SPI devices share MISO)
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS    15  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   4  // Reset pin (could connect to RST pin)

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

// #define SPI_FREQUENCY  20000000
#define SPI_FREQUENCY  27000000
// #define SPI_FREQUENCY  40000000
// #define SPI_FREQUENCY  80000000

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  16000000

#define SPI_TOUCH_FREQUENCY  2500000
```

TSJim, 2020-01-29:
Hi Bodmer...

I'm always amazed at how helpful you are and how quickly you reply to issues.  Thank you for that.

I swapped in the code that you provided, but I'm getting the same error.  I had actually already tried to define the pins that way.

Here is the first error that happens when I try to compile:

`In file included from /Users/jim/Documents/Arduino/libraries/TFT_eSPI/examples/320 x 240/TFT_graphicstest_one_lib/TFT_graphicstest_one_lib.ino:17:0:
/Users/jim/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.h:661:15: warning: inline function 'void TFT_eSPI::begin_tft_write()' used but never defined`

There are also similar errors for "begin_tft_read()" and "end_tft_read()".

I'm just running the stock unaltered "TFT_graphicstest_one_lib.ino" sketch and I have the latest version of the Arduino IDE and the library.  I did a clean install and got rid of all other libraries but it still won't compile.

Any other ideas why this will not compile?

Thank you!

Bodmer, 2020-01-29:
You have listed a warning, that will not stop the compile. Find the first few errors.

Bodmer, 2020-02-01:
I have weeded out the warnings sources and fixed a bug in the 4 bpp Sprite so you may get more success with the latest issue 2.0.2 version.

Bodmer, 2020-02-06:
Post back if you still have the issue and I will open again.

closed 2020-02-06 by Bodmer