Bodmer/TFT_eSPI #535: How to use with Particle Argon?

GeerGuy, 2020-01-30:
There is a version of this library available from the Particle IDE (TFT_eSPI 0.16.15), but no examples for how to set it up or use it with Particle devices.

Any tips?

Bodmer, 2020-01-30:
I have no knowledge of that adapted library or who ported it, nor am I familiar with the Particle Argon board or IDE. So I am unable to help. In principle it should only be necessary to edit the User_Setup.h file to suit your particular setup.

One option is to get an ESP8266 or ESP32 and try the examples supplied with this master library.

This master copy now supports any generic processor but it will be slow as it has performance optimisation code for the ESP32/ESP8266 and STM32 processors.

closed 2020-01-30 by Bodmer

GeerGuy, 2020-01-30:
The Particle Argon uses the nRF52840 SoC with the ARM Cortex-M4F 32-bit processor @ 64MHz, so I assume that it should be included in the STM32.  I will try it with your newest files.

Hynesman, 2022-09-29:
did you get it working?  I'm also interested in getting it working on a particle argon.