Bodmer/TFT_eSPI #536: Better fps on

tobozo, 2020-02-01:
As observed, psram can [slow down high-speed rendering](https://twitter.com/M5Stack/status/1214455521052250112?s=20)

In order to add control to this behaviour, I've been testing this for a while and it does the job:

    void TFT_eSprite::setPsram( bool enabled ) {
      _usePsram = enabled;
    }

    (....) 

    if ( psramFound() && _usePsram ) ptr8 = ( uint8_t*) ps_calloc(w * h + 1, sizeof(uint16_t));

    (.... )

It doesn't alter the initial behaviour, so it's still possible to mix non-psram and psram sprites.

Bodmer, 2020-02-01:
The video suggested to me that PSRAM is faster and I could not see any frame rate figures to judge if this is the case.

I did some tests a while ago and PSRAM was only 12% slower which is quite impressive considering it is serial accessed.  Do you have a test that gives a measured frame rate for comparison. You may find bigger performance gains can be made by making other parts of the code more efficient.

tobozo, 2020-02-01:
Maybe the 12% measure had benefits from other optimizations ?

In order to be sure, I've produced an example where *all* pixels are sent.

This sketch gives ~38fps with psram off, and ~20fps when psram is enabled, that's 48% slower.

```C
  sprite.setColorDepth( 16 );
  sprite.createSprite( 237, 237 ); // don't make it any bigger
  float fps = 0;
  int fpscount;
  unsigned long lastfps = millis();

  while( 1) {
    uint16_t randomColor = tft.color565( random( 255 ), random( 255 ), random( 255 ) );
    sprite.fillSprite( randomColor );
    sprite.setCursor( 0, 0 );
    sprite.printf("%.2f", fps);
    sprite.pushSprite( 0,0 );
    fpscount++;
    unsigned long fpsdelay = millis() - lastfps;
    if( fpsdelay >= 1000 ) {
      fps = float(fpscount*1000)/float(fpsdelay);
      lastfps = millis();
      fpscount = 0;
    }
  }

```

Bodmer, 2020-02-01:
PSRAM data is cached in a 16kbyte buffer, this means that 16 bit Sprites larger than 90x90  in total area require the buffer to be flushed.  This causes the effects you observe. For example this is the frame rate for different 16 bit square Sprites. You can see that above 128x128 pixels the frame rate drops significantly lower than SRAM only Sprites:
[image]

The nice thing about this cache is that it is in an area of RAM not normally accessible to sketches so it effectively acts as a virtual SRAM extension when handling the smaller Sprites and performance impact is then is maybe 3%.

Clearly 8 bit Sprites can be up to 128x128 in area, 4 bit Sprites can be 256x256 in area and 1 bit Sprites can be 362z362 in area before the cache must be flushed for the next block.

In general is is best to use PSRAM for large Sprites during development until you know what workspace is needed and how much can be allocated to Sprites before the processor starts crashing.

Rather than  use large Sprites for image handling in web/WiFi applications it is usually best to create a smaller buffer and update the screen in blocks rather than try to cache images in PSRAM.

I think the need for a PSRAM enable switch will be rare but it is very easy to add so I will put it on my TODO list.

Bodmer, 2020-02-06:
Library updated.  To disable/enable use of PSRAM you can now use for example:

sprite.setAttribute(PSRAM_ENABLE, false); // true to enable

Where "sprite" is the Sprite class instance.

closed 2020-02-06 by Bodmer