Bodmer/TFT_eSPI #537: include order in TFT_eSPI.h

joerglemke, 2020-02-01:
Adaption of display type is meant to be done by editing within the library code.
I find this problematic when working with different boards at the same time. Also I have to do this every time the library is updated.
Furthermore when the board comes with an integrated display, it would be nice to have the display configured by just choosing the right board.

So, for PlatformIO there is the suggestions given in the tools folder, you should provide the settings through build time compiler switches. I even succeeded to do likewise for the Arduino IDE (although personally I would stick to PlatformIO). However, the maintainers of the Arduino board specs do not like the idea of supplying pin definitions on on the command line. They say, these should be supplied in the board specific `pins_arduino.h` .

I also succeeded to create a `pins_arduino.h` for my board (TTGO T-Display). There is only a little problem left.

The `User_Setup_Select.h` is included from `TFT_eSPI.h` in the first place. So at that time `pins_arduino.h` has not been included yet. `Arduino.h` is included just after. Swapping these would help. I verified that this is possible, so that my `pins_arduino.h` can provide the definitions.

**So, why isn't `Arduino.h` included BEFORE `User_Setup_Select.h` ?  Could this be changed?**

Like this maybe:

```
#ifndef _TFT_eSPIH_
#define _TFT_eSPIH_

#define TFT_ESPI_VERSION "2.0.2"

#include <Arduino.h>    // include first to include pins_arduino.h <--------------------

/***************************************************************************************
**                         Section 1: Load required header files
***************************************************************************************/
// Include header file that defines the fonts loaded, the TFT drivers
// available and the pins to be used, etc, etc
#include <User_Setup_Select.h>

//Standard support
// #include <Arduino.h>    // DO NOT include again <--------------------
#include <Print.h>
#include <SPI.h>
```

Bodmer, 2020-02-02:
The Arduino.h could be included before User_Setup_Select.h, because the order at this point is arbitrary. However it appears that this is only needed if a user edits the board core files (which will also get over-written on the next update) and this is beyond the ability of most users.

The user setup files do not need to be in the library folder, see ReadMe "Tips" section.

Unfortunately there is no easy solution. I find I can add and remove // from a file quickly to swap between setups but I know this is not ideal.

One solution I though of was to write a helper app that scans your setup folder, lists them in a menu and modifies the library setup select file to point a selected file. But I figured I would spend too much time writing that app than I would save.  When the library was originally created there were no ESPxxx+display boards available and all setups became a configuration file, of course the landscape has changed and now there is a plethora of boards with built in displays...

joerglemke, 2020-02-02:
Thanks for commenting.

You say: "_it appears that this is only needed if a user edits the board core files (which will also get over-written on the next update) and this is beyond the ability of most users_"

I do not see it as a user's task. It should be provided by the board supplier. So a user would simply choose his board type and everything would be readily configured.

At this time I see myself playing role of a board supplier, because presently the concerning board does not have its specific variant yet.

I attempted a different approach though:
I put a file `User_Setup.h` (as required by TFT_eSPI) into the boards `variant` folder.
Then under Arduino IDE it works out-of-the-box, because the `variant` folder is on the include path before the TFT_eSPI library folder.
Under PlatformIO however the include order is the other way, so the default `User_Setup.h` from the library is seen first. So I just deleted this file.

What do you think of this idea:
Do not deliver the `User_Setup.h` with the library. You could deliver it maybe under a different name like `User_Setup.h.template`. So the user would additionally simply rename it while editing to his needs.
In case the board variant delivers a `User_Setup.h` the library would be ready to go.
In any other case the user would have to edit the library after every update anyways. This way he would be forced to to at least something or compilation will fail. Currently after the update, the library would pick some invalid pin definitions and do strange things with the board. That would be a forced reminder to the user, that he needs to do something.

Bodmer, 2020-02-02:
OK, I understand the problem. As a board supplier it would be worth creating a new board variant. The way other suppliers managed the issue if to take a renamed copy of TFT_eSPI and build it into their own version of the library. (M5Stack and Seeed have followed this route). This makes tracking updates to the TFT_eSPI library more difficult and so compatibility has diverged.

I'll move Arduino.h if you wish but User_Setup.h has to stay until I find a better way forward.

I can add another #ifndef ([like this one which the PlatformIO users requested](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L17)) if it helps you.

Espressiff build in TFT pins into their board package to support their own graphics library but eventually removed them after users objected as it removed flexibility and caused issues with other graphics libraries.

I think a more universal approach is needed but that needs a level of input from the Arduino team to set the board<>library interface specifications and I doubt that is a path they want to follow. The problem as I see it is that the term "board" originally meant a basic processor so only pins had to be defined. Now the "board" package is being used for more sophisticated system with interfaces and hardwired functionality far more complex that basic digital and analog IO.

Unfortunately I do not have a solution for you but I will make simple changes if it helps you.

joerglemke, 2020-02-02:
The approach of supplying those defines in pins_arduino.h already makes use of USER_SETUP_LOADED. So no need to add another #ifdef

For the remainder of the discussion I think you see similar issues as I do. I only recently started with the Arduino PlatformIO stuff. But I already see, that the stack misses a few layers of concern. When one creates a project, he puts several things together, like a build platform (Arduino IDE or PlatformIO), firmware platform (Arduino or EPS-IDF), a pre-manufactured base board with MPU and optionally other on-board components (like display, radios, input devices), project specific add-on hardware (which again could be off-the-shelf), libraries supporting the hardware components (eg. there might be several libraries supporting the same hardware, so one could choose), libraries for algorithms, and last but not least the business code that should have some kind of abstraction from the concrete hardware in order to be able to migrate or produce different models. Everything must be configurable in some way but presently you can only provide a "board" definition, where Arduino people only see the MPU, and a set of libraries, where you have to configure hardware adaption by modifying (editing  or choosing a different version) the library. 

As you say USER_SETUP_LOADED was requested by PlatformIO users. They would put command line switches into the project specific build recipe. Only half way done, because you need to copy this from project to project. And on the Arduino IDE there seems to be no way to do this at all. Wouldn't it be nice to have it in a common place? It is possible to put these compiler switches into the board definition files, even with the Arduino IDE. But as I said the maintainers of arduino-esp32 rejected this. So I am not yet sure about the opinions of the PlatformIO maintainers.

That said, I fully understand your position as a provider of a general library supporting many different hardware models and configurations.

Coming back to the problem at hand. If you could move this include of Arduino.h up before User_Setup_Select.h (and thereby most importantly before this #ifndef USER_SETUP_LOADED) you would open up an additional way for providing the TFT pin definitions, not only via the compiler switches but also through pins_arduino.h. It shouldn't break backwards compatibilty. Actually, in the user's code this include could be done explicitly and with Arduino IDE is done automatically anyways for the sketch. But for TFT_eSPI.cpp currently this is not the case, so the library code would see different definitions as the sketch code.

Yes I know, that by putting the TFT definitions into the board variant, one would only supply configuration for this particular TFT_eSPI library. If one would want to use a different display library with the same board, it wouldn't help. But hey, let's say, TFT_eSPI is the de-facto standard ...

Thanks for your patience.

Bodmer, 2020-02-02:
OK, done.

Since sketch #define's are not seen by the library the only option for sketch configuration is to pass the required configuration, perhaps via a struct, to the constructor.  I am not keen on that approach because at the moment once the setup file is correct then I can run any example without modification.

Closing but I will be monitoring postings.  Good luck with your project.

closed 2020-02-02 by Bodmer