Bodmer/TFT_eSPI #539: 3.5" TFT LCD ILI9486 not working

DJ534, 2020-02-05:
Hi, I am trying to get 3.5" TFT LCD shield for arduino UNO working on ESP32. It has 8 bit interface labeled D0 - D7, resolution 320 x 480 and driver ILI9486. Unfortunately, it does not work. All I see is a white screen. I have edited the User_Setup.h as follows:

`
#define ILI9486_DRIVER
#define ESP32_PARALLEL
#define TFT_CS   33
#define TFT_DC   15
#define TFT_RST  32
#define TFT_WR    4
#define TFT_RD    2
#define TFT_D0   12
#define TFT_D1   13
#define TFT_D2   26
#define TFT_D3   25
#define TFT_D4   17
#define TFT_D5   16
#define TFT_D6   27
#define TFT_D7   14

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF
`

I have connected the above mentioned pins on the shield to corresponding GPIO pins on ESP32. The TFT shield has pin labeled with 5V. I realized, that it is necessary to connect it on 5V otherwise, the backlit does not work. Since ESP 32 does not provide 5V output, I use arduino uno's 5V pin as a 5V source. I have connected the GND's of UNO and ESP32 to make sure, that the grounds are on the same potential. I am powering the UNO and ESP 32 from PC by USB cables.

Can anybody give me a suggestion how to get the display working on ESP 32? I have tested it successfully on arduino UNO with MCUFRIEND library.

Bodmer, 2020-02-05:
I assume you are using an ESP32 UNO style board so it should supply 5V (mine does).

Run the Read_User_Setup diagnostic sketch to check the right pins and settings are being picker up by the compiler.

I don't have an 8 bit parallel ILI9486 board so am unable to test. It could be that the write strobe may be too fast for that chip driver and that it needs to be slowed down. If you change both [this line](https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_ESP32.h#L63) and also [this line](https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_ESP32.h#L105) to:
```
#if defined (XXXXTFT_PARALLEL_8_BIT)
```
The library will slow down the DC + CS strobes and that *may* help. If that does not work then unfortunately you will have to stick with a library that you find does work.

DJ534, 2020-02-06:
Hi Bodmer,
thank you for your reply.

1. I am using Arduino UNO with atmega328 for powering the backlit of the TFT (see attached picture).

[image]

2. I ran the diagnostics twice - first with latest version of TFT_eSPI , which is available through library manager on Arduino IDE - TFT_eSPI ver = 1.4.20. And again with latest version available TFT_eSPI ver = 2.0.2, downloaded from github. The outputs are slightly different, see the attached file.  

[bodmer_reply.txt](https://github.com/Bodmer/TFT_eSPI/files/4164256/bodmer_reply.txt)

3. I have changed the two lines in the TFT_eSPI_ESP32.h, but the result is still the same. If I run the Colour_Test, the colours are changing correctly, but the text is not displayed. Instead, few green and red dots appear on left side of the screen. The Demo_3D_cube makes the display flicker with and few pixels changing.

Meanwhile, I have successfully tested the display with ESP32 and MCUFRIEND library. I have checked the display's driver through readID and it is really 9486. If you were interested in the  output of the LCD_read_reg sketch, I attach it below.

[LCD_ID_readreg.txt](https://github.com/Bodmer/TFT_eSPI/files/4164344/LCD_ID_readreg.txt)

If there were any way how to get TFT_eSPI library working, I would really appreciate it. The sprite class would be helpful for my purposes.

Bodmer, 2020-02-06:
Thanks for the update. I am not sure what is happening. There must be some incompatibility in the library. There have been some drastic restructuring going on recently so it is possible I broke something.  I will have a look but the symptoms you describe and the testing indicate this is a timing issue but as far as I can see the timing should be OK.

I will investigate further as I am sure it could be made to work!

DJ534, 2020-02-09:
Thank you for the effort to help me Bodmer!

Bodmer, 2020-02-09:
Try changing [this line](https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_ESP32.h#L147) to:
```
  #define WR_H GPIO.out_w1tc = (1 << TFT_WR); GPIO.out_w1tc = (1 << TFT_WR); GPIO.out_w1ts = (1 << TFT_WR)
```
This will slow down the write strobe by adding two low periods to allow more data setup time. 
If that works then you may find only 1 extra low period works:
```
  #define WR_H GPIO.out_w1tc = (1 << TFT_WR); GPIO.out_w1ts = (1 << TFT_WR)
```

DJ534, 2020-02-11:
Hi Bodmer,
I apologize for the delayed answer. I tested the code with updates, that you mentioned in your previous post. Unfortunately, the display still does not work properly. During the tests I realized, that one or a few jumper wires are loosely connected either to display or ESP32. That could cause the improper behavior of the display. I will make a PCB with soldered connections a make the tests once again. I will be back later with the results.

DJ534, 2020-02-13:
> Try changing [this line](https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_ESP32.h#L147) to:
>
> ```
>   #define WR_H GPIO.out_w1tc = (1 << TFT_WR); GPIO.out_w1tc = (1 << TFT_WR); [...]

This seems to work. But the text in the image is distorted. See result of the Colour_Test: 

[image]
[image]

Bodmer, 2020-02-13:
Looks like the write strobe is still not slow enough! Try this line:
```
 #define WR_H GPIO.out_w1tc = (1 << TFT_WR); GPIO.out_w1tc = (1 << TFT_WR); GPIO.out_w1tc = (1 << TFT_WR); GPIO.out_w1tc = (1 << TFT_WR); GPIO.out_w1ts = (1 << TFT_WR); GPIO.out_w1ts = (1 << TFT_WR); GPIO.out_w1ts = (1 << TFT_WR)
```
That is 4  logic low periods and 3 logic high periods.

RmkJersey, 2020-02-14:
I have the same issue with the fonts displayed corrupted.

Bodmer, 2020-02-16:
Should be OK now.

closed 2020-02-16 by Bodmer

RmkJersey, 2020-02-16:
Thanks for the update.
 
Has this fixed the readrect and readpixel issue or the corrupt fonts?

RmkJersey, 2020-02-16:
All of the font test sketches seem to work but the colour_test demo does not run and all you see is a white screen initially and after a few seconds, the last 5th of the screen is grey.

DJ534, 2020-02-17:
> Should be OK now.

The font in Colour_Test is now displayd properly except the font in white color. In this case, the font is still corrupted. Demo_3D_cube and TFT_Ellipse work flawlessly. UTFT_Demo_480x320 displays corrupted font in red rectangle. The green one is ok. It seems that displaying font in some colors still remains a problem.

RmkJersey, 2020-02-17:
Many thanks for sorting so quickly.
 
I am heavily involved in the games industry (formerly Software Creations) and I would like to discuss with you ,offline ,creating a gaming engine with you, if you are interested?

Bodmer, 2020-02-17:
@DJ534 I reproduced the font corruption issue on one of my 8 bit parallel displays and this is now definitely fixed. I found an ILI9486 display identical to yours and it works without errors when coupled to an ESP32 UNO board.

I suspect your setup is suffering from signal integrity induced errors. The ESP32 outputs fast edges on the logic lines and this causes ringing on long wires and ground bounce when a large number of logic lines switch at the same time (e.g. for black to white colour changes). Beef up the 0V (GND) wires betwen the ESP32 and display by adding extra wires and keeping as short as possible. Shorten the data lines and control lines. Drive the setup from a good power source. The ESP32 board should be able to provide the 5V for the display, USB power pin on ESP32 may be called VIN. Ideally use an UNO style ESP32 board with the wiring modes in the README.

Unfortunately these low cost shields have no decoupling capacitors on the logic buffers or the regulator which is bad design practice but that is unlikely to be the cause of your corruption.

Checking the data sheet and tests with a 'scope show that the strobe timing is fine for an ILI9486 so that was not the problem.

Bodmer, 2020-02-17:
@RmkJersey, I have too many of my own projects!

RmkJersey, 2020-02-17:
Many thanks and I will shorten all of the lines and do as you have advised for the other connections, I will also power the screen via a separate supply, and I will let you know how I get on.

RmkJersey, 2020-02-17:
I have now shortened all of the connections as requested and have now added an external power source for the TFT screen and the font corruption has now been fixed.  I removed the TFT external power supply and the fonts remain uncorrupted, so I suspect the "ground noise" as indicated was the cause of the corruption.

I still can't get the readRect() or ReadPixel() to work so is this issue unrelated to the font issue?

DJ534, 2020-02-18:
> @DJ534 I reproduced the font corruption issue on one of my 8 bit parallel displays and this is now definitely fixed. I found an ILI9486 display identical to yours and it works without errors when coupled to an ESP32 UNO board.
> 
> I suspect your setup is suffering from signal integrity induced errors. The ESP32 outputs fast edges on the logic lines and this causes ringing on long wires and ground bounce when a large number of logic lines switch at the same time (e.g. for black to white colour changes). Beef up the 0V (GND) wires betwen the ESP32 and display by adding extra wires and keeping as short as possible. Shorten the data lines and control lines. Drive the setup from a good power source. The ESP32 board should be able to provide the 5V for the display, USB power pin on ESP32 may be called VIN. Ideally use an UNO style ESP32 board with the wiring modifications in the README.
> 
> Unfortunately these low cost shields have no decoupling capacitors on the logic buffers or the regulator which is bad design practice but that is unlikely to be the cause of your corruption.
> 
> Checking the data sheet and tests with a 'scope show that the strobe timing is fine for an ILI9486 so that was not the problem.

Thanks for the suggestions. If I have shield LCD, I will put on ESP32 UNO board next time, as you recommend. I don't know what you exactly mean with ,,beef up the GND wires betwen the ESP32 and display by adding extra wires,, can you explain it please?

Bodmer, 2020-02-18:
I mean increase the number of GND connections (wires) between the GND of the ESP32 and the GND of the TFT, so use say 3 wires for this connection to reduce the inductance (resistance is not the problem). On a PCB reducing the GND connection inductance is achieved by using a "ground plane" or wide PCB tracks. Shorter signal (logic line) connections reduce the edge ringing. The critical connection is the TFT_WR line as the rising edge is used rather than the level, so any ringing on the rising or falling edge can cause multiple writes on what should be a single write. Adding a 22 Ohm resistor at the ESP32 end of the TFT_WR connection can be used to reduce the rise time of the signal at the TFT and reduce the ringing. [See here](https://en.wikipedia.org/wiki/Ground_bounce).

DJ534, 2020-02-18:
> I mean increase the number of GND connections (wires) between the GND of the ESP32 and the GND of the TFT, so use say 3 wires for this connection to reduce the inductance (resistance is not the [...]

Thanks for explanation!

RmkJersey, 2020-02-19:
I have now added the additional wires to the ground pins and added an external power source for the TFT screen and all appears to be working well.
 
I am trying to move a sprite across the screen by reading what is under the sprite before pushing the sprite to the screen, therefore preserving what was on the screen.  For each loop I am then writing back what was originally under the sprite to another sprite buffer, then I push the next frame of the animation to the same sprite buffer.
 
My question is when you are pushing two images to the same sprite buffer does the 2nd push overwrite the 1st push?  Basically what I am trying to do is build up the layers of the sprite buffer and then push this buffer to the TFT screen.
 
What are the correct methods for doing this as I think the 2nd push is nuking the 1st image push?

commit 55e97ffe (2020-02-16): Fix #539