Bodmer/TFT_eSPI #542: Rotating a triangle

stephensaid, 2020-02-10:
Hi,
I do not think there is a rotation functionality for a triangle. I need to rotate a small triangle to indicate the wind direction so I came up with this:

```
#include <TFT_eSPI.h>

TFT_eSPI tft = TFT_eSPI();
// xpos, ypos: center of triangle
// h, w: height and width
// r: angle of rotation
// color: color of triangle
void drawArrow( int32_t xpos, int32_t ypos, int32_t h, int32_t w, int32_t r, uint32_t color) {
  int32_t x0, y0, x1, y1, x2, y2;
  x0 = xpos + (w / 2);
  y0 = ypos + (h / 2);
  x1 = xpos - (w / 2);
  y1 = ypos + (h / 2);
  x2 = xpos;
  y2 = ypos - (h / 2);

  if ( r > 360 || r < -360 ) r = 0;
  if ( r != 0 ) {
    int32_t _x0 = cos(r) * (x0 - xpos) - sin(r) * (y0 - ypos) + xpos;
    int32_t _y0 = sin(r) * (x0 - xpos) + cos(r) * (y0 - ypos) + ypos;
    int32_t _x1 = cos(r) * (x1 - xpos) - sin(r) * (y1 - ypos) + xpos;
    int32_t _y1 = sin(r) * (x1 - xpos) + cos(r) * (y1 - ypos) + ypos;
    int32_t _x2 = cos(r) * (x2 - xpos) - sin(r) * (y2 - ypos) + xpos;
    int32_t _y2 = sin(r) * (x2 - xpos) + cos(r) * (y2 - ypos) + ypos;

    x0 = _x0;
    y0 = _y0;
    x1 = _x1;
    y1 = _y1;
    x2 = _x2;
    y2 = _y2;
  }

  Serial.printf ("r: %i, x0: %i, y0: %i, x1: %i, y1: %i, x2: %i, y2: %i \n", r, x0, y0, x1, y1, x2, y2);

  tft.fillTriangle(x0, y0, x1, y1, x2, y2, color);
}

void setup() {
  Serial.begin(115200);
  delay(100);
  tft.init();
  tft.setRotation(1);
  tft.fillScreen(TFT_BLACK);
  for (int i = 0; i <= 360; i++) {
    drawArrow(160, 120, 120, 100, i, TFT_GREEN);
    delay(1000);
    drawArrow(160, 120, 120, 100, i, TFT_BLACK);
  }
}

void loop() {}
```

Thing is that, although the triangle is rotating, it is not rotating 1 degree at a time.

I referred to this webpage for the formula: https://homepages.inf.ed.ac.uk/rbf/HIPR2/rotate.htm

Could you have a look at this? I think if we can make this work properly, it should go into the library unless there is something which I missed.

Thanks in advance

Bodmer, 2020-02-10:
The sine and cosine functions operate with angles in radians not degrees.

The library already provides a more general purpose Sprite rotation capability.

closed 2020-02-10 by Bodmer

daxliniere, 2021-12-29:
Hey @Bodmer, you mentioned the library has a way to rotate sprites, but I can't find any documentation on it. I, too, have a triangle I need to rotate (around the apex).

Could you please point me in the right direction for this so I can read about it?

All the best,
Dax.

daxliniere, 2021-12-29:
Found it in the Examples. Cheers!!

SaKiEQ, 2023-06-13:
Thx - very useful ! Needed to add a heading indicator, that actual is within a sprite - so rotating the sprite was not really an option. Using another sprite made no sense in my case, as i am already quiet busy on memory and processing. 

Degrees convert to radians by multiplying by π/180 apparently.