Bodmer/TFT_eSPI #545: 3.5" TFT LCD ILI9486 Screen Mirrored in Portrait Mode + ILI9486 drivers not working

RmkJersey, 2020-02-14:
My screen is displaying mirrored in portrait mode and I have tested all 7 rotation modes.

The only setting that displays the screen correctly is landscape set to tft.setRotation(7);

Also, I have had to use the Setup13_ILI9481_Parallel.h in order to get the screen to work as none of the ILI9486 drivers appear to work, so what am I doing wrong.

Here are the pin definitions that I am using:

#define ESP32_PARALLEL

#define ILI9481_DRIVER

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18

// ESP32 pins used for UNO format board
#define TFT_RD   23
#define TFT_WR   19  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_DC   18  // Data Command control pin - must use a pin in the range 0-31
#define TFT_CS   33  // Chip select control pin
#define TFT_RST  32  // Reset pin

#define TFT_D0   12  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   13  // so a single register write sets/clears all bits
#define TFT_D2   26
#define TFT_D3   25
#define TFT_D4   21
#define TFT_D5   5
#define TFT_D6   27
#define TFT_D7   14

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

RmkJersey, 2020-02-14:
I got it to work with the RPI_ILI9486_DRIVER but had to add define: #define ESP32_PARALLEL

Not sure why this made it work but it did.

Bodmer, 2020-02-15:
You seem to have defined the pins for both SPI and 8 bit parallel displays, so this is going to cause library conflicts. If you have a parallel display then SPI pins should not be defined. The library supports SPI base RPi displays only.

If it works for you then it is more by a lucky sequence of compile options and thus future changes to the library could cause problems.

It may be the case that you have an ILI9486 display. Check the vendors web page.

closed 2020-02-15 by Bodmer

RmkJersey, 2020-02-15:
Many thanks for getting back so quickly.
 
So please could you tell me which pin definitions should be defined as I'm not sure which ones to set?

Richard

Bodmer, 2020-02-15:
See #548

RmkJersey, 2020-02-16:
Do you want me to send you the manufacturer manual which has all of the timing data?