Bodmer/TFT_eSPI #547: Incompatible Examples

dave-greeko, 2020-02-15:
Hi There,
First off thank you so much for your work. I am wondering what's the difference between TFT_HX8357 and TFT_eSPI. I have HiLetgo 3.5" IPS TFT LCD Display ILI9486/ILI9488 with Arduino Mega2560 that's working perfectly with the awesome TFT_HX8357 library but when I tried the TFT_eSPI I get an error compiling it under Arduino IDE. In addition, the IDE is listing the examples of the library under "INCOMPATIBLE" section and the compiler is throwing an error. Finally, when I compile it with PlatformIO I get no errors but just white screen. Here is the Arduino IDE error:

**`Arduino: 1.8.10 (Windows 7), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

Multiple libraries were found for "TFT_eSPI.h"
 Used: C:\Users\main\Documents\Arduino\libraries\TFT_eSPI
Multiple libraries were found for "SPI.h"
 Used: E:\Program
In file included from C:\Users\main\Documents\Arduino\libraries\TFT_eSPI\examples\480 x 320\Free_Font_Demo\Free_Font_Demo.ino:28:0:

C:\Users\main\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:120:12: fatal error: FS.h: No such file or directory

   #include <FS.h>
            ^~~~~~
compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`**

Best Regards,
Dave

Bodmer, 2020-02-15:
TFT_HX8357 is for 16 bit parallel boards, TFT_eSPI is targeted at 32 bit processors with greater RAM and FLASH capacity so is not recommended for the Mega.

The latest commit will compile for a Mega but only the simple examples will run on compatible (SPI and 8 bit parallel) displays and since the library is not optimised for the Mega it will run very slowly.

closed 2020-02-15 by Bodmer

dave-greeko, 2020-02-15:
Many thanks for the clarifications.