Bodmer/TFT_eSPI #548: ReadRect() not working on Kuman (ILI9486) 3.5 inch Tft Touch Screen LCD Display

RmkJersey, 2020-02-15:
Does the Kuman 3.5" screen had the ability to read the screen with readRect function as I have tried everything but always get the same read results irrespective of what is on the screen?  I also tried the readPixel function and the same issue!

Also, which is the correct driver to use for this display as all of the drivers are prefixed with _"RPI" and this is not an RPI screen?

[image]

RmkJersey, 2020-02-15:
I have just tested the MCUFRIEND_kbv ReadPixel example and it reads the screen correctly.  I tested it on the Arduino Mega 2560 dev board.

Here is the code:

#define LCD_CS A3 // Chip Select goes to Analog 3
#define LCD_CD A2 // Command/Data goes to Analog 2
#define LCD_WR A1 // LCD Write goes to Analog 1
#define LCD_RD A0 // LCD Read goes to Analog 0
#define LCD_RESET A4 // Can alternately just connect to Arduino's reset pin

#include <Adafruit_GFX.h> // Hardware-specific library
#include <MCUFRIEND_kbv.h>
MCUFRIEND_kbv tft;

//#include <Adafruit_TFTLCD.h> // Hardware-specific library
//Adafruit_TFTLCD tft(LCD_CS, LCD_CD, LCD_WR, LCD_RD, LCD_RESET);
//Adafruit_TFTLCD tft;

// Assign human-readable names to some common 16-bit color values:
#define	BLACK   0x0000
#define	BLUE    0x001F
#define	RED     0xF800
#define	GREEN   0x07E0
#define CYAN    0x07FF
#define MAGENTA 0xF81F
#define YELLOW  0xFFE0
#define WHITE   0xFFFF

void setup(void);
void loop(void);

uint16_t g_identifier;

void setup(void) {
    tft.reset();
    g_identifier = tft.readID();

    Serial.begin(9600);
    Serial.print("readID = 0x");
    Serial.println(g_identifier, HEX);
    if (g_identifier == 0xFFFF) g_identifier = 0x9341;
    if (g_identifier == 0) {
        Serial.println("Unknown ID");
        while (1);
    }

    tft.begin(g_identifier);
    tft.setRotation(0);
}

uint16_t colors[] = { 
    BLACK, BLUE 
};

void colordump(uint16_t x, uint16_t y)
{
    uint16_t pixel, pixels[32];
    char i, j, buf[20], dirty;
    uint8_t wid = (tft.width() - 9 * 6)/ (5 * 6), ht = (tft.height() / 8) - 1;
    tft.setTextColor(WHITE);
    tft.setTextSize(1);
    for (j = 0; j < ht; j++) {
        sprintf(buf, "%3d,%3d:", x, y + j);
        tft.print(buf);
        dirty = 1;
        for (i = 0; i < wid; i++) {
#if 1 && defined(MCUFRIEND_KBV_H_)        
            if (dirty) tft.readGRAM(x, y + j, pixels, wid, 1);
            dirty = 0;
            pixel = pixels[i];
#else
            pixel = tft.readPixel(x + i, y + j);
#endif
            tft.print(" ");
            if (pixel == WHITE) tft.setTextColor(GREEN);
            sprintf(buf, "%04X", pixel);
            tft.print(buf);
            tft.setTextColor(WHITE);
        }
        tft.println();
    }
}

uint16_t bgr(uint16_t color)
{
    return ((color & 0xF800) >> 11) | (color & 0x7E0) | (color & 0x1F) << 11;
}

void duffcolor(uint16_t color)
{
    uint16_t pixel, x, y;
    char done, buf[20];
    uint16_t BGR = bgr(color);
    for (done = 0, y = 0; y < 320 && !done; y++) {
        for (x= 0; x < 240; x++) {
            //            pixel = readxy(x, y);
            pixel = tft.readPixel(x, y);
            if (pixel != BGR) {
                done = 1;
                sprintf(buf, "0x%04X @ %d, %d", pixel, x, y);
                tft.println(buf);
                break;
            }
        }
    }
}

uint8_t aspect;
char *aspectname[] = {
    "PORTRAIT", "LANDSCAPE", "PORTRAIT_REV", "LANDSCAPE_REV" 
};

void loop(void) 
{
    uint16_t iter, color;
    char buf[80];
    aspect = (aspect + 1) & 3;
    tft.setRotation(aspect);
    for (iter = 0; iter < sizeof(colors) / sizeof(uint16_t); iter++) {
        color = colors[iter];
        tft.fillScreen(color);
        tft.setTextColor(WHITE);
        tft.setTextSize(1);
        tft.setCursor(0, 0);
        sprintf(buf, "ID=0x%04X Background=%04X %s",
           tft.readID(), color, aspectname[aspect]);
        tft.println(buf);
        colordump(6 * 6, 0);
        //    duffcolor(color);
        delay(5000);
    }
}

Bodmer, 2020-02-15:
The suggested setup is:
```
#define ESP32_PARALLEL

#define ILI9486_DRIVER

// ESP32 pins used for UNO format board
#define TFT_CS   33  // Chip select control pin
#define TFT_DC   15  // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST  32  // Reset pin

#define TFT_WR    4  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD    2

#define TFT_D0   12  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   13  // so a single register write sets/clears all bits
#define TFT_D2   26
#define TFT_D3   25
#define TFT_D4   17
#define TFT_D5   16
#define TFT_D6   27
#define TFT_D7   14

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT
```
Step 1 is then to see if you can write to the display OK. If writes are OK then reads should work. I do not have one of these displays and there may be timing issues, see #539.

RmkJersey, 2020-02-15:
Thank you for getting back to me so quickly, it is appreciated.
 
The ESP32-VROOM-32 dev board I have does not have a pin 17, so what alternative can I use for this?
 
Also, I was under the impression that this screen has SPI capability and if so I cannot see your definitions for this in your example,

RmkJersey, 2020-02-15:
I finally got the screen working with the following pin definitions, but the readRect is still bringing random data back from the screen, irrespective of where on the screen I read.  I am only reading a small area 34 * 50 area.
 
I have also modified the line you suggested:
 
#define WR_H GPIO.out_w1tc = (1 << TFT_WR); GPIO.out_w1tc = (1 << TFT_WR); GPIO.out_w1tc = (1 << TFT_WR); GPIO.out_w1tc = (1 << TFT_WR); GPIO.out_w1ts = (1 << TFT_WR); GPIO.out_w1ts = (1 << TFT_WR); GPIO.out_w1ts = (1 << TFT_WR)
 
Adding this line made no difference and the colour test still displayed corrupt font as previously reported.

RmkJersey, 2020-02-16:
I have just ran your TFT_ring_meter example and the numbers in the centre are corrupt, so I'm assuming this is the same issue as the corrupt fonts in the colour_test demo and also the issue I'm having with ReadRect and ReadPixel not working?

I have attached the timing manual from the manufacturer to see if this helps with your investigation.

Your TFT library is superb so I hope I can help resolve some of the issues with you.

[image]

[Driver chip ILI9486.pdf](https://github.com/Bodmer/TFT_eSPI/files/4209823/Driver.chip.ILI9486.pdf)

RmkJersey, 2020-02-16:
Not sure if this is significant but if I change your tft.setTextSize all to 2 and above then the drawstring works, anything below that setting and the text displays incorrectly.

Obviously the numbers are now too large for the demo, but it shows they displayed correctly if set to >= 2

[image]

closed 2020-02-17 by Bodmer

RmkJersey, 2020-02-17:
Has this now fixed the readrect() and readpixel() issue?

Closed #548 <https://github.com/Bodmer/TFT_eSPI/issues/548> .

-

RmkJersey, 2020-02-18:
I can confirm that this is now fixed, excellent work yet again.

commit 8d163618 (2020-02-17): Add parallel TFT readRectRGB(), fix #548