Bodmer/TFT_eSPI #550: ESP32 and the 3,5in TFT featherwing

DavidTruyens, 2020-02-17:
Hi,

I'm using [this adafruit shield](https://learn.adafruit.com/adafruit-3-5-tft-featherwing/tft-graphics-test) for testing because it uses the same driver as my real project. But I can't get the eSPI libery working. It uses the HX8357D library. So within Platformio I'm using these settings:

```

[env:featheresp32]
platform = espressif32
board = featheresp32
framework = arduino
monitor_speed = 115200

lib_deps = 
    TFT_eSPI

build_flags =
  -D USER_SETUP_LOADED=1
  -include $PROJECT_LIBDEPS_DIR/$PIOENV/TFT_eSPI_ID1559/User_Setups/Setup15_HX8357D.h
  -D HX8357D_DRIVER=1                           ; Select HX8357D driver
  -D TFT_WIDTH=480                              ; Set TFT size
  -D TFT_HEIGHT=320
  -D TFT_MISO=19                                ; Define SPI pins
  -D TFT_MOSI=18
  -D TFT_SCLK=5
  -D TFT_CS=15
  -D TFT_DC=33                                  ; Data/Comand pin
  -D TFT_RST=-1                                 ; Reset pin
  -D LOAD_GLCD=1                                ; Load Fonts
   ;-D SPI_FREQUENCY=24000000                    ; Set SPI frequency 27000000  
```

If I add theses diagnostics (from Adafruit):
```
  tft.init();

   // read diagnostics (optional but can help debug problems)
  uint8_t x = tft.readcommand8(HX8357_RDPOWMODE);
  Serial.print("Display Power Mode: 0x"); Serial.println(x, HEX);
  x = tft.readcommand8(HX8357_RDMADCTL);
  Serial.print("MADCTL Mode: 0x"); Serial.println(x, HEX);
  x = tft.readcommand8(HX8357_RDCOLMOD);
  Serial.print("Pixel Format: 0x"); Serial.println(x, HEX);
  x = tft.readcommand8(HX8357_RDDIM);
  Serial.print("Image Format: 0x"); Serial.println(x, HEX);
  x = tft.readcommand8(HX8357_RDDSDR);
  Serial.print("Self Diagnostic: 0x"); Serial.println(x, HEX); 
```
I get this as a response:
```
Display Power Mode: 0x0
MADCTL Mode: 0x0
Pixel Format: 0x0
Image Format: 0x0
Self Diagnostic: 0x0
```

If I just run the adafruit featherwing graphics test is works fine. Maybe something to do with the SPI mapping?

Thx,
David

Bodmer, 2020-02-17:
Don't include both Setup15_HX8357D.h and define your own settings.
Preferably use a pin to drive the RST line.

DavidTruyens, 2020-02-17:
Yes, thanks! It's working now.

closed 2020-02-17 by Bodmer