Bodmer/TFT_eSPI #552: Example Comparing SPI to TFT_PARALLEL_8_BIT

32teeth, 2020-02-19:
Not an issue
More a request

Do you have an example comparing SPI to 8 Bit?
Pros?
Cons?

Thanks

Bodmer, 2020-03-01:
In terms of the physical hardwaew, clearly SPI uses less GPIO pins, so this leaves more free for other hardware interfacing.

From the performance aspect then 8 bit parallel is about 1.7x faster on the ESP32 compared to SPI (clocked at 40MHz). If you need to read from the TFT CGRAM then 8 bit parallel is more likely to work and is faster than SPI. Some displays, such as the MHS 4.0 can be reliably clocked at 80MHz by the ESP32 and then this gives a slightly faster performance for SPI compared to 8 bit parallel.

Not sure what you mean by "example" but for performance the graphicstest example is often used,
example performances:

ESP32 parallel ILI9341
Benchmark,                Time (microseconds)
Screen fill (5 times),    100717
Text,                     6594
Lines,                    39054
Horiz/Vert Lines,         9561
Rectangles (outline),     6094
Rectangles (filled),      160847
Circles (filled),         36709
Circles (outline),        21454
Triangles (outline),      9732
Triangles (filled),       80560
Rounded rects (outline),  12798
Rounded rects (filled),   263675
Total = 747795us
Total = 0.7478s

ESP32 SPI ILI9341 40MHz
Benchmark,                Time (microseconds)
Screen fill (5 times),    166771
Text,                     13485
Lines,                    78008
Horiz/Vert Lines,         14579
Rectangles (outline),     9247
Rectangles (filled),      346275
Circles (filled),         62127
Circles (outline),        47635
Triangles (outline),      18742
Triangles (filled),       120225
Rounded rects (outline),  24405
Rounded rects (filled),   381134
Total = 1282633us
Total = 1.2826s

closed 2020-03-01 by Bodmer

Derek-K, 2020-03-03:
Hi @Bodmer 

  I was going to ask the same question, seem like someone beat me to it.

> Not sure what you mean by "example" but for performance the graphicstest example is often used,
> example performances:
> 
> ESP32 parallel ILI9341
> ESP32 SPI ILI9341 40MHz

  Specifically, I am planning to connect TFT01 (S6D1121) to ESP8266 in **PARALLEL** mode. I know the driver is not yet available for TFT_eSPI at the moment, however, there is support from UTFT (Arduino). (sidenote, mod maybe require to enable GPIO9 & GPIO10, see http://smarpl.com/content/esp8266-esp-201-module-freeing-gpio9-and-gpio10) Or in a simple sense, use something like MAX7317 (10 port SPI I/O expander)

  I would like to stick with TFT_eSPI since it has a few nice features that's missing from UTFT. (sidenote: someone has made a mod for UTFT to run on ESP8266/ESP32 but for serial S6D1121 only)

  Before I start, I just want to make sure my assumption is correct...

- Currently, TFT_eSPI only supports parallel mode for ESP32 and not ESP8266?
- My best approach is to use ILI9341_Defines.h, ILI9341_Init.h and ILI9341_Rotation.h as the template to start?

   Any additional pointers that may be helpful?

Thanks.

Derek-K, 2022-01-15:
Hey @Bodmer, FYI, I just come across this 'driver' for S6D1121 in parallel mode on ESP32

https://github.com/MinatsuT/esp32_lcd_S6D1121