Bodmer/TFT_eSPI #555: Using TFT_eSPI with multiple applications (and different hardware configurations)

WA2FZW, 2020-02-23:
I used the TFT_eSPI library in a digital VFO application which I started almost a year ago with great success. Now there is another project that has come along that also uses the library, but with a different hardware configuration (SPI GPIO assignments and the addition of a touch screen).

I know how to create different "User_Setups" for the two applications and have done so successfully, but as far as I and the creator of the other application can tell, the only way to switch them is by changing the "User_Setup_Select.h" file.

What we haven't been able to figure out is a way to select the appropriate "User_Setup" from the application as opposed to one having to edit the library files (which non-developers using out programs are apt to mess up)!

Any thoughts?

Thanks,
John
WA2FZW

WA2FZW, 2020-02-24:
Well, I may have answered my own question! Using the Arduino IDE, I added the following code to one of the configuration headers for my VFO program and it seems to work; at least with the ILI9341 display. I still need to test with at least one more display and we need to test the other application which probably won't happen for a while as the author is on the road.

[code]
/*
 *	The following definitions control the behavior of the TFT_eSPI library. By defining the
 *	"USER_SETUP_LOADED" symbol, the library will NOT load any of its "User_Setup" files but
 *	rather use the settings we define here.
 */

#define	USER_SETUP_LOADED

/*
 *	The width and height have to be defined in portrait mode, which is the opposite of how
 *	the program deals with the display which will make other places where we refer to the
 *	width and height a bit confusing!
 */

#define	TFT_WIDTH	240
#define	TFT_HEIGHT	320

/*
 *	Un-comment the appropriate display driver type.
 */

	#define ILI9341_DRIVER	// Only tested this display so far
//	#define ST7735_DRIVER	// Might need additional parameters for this display

#define TFT_MOSI 23	// Sends data to the display
#define TFT_SCLK 18	// Display clock line
#define TFT_CS    5	// Display chip select control pin
#define TFT_DC    2	// Display data/command control pin
#define TFT_RST  -1	// Set TFT_RST to -1 if display RESET is connected to ESP32 board RST

#define LOAD_GLCD	// Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2	// Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4	// Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6	// Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7	// Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8	// Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.

#define LOAD_GFXFF	// FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

#ifdef	ILI9341_DRIVER				// Using ILI9341?
	#define SPI_FREQUENCY	40000000	// Max speed for the ILI9341
#endif

#ifdef	ST7735_DRIVER				// Using ST7735?
	#define SPI_FREQUENCY	27000000	// Max speed for the ST7735
#endif

#define SPI_READ_FREQUENCY	20000000	// Not used in the VFO
#define SPI_TOUCH_FREQUENCY	 2500000	// Not used in the VFO
[/code]

WA2FZW, 2020-02-24:
Well, I may have answered my own question! Using the Arduino IDE, I added the following code to one of the configuration headers for my VFO program and it seems to work; at least with the ILI9341 display. I still need to test with at least one more display and we need to test the other application which probably won't happen for a while as the author is on the road.

/*
 *	The following definitions control the behavior of the TFT_eSPI library. By defining the
 *	"USER_SETUP_LOADED" symbol, the library will NOT load any of its "User_Setup" files but
 *	rather use the settings we define here.
 */

#define	USER_SETUP_LOADED

/*
 *	The width and height have to be defined in portrait mode, which is the opposite of how
 *	the program deals with the display which will make other places where we refer to the
 *	width and height a bit confusing!
 */

#define	TFT_WIDTH	240
#define	TFT_HEIGHT	320

/*
 *	Un-comment the appropriate display driver type.
 */

	#define ILI9341_DRIVER	// Only tested this display so far
//	#define ST7735_DRIVER	// Might need additional parameters for this display

#define TFT_MOSI 23	// Sends data to the display
#define TFT_SCLK 18	// Display clock line
#define TFT_CS    5	// Display chip select control pin
#define TFT_DC    2	// Display data/command control pin
#define TFT_RST  -1	// Set TFT_RST to -1 if display RESET is connected to ESP32 board RST

#define LOAD_GLCD	// Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2	// Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4	// Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6	// Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7	// Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8	// Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.

#define LOAD_GFXFF	// FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

#ifdef	ILI9341_DRIVER				// Using ILI9341?
	#define SPI_FREQUENCY	40000000	// Max speed for the ILI9341
#endif

#ifdef	ST7735_DRIVER				// Using ST7735?
	#define SPI_FREQUENCY	27000000	// Max speed for the ST7735
#endif

#define SPI_READ_FREQUENCY	20000000	// Not used in the VFO
#define SPI_TOUCH_FREQUENCY	 2500000	// Not used in the VFO

Derek-K, 2020-03-03:
Nice. I was trying to figure that out for a while!

Bodmer, 2020-03-03:
The proposed solution will not work when the setup header is attached to a sketch or another library. The Arduino IDE does not permit the #defines in a sketch or another library to be visible to another library, clearly this helps avoid name conflicts.  So you will find the TFT_eSPI library is simply picking up your old setup (i.e the one you previously defined within the TFT_eSPI library).

Settings in "board packages" ARE visible to the sketch and libraries. So manufacturers can have bespoke processor and board specific setups.

In the PlatformIO IDE you can have project specific setups, see Tools/PlatformIO folder.

WA2FZW, 2020-03-03:
We’re using the Arduino IDE, and my approach works fine. It’s now been tested in two different applications. Many of the common libraries not only allow definitions in the application, but require them.
 
JP

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

Bodmer, 2020-03-03:
Interesting. Can you provide a working example.

WA2FZW, 2020-03-03:
There's a working example in the issue thread.

JP

Bodmer, 2020-03-03:
Sadly that does not work for me, so I would like to understand more about the setup you are using. Perhaps I misunderstand what you mean by "I added the following code to one of the configuration headers for my VFO program", I assumed you meant attach to a sketch, correct?

WA2FZW, 2020-03-03:
Not home now. I'll send more information later.

JP

WA2FZW, 2020-03-03:
You may be right! I did some more testing this evening just to make sure and found that I had left the “#include User_Setup.h” enabled in the “User_Setup_Select.h” file and, in fact, that was where it was picking the correct pin numbers up. No more time to work on it this evening (and probably not tomorrow).
 
Close the issue and if I do come up with a working solution, I’ll post it. Thanks for the help
 
John
WA2FZW

 

https://www.avast.com/antivirus

closed 2020-03-04 by Bodmer