Bodmer/TFT_eSPI #556: PR Request

hasenradball, 2020-02-24:
Hi can I make a Pullrequest?

-> Sorting the Keyword List?

Bodmer, 2020-02-29:
Check Github tutorials, however a sorted keywords list is not something I am interestes in adding. Originally the list was in the same order as that in the header which groups functions by there functionality. You should use the header files as the function reference si you can see the parameters needed and returned type.

Bodmer, 2020-02-29:
Check Github tutorials, however a sorted keywords list is not something I am interestes in adding. Originally the list was in the same order as that in the header files which groups functions by there functionality. You should use the header files as the function reference si you can see the parameters needed and returned type.

closed 2020-02-29 by Bodmer